arkq/bluez-alsa #180: Cannot play stereo

funnyflowerpot, 2018-12-24:
First, thanks a lot for this great piece of software!

Apparently I cannot play stereo sound via my `bluealsa` device. I use Gentoo without Pulseaudio, below is my `.asoundrc`. Please let me know what other output I can provide here! :-)

I tested with two equal WAV files that, where one is encoded with one channel and the other with two. I used `ffmpeg` with `-ac` to create the other file. `aplay` works for the mono-file, but not for the stereo-file. 

Similarly, I created a mono MP3 from a stereo MP3 file. Those and the WAV files yield the same results with `mpv`. When I use `--af=channels=1` for `mpv`, playback works for all files.

How can I convince my setup to play stereo? Thanks!

```
pcm.btheadset {
  type plug
  slave {
    pcm {
      type bluealsa
      device "VE:RY:DE:AD:BE:EF"
      profile "a2dp"
#      delay 1000
    }
  }
  hint {
    show on
    description "Mpow Judge"
  }
}
```

borine, 2019-03-01:
@funnyflowerpot : try launching aplay as:
`LIBASOUND_THREAD_SAFE=0 aplay -D btheadset  filename.wav`
That fixes this issue for me. Which version of alsa-lib are you using? You can tell by using
`aplay --version`

funnyflowerpot, 2019-03-02:
@borine Thanks for getting back to this issue!

Time has passed. I couldn't figure out how to do it with Alsa, and I took this as final reason to install PulseAudio. After some initial struggles I managed to get it working. Practically speaking, I don't have this problem anymore. But I am happy to resolve this issue, if I can help. 

However, when I try to run the command you're suggesting, I get following response:
`ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.1.6-r1/work/alsa-lib-1.1.6/src/dlmisc.c:287:(snd1_dlobj_cache_get) Cannot open shared library /usr/lib64/alsa-lib/libasound_module_pcm_bluetooth.so ((null): /usr/lib64/alsa-lib/libasound_module_pcm_bluetooth.so: cannot open shared object file: No such file or directory)`

I already re-installed bluez, bluez-alsa and alsa-lib. The error message stays the same. I am note sure what changed. Output from your second command is:
`aplay: version 1.1.6 by Jaroslav Kysela <perex@perex.cz>`

borine, 2019-03-03:
@funnyflowerpot  That error message suggests that the bluez-alsa installation is wrong - maybe you need to pass a --prefix option to configure when building bluealsa on your system. Its odd that alsa is looking in /var/tmp for libraries though - I've never seen that before.
I commented on your issue just because i experienced exactly the same mono-only problem when experimenting with bluealsa last week, and was surprised when it turned out to be the alsa thread locking that was the cause, and thought that posting here might help others.
I hope that pulse works out OK for you, I'm continuing to experiment with blue alsa as I think that @Arkq has done a fantastic job here!

funnyflowerpot, 2019-03-03:
@borine I have already checked if that library lingers somewhere else in my file system, but to no avail. I vaguely remember something related to different versions of certain libraries, though. What BlueZ version do you use?

The path with `/var/tmp/portage` was valid during compilation. In Gentoo all packages are compiled locally and the working directory per package is in `/var/tmp/portage`. So that should be fine.

Thanks for the feedback! And yes, I think bluez-alsa is a pretty good piece of software!

closed 2021-06-04 by borine