arkq/bluez-alsa #190: j2ba : a jackd client that directly writes to bluealsa.

egag55, 2019-03-04:
Hello,

Thank you for this project. For users like me that don't use pulseaudio it is realy usefull. Finally i have music from my PC on my BT headphone. Very happy :)

It works ok here with jackd using the alsa_out utility to stream to bluealsa_pcm.
Only point i have atm. is that i have to reconnect once or twice at startup to get an
a2dp connection.

I've looked at the code of the bluealsa_pcm plugin and i think that it won't be to
hard to write a jackd client that outputs directly to bluealsa, instead of using the alsa plugin.  I searched for such a project but couldn't find anything . (I'm in no way an experienced programmer but i'm gonna give it a try anyway ). 

I hope you can give me some info.

To start i'd like to know what audio-format(s) and samplerate(s) are suppoted by bluealsa ? I only see signed short mentioned in the plugin code but there might be more than that ?

And for samplerate(s) all i know is i'm using 48000 here on my PC.

With that info i should be able to write some test code.
Thanks in advance.

Ed.

capoei, 2019-06-15:
Hi.
what would the advantage be?
got alsa_out stable here. but only with "-p 10000".
I doubt there can be anything (much) done about the lag

borine, 2021-06-08:
This question has remained unanswered for quite some time now. If the OP is still interested, I suggest to checkout the bluealsa manual pages and wiki pages for info on supported codecs and audio formats etc. The bluealsa plugin works quite well with jack, there is also a wiki page here about that. Closing.

closed 2021-06-08 by borine