arkq/bluez-alsa #193: bluealsas and bluealsa-aplay sound is non constant

flatmax, 2019-03-12:
I'm using the system as a way to play audio from my phone to a Pi with an I2S sound card.

Bluez5 is setup and I have the following commands for bluez-alsa :

* bluealsa -i hci0 -p a2dp-sink
* bluealsa-aplay -d hw:0  -v  00:00:00:00:00:00

My android phone connects and the audio is routed to the Pi. I can hear the audio coming through in patches from the phone, however long gaps between. It is stereo.

You can listen here :
https://drive.google.com/file/d/1922x6iXofnortUnxkr511fsLj8-GjROn/view?usp=sharing

flatmax, 2019-03-12:
I've also done a test with altering the buffer and period time : 

bluealsa-aplay -d hw:0 --profile-a2dp --pcm-period-time=200  --pcm-buffer-time
=100 -v 00:00:00:00:00:00

Still sounds patchy ...

https://drive.google.com/file/d/1PRboBiSfkXn7xHEXbU5e_cuOYtjHkz4u/view?usp=sharing

flatmax, 2019-03-21:
I tested with a USB bluetooth dongle. It works well.

Turns out the problem is with the Pi3 inbuilt bluetooth hardware. I have tried turning wifi off and playing with flow control - but nothing comes of it.

Anyone who has setup the Pi3 using buildroot (not raspbian) with suggestions are welcome.

mbrebion, 2020-01-19:
I've encountered similar problems with Bluetooth streaming from devices to RPIs 3B and 3 zero (I use a hifiberry sound card for decent audio).
I've found two working solutions : 
 - Using a Bluetooth dongle (and disconnect the onboard Bluetooth)
 - disconnecting Wifi as soon as Bluetooth stream is coming. This works well but implies ssh disconnections and so on

Apparently, newer RPIs do not suffer from this problem

borine, 2021-06-21:
This appears to be an issue with RPi firmware and/or UART config. It is very likely fixed by updating firmware and OS. I think it is not a bluez-alsa issue.

closed 2021-06-21 by borine