arkq/bluez-alsa #195: alsa default config file

borine, 2019-03-15:
Is there a specfic reason why the file 20-bluealsa.conf file defines pcm.bluealsa to be of type "plug" ? I struggle with alsa conguration, so please forgive if there is some obvious reason that I have missed.

The reason I ask is that the alsa deadlock bug that requires bluealsa to have the env var LIBASOUND_THREAD_SAFE=0 set appears to be triggered by accessing a pcm ioplug plugin from within the "plug" plugin. If I change the bluealsa conf file so that pcm.bluealsa is defined as type "bluealsa" then I no longer need to export the env var and I see no lockups of bluealsa applications at all. For reference I've attached my 20-bluealsa.conf file.
[20-bluealsa.conf.gz](https://github.com/Arkq/bluez-alsa/files/2971863/20-bluealsa.conf.gz)

Of course I still get lockups if I use a local config such as:
```
pcm.bt {
    type plug
    slave.pcm {
        type bluealsa
    }
}
```
... but then I do also with
```
pcm.pulse2 {
    type plug
    slave.pcm {
        type pulse
    }
}
```
so in that sense bluealsa is no worse than pulse.
If there are no undesirable side-effects, then I think this default conf file could improve the first impressions of new bluealsa users, and possibly resolve some of the open issues.

Arkq, 2019-03-16:
The default usage of `plug` plugin was dedicated by that fact that bluealsa supports only single PCM configuration. So, in order to simplify the usage I've introduced `plug` plugin in the default configuration. I think, that many hardware cards which lacks support for e.g. mono output, or other sampling frequency than 48000 use `plug` plugin as well: `grep -r plug /usr/share/alsa/cards/`. The case is, that the hardware PCM just woks with `plug`. So, the problem is in the alsa-lib... again.

borine, 2019-03-16:
Ah yes, thanks. I only just discovered that with my config, all tracks are being played at 48kHz stereo - so the 44.1kHz sampled files are playing back "speeded up" and mono files do not play at all. After restoring your default config, all files that need to be resampled (44.1kHz or mono for example) play perfectly but any that do not need conversion (48kHz stereo) cause a deadlock before any audio is played. Back to the drawing board for me then ...

closed 2019-03-16 by borine