arkq/bluez-alsa #196: Does record from bluetooth supported now?

AndroiDjo, 2019-03-20:
Hello.
Sorry for my noob question, im new in linux-like systems and im trying to make something like smart speaker on my raspbian. I can use my speaker through a2dp profile successfully, but i cant change profile (hfp-hf, hfp-ofono) to record sound.

Im trying change profile by command:
pi@raspberrypi:~/exp $ bluealsa -p hfp-hf
bluealsa: Couldn't bind controller socket: /var/run/bluealsa/hci0: Address already in use
bluealsa: ../../src/bluez.c:1099: Registering profile: /HFP/HandsFree
bluealsa: ../../src/main.c:299: Starting main dispatching loop
then nothing happens and i have to cancel this process by Ctrl+C

when i try to record sound, i get error:
pi@raspberrypi:~/exp $ arecord -d 5 record.wav
../../../src/asound/../shared/ctl-client.c:108: Connecting to socket: /var/run/bluealsa/hci0
../../../src/asound/../shared/ctl-client.c:238: Getting transport for F4:4E:FD:32:0D:6C type 0x81
ALSA lib ../../../src/asound/bluealsa-pcm.c:714:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA transport: No such device or address
arecord: main:788: audio open error: No such device or address

AndroiDjo, 2019-03-22:
Upd.
Now i know how to record with bluealsa, but i have error:
$ sudo arecord -f S16_LE -D bluealsa:HCI=hci,DEV=F4:4E:FD:32:0D:6C,PROFILE=sco record.wav
../../../src/asound/../shared/ctl-client.c:108: Connecting to socket: /var/run/bluealsa/hci
ALSA lib ../../../src/asound/bluealsa-pcm.c:702:(_snd_pcm_bluealsa_open) BlueALSA connection failed: No such file or directory
arecord: main:788: audio open error: No such file or directory

SeppiR, 2019-03-28:
I am facing the same Problem
aplay works great with my bluetooth devic but arecord gives me the same error as @AndroiDjo 
$ arecord -l  --> list is empty

bryanhughes, 2019-10-31:
I too am facing the same problem. After a lot of struggling, I compiled from source on my Beagle Bone Green (Debian 9.9) without issue, purged pulseaudio and got playback working. Yay!

```
$ arecord -D bluealsa capture.wav
ALSA lib ../../../src/asound/bluealsa-pcm.c:755:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: PCM not found
arecord: main:788: audio open error: No such device
```

Is there more configuration or settings needed to record audio from bluetooth?

Arkq, 2019-11-05:
With the latest commit in the master recording with HFP/HSP in headset mode (incoming call from phone) should be possible.

Also please note, that when SCO is acquired, the A2DP might be muted by the headset, so you can not use SCO and A2DP simultaneously.

> update2 not working

You've made a typo: not "sOc" but "scO" :)

closed 2019-11-05 by Arkq

rkachach, 2020-10-26:
Facing the same problem with the latest version (v3.0.0-9-g63e2eea) running on a RPI 4. I'm able to play wav files when using **a2dp** but I can't capture/record. I get the following error:

```
pi@raspberrypi:~ $ arecord -D jabra capture.wav
D: ../../../src/asound/bluealsa-pcm.c:1004: Getting BlueALSA PCM: CAPTURE 70:BF:92:15:BF:EE a2dp
ALSA lib ../../../src/asound/bluealsa-pcm.c:1006:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: PCM not found
arecord: main:828: audio open error: No existe el dispositivo

```
Is there something i'm missing? my configuration is the following:

defaults.bluealsa.interface "hci0"
defaults.bluealsa.device "70:BF:92:15:BF:EE"
defaults.bluealsa.profile "sco"

```
pcm.jabra {
    type plug
    slave.pcm {
        type bluealsa
        service "org.bluealsa"
        device "70:BF:92:15:BF:EE"
        profile "sco"
    }
    hint {
       show on
       description "Bluetooth Jabra Speaker"
    }
}

```