arkq/bluez-alsa #201: More D-Bus errors

gearhead, 2019-03-31:
I am running this on Arch on a RPi3. My goal is to have  the Pi as a bluetooth sink. Arch uses Systemd and I have built and installed the AUR package:
https://aur.archlinux.org/packages/bluez-alsa-git/
The installed package is advertised as this:
bluez-alsa-git 1.4.0.r9.g1557602-1

bluealsa 'worked' at one point, but no longer does. 

I cannot get the bluealsa service to run. Every time I try, I get this error:
```
 bluealsa[279]: /usr/bin/bluealsa: Couldn't get managed objects: GDBus.Error:org.freedesktop.DBus.Error.AccessDen>
bluealsa[279]: /usr/bin/bluealsa: Couldn't acquire D-Bus name: org.bluealsa
```
When I interrogate the dbus, I get this:

```
Monitoring bus message stream.
‣ Type=method_call  Endian=l  Flags=0  Version=1  Priority=0 Cookie=1
  Sender=:1.50  Destination=org.freedesktop.DBus  Path=/org/freedesktop/DBus  Interface=org.freedesktop.DBus  Member=Hello
  UniqueName=:1.50
  MESSAGE "" {
  };

‣ Type=method_return  Endian=l  Flags=1  Version=1  Priority=0 Cookie=1  ReplyCookie=1
  Sender=org.freedesktop.DBus  Destination=:1.50
  MESSAGE "s" {
          STRING ":1.50";
  };

‣ Type=signal  Endian=l  Flags=1  Version=1  Priority=0 Cookie=56
  Sender=org.freedesktop.DBus  Path=/org/freedesktop/DBus  Interface=org.freedesktop.DBus  Member=NameOwnerChanged
  MESSAGE "sss" {
          STRING ":1.50";
          STRING "";
          STRING ":1.50";
  };

‣ Type=signal  Endian=l  Flags=1  Version=1  Priority=0 Cookie=2
  Sender=org.freedesktop.DBus  Destination=:1.50  Path=/org/freedesktop/DBus  Interface=org.freedesktop.DBus  Member=NameAcquired
  MESSAGE "s" {
          STRING ":1.50";
  };

‣ Type=method_call  Endian=l  Flags=0  Version=1  Priority=0 Cookie=2
  Sender=:1.50  Destination=org.freedesktop.systemd1  Path=/org/freedesktop/systemd1  Interface=org.freedesktop.systemd1.Manager  Member=GetDynamicUsers
  UniqueName=:1.50
  MESSAGE "" {
  };

‣ Type=method_return  Endian=l  Flags=1  Version=1  Priority=0 Cookie=58  ReplyCookie=2
  Sender=:1.2  Destination=:1.50
  UniqueName=:1.2
  MESSAGE "a(us)" {
          ARRAY "(us)" {
                  STRUCT "us" {
                          UINT32 65306;
                          STRING "haveged";
                  };
          };
  };

‣ Type=signal  Endian=l  Flags=1  Version=1  Priority=0 Cookie=5
  Sender=org.freedesktop.DBus  Destination=:1.50  Path=/org/freedesktop/DBus  Interface=org.freedesktop.DBus  Member=NameLost
  MESSAGE "s" {
          STRING ":1.50";
  };

‣ Type=signal  Endian=l  Flags=1  Version=1  Priority=0 Cookie=57
  Sender=org.freedesktop.DBus  Path=/org/freedesktop/DBus  Interface=org.freedesktop.DBus  Member=NameOwnerChanged
  MESSAGE "sss" {
          STRING ":1.50";
          STRING ":1.50";
          STRING "";
  };

‣ Type=method_call  Endian=l  Flags=0  Version=1  Priority=0 Cookie=1
  Sender=:1.51  Destination=org.freedesktop.DBus  Path=/org/freedesktop/DBus  Interface=org.freedesktop.DBus  Member=Hello
  UniqueName=:1.51
  MESSAGE "" {
  };

‣ Type=method_return  Endian=l  Flags=1  Version=1  Priority=0 Cookie=1  ReplyCookie=1
  Sender=org.freedesktop.DBus  Destination=:1.51
  MESSAGE "s" {
          STRING ":1.51";
  };

‣ Type=signal  Endian=l  Flags=1  Version=1  Priority=0 Cookie=58
  Sender=org.freedesktop.DBus  Path=/org/freedesktop/DBus  Interface=org.freedesktop.DBus  Member=NameOwnerChanged
  MESSAGE "sss" {
          STRING ":1.51";
          STRING "";
          STRING ":1.51";
  };

‣ Type=signal  Endian=l  Flags=1  Version=1  Priority=0 Cookie=2
  Sender=org.freedesktop.DBus  Destination=:1.51  Path=/org/freedesktop/DBus  Interface=org.freedesktop.DBus  Member=NameAcquired
  MESSAGE "s" {
          STRING ":1.51";
  };

‣ Type=method_call  Endian=l  Flags=0  Version=1  Priority=0 Cookie=2
  Sender=:1.51  Destination=org.freedesktop.DBus  Path=/org/freedesktop/DBus  Interface=org.freedesktop.DBus  Member=AddMatch
  UniqueName=:1.51
  MESSAGE "s" {
          STRING "type='signal',sender='org.bluez',interface='org.freedesktop.DBus.ObjectManager',member='InterfacesAdded'";
  };

‣ Type=method_return  Endian=l  Flags=1  Version=1  Priority=0 Cookie=3  ReplyCookie=2
  Sender=org.freedesktop.DBus  Destination=:1.51
  MESSAGE "" {
  };

‣ Type=method_call  Endian=l  Flags=0  Version=1  Priority=0 Cookie=3
  Sender=:1.51  Destination=org.freedesktop.DBus  Path=/org/freedesktop/DBus  Interface=org.freedesktop.DBus  Member=AddMatch
  UniqueName=:1.51
  MESSAGE "s" {
          STRING "type='signal',sender='org.bluez',interface='org.freedesktop.DBus.ObjectManager',member='InterfacesRemoved'";
  };

‣ Type=method_return  Endian=l  Flags=1  Version=1  Priority=0 Cookie=4  ReplyCookie=3
  Sender=org.freedesktop.DBus  Destination=:1.51
  MESSAGE "" {
  };

‣ Type=method_call  Endian=l  Flags=0  Version=1  Priority=0 Cookie=4
  Sender=:1.51  Destination=org.freedesktop.DBus  Path=/org/freedesktop/DBus  Interface=org.freedesktop.DBus  Member=AddMatch
  UniqueName=:1.51
  MESSAGE "s" {
          STRING "type='signal',sender='org.bluez',interface='org.freedesktop.DBus.Properties',member='PropertiesChanged',arg0='org.bluez.Device1'";
  };

‣ Type=method_return  Endian=l  Flags=1  Version=1  Priority=0 Cookie=5  ReplyCookie=4
  Sender=org.freedesktop.DBus  Destination=:1.51
  MESSAGE "" {
  };

‣ Type=method_call  Endian=l  Flags=0  Version=1  Priority=0 Cookie=5
  Sender=:1.51  Destination=org.freedesktop.DBus  Path=/org/freedesktop/DBus  Interface=org.freedesktop.DBus  Member=AddMatch
  UniqueName=:1.51
  MESSAGE "s" {
          STRING "type='signal',sender='org.bluez',interface='org.freedesktop.DBus.Properties',member='PropertiesChanged',arg0='org.bluez.MediaTransport1'";
  };

‣ Type=method_return  Endian=l  Flags=1  Version=1  Priority=0 Cookie=6  ReplyCookie=5
  Sender=org.freedesktop.DBus  Destination=:1.51
  MESSAGE "" {
  };

‣ Type=method_call  Endian=l  Flags=0  Version=1  Priority=0 Cookie=6
  Sender=:1.51  Destination=org.bluez  Path=/  Interface=org.freedesktop.DBus.ObjectManager  Member=GetManagedObjects
  UniqueName=:1.51
  MESSAGE "" {
  };

‣ Type=error  Endian=l  Flags=1  Version=1  Priority=0 Cookie=7  ReplyCookie=6
  Sender=org.freedesktop.DBus  Destination=:1.51
  ErrorName=org.freedesktop.DBus.Error.AccessDenied  ErrorMessage="Rejected send message, 2 matched rules; type="method_call", sender=":1.51" (uid=974 pid=1385 comm="/usr/bin/bluealsa ") interface="org.freedesktop.DBus.ObjectManager" member="GetManagedObjects" error name="(unset)" requested_reply="0" destination="org.bluez" (uid=0 pid=272 comm="/usr/lib/bluetooth/bluetoothd ")"
  MESSAGE "s" {
          STRING "Rejected send message, 2 matched rules; type="method_call", sender=":1.51" (uid=974 pid=1385 comm="/usr/bin/bluealsa ") interface="org.freedesktop.DBus.ObjectManager" member="GetManagedObjects" error name="(unset)" requested_reply="0" destination="org.bluez" (uid=0 pid=272 comm="/usr/lib/bluetooth/bluetoothd ")";
  };

‣ Type=method_call  Endian=l  Flags=0  Version=1  Priority=0 Cookie=7
  Sender=:1.51  Destination=org.freedesktop.DBus  Path=/org/freedesktop/DBus  Interface=org.freedesktop.DBus  Member=RequestName
  UniqueName=:1.51
  MESSAGE "su" {
          STRING "org.bluealsa";
          UINT32 0;
  };

‣ Type=error  Endian=l  Flags=1  Version=1  Priority=0 Cookie=8  ReplyCookie=7
  Sender=org.freedesktop.DBus  Destination=:1.51
  ErrorName=org.freedesktop.DBus.Error.AccessDenied  ErrorMessage="Connection ":1.51" is not allowed to own the service "org.bluealsa" due to security policies in the configuration file"
  MESSAGE "s" {
          STRING "Connection ":1.51" is not allowed to own the service "org.bluealsa" due to security policies in the configuration file";
  };

‣ Type=signal  Endian=l  Flags=1  Version=1  Priority=0 Cookie=6
  Sender=org.freedesktop.DBus  Destination=:1.51  Path=/org/freedesktop/DBus  Interface=org.freedesktop.DBus  Member=NameLost
  MESSAGE "s" {
          STRING ":1.51";
  };

‣ Type=signal  Endian=l  Flags=1  Version=1  Priority=0 Cookie=59
  Sender=org.freedesktop.DBus  Path=/org/freedesktop/DBus  Interface=org.freedesktop.DBus  Member=NameOwnerChanged
  MESSAGE "sss" {
          STRING ":1.51";
          STRING ":1.51";
          STRING "";
  };
```

infirit, 2019-03-31:
You need to make sure ``https://github.com/Arkq/bluez-alsa/blob/master/src/bluealsa.conf`` is installed in the dbus system service directory. For me this is ``/etc/dbus-1/system.d`` which I suspect is the same on Arch.

The git pkgbuild has not been updated for the latest changes. Someone should report it out of date.

gearhead, 2019-04-01:
Thanks for the help!.

I checked, that file is in my /etc/dbus-1/system.d/ folder. It appears that the git PKGBUILD grabs the latest from git. Is there something missing that I need to get or edit?

Arkq, 2019-04-01:
>  Is there something missing that I need to get or edit?

When I've checked D-Bus configuration on Ubuntu 18 I had to perform reboot
before D-Bus has grabbed the configuration - restarting D-Bus alone has
seemed not to work (I'm not sure why though). On my Gentoo PC, everything
has worked flawlessly after D-Bus reload.

>

infirit, 2019-04-01:
@gearhead Are you running ``bluealsa`` as root? I would expect it to error like that if you don't.

gearhead, 2019-04-01:
I appreciate the help... I dug through this again. Tried on a fresh install. Discovered that the /etc/dbus-1/system.d/bluetooth.conf file was missing. Do no t know why, but when I installed (and re-installed) bluez-5.50 and bluez-utils, it was never installed. I grabbed one from the official repository (I'll let arch know) and put it in. I edited the resulting file to add the lines:

  <policy user="bluealsa">
    <allow send_destination="org.bluez"/>
  </policy>

again. then rebooted. I get a different but similar response and still the process does not start:
Apr 01 12:13:17 alarm systemd[1]: Started BluezAlsa proxy.
Apr 01 12:13:17 alarm bluealsa[2105]: /usr/bin/bluealsa: Couldn't acquire D-Bus name: org.bluealsa
Apr 01 12:13:17 alarm systemd[1]: bluealsa.service: Main process exited, code=exited, status=1/FAILURE
Apr 01 12:13:17 alarm systemd[1]: bluealsa.service: Failed with result 'exit-code'.

I start to wonder if this be a systemd issue? When I try to run it from the cli, I get nothing. 
/usr/bin/bluealsa never returns with anything and requires a ctrl c to get a prompt back. I edited the service file to comment out the user and group (so it launches as root) and now it runs. Woo Hoo! Now to see if audio comes out and then dig into what I need to do with systemd user and group...

nheir, 2019-04-01:
So starting from 86b174aba7485ec7f6159bf05baf080250adda25, is it mandatory to run bluealsa as root ?

gearhead, 2019-04-01:
I believe my problem was resulting from systemd and not bluezalsa,

Arkq, 2019-04-02:
@nheir by default, yes. However, anyone can change D-Bus policy file in
order to allow other users (groups) to run and connect to bluealsa.

closed 2019-06-01 by Arkq