arkq/bluez-alsa #202: Update mute status on volume update

snehaltan, 2019-04-03:
A2DP Capture Switch is not updated based on volume update. This will create odd behaviour when controlling the volume and mute function. Consider below scenario:
1. BT Sink Device -> bluealsa capture switch - mute -> Bluez -> BT source volume 0
2. BT source volume 10% -> Bluez -> bluealsa capture volume - 10% -> BT Sink Device
Volume is not zero now, but the capture switch was not updated and is still muted. A solution is to update the capture switch everytime a volume update is received on bluez. i.e. mute on 0 volume and unmute on non-zero volume.

Log at end of step 2 above (Note: switch off = muted):
```
amixer -D bluealsa contents '<Device> - A2DP'
../../git/src/shared/bluealsa/ctl-client.c:105: Connecting to socket: /var/run/bluealsa/hci0
../../git/src/shared/bluealsa/ctl-client.c:105: Connecting to socket: /var/run/bluealsa/hci0
../../git/src/shared/bluealsa/ctl-client.c:132: Subscribing for events: 1111111111111111
numid=2,iface=MIXER,name='<Device> Capture Switch'
  ; type=BOOLEAN,access=rw------,values=2
  : values=off,off
numid=1,iface=MIXER,name='<Device> Capture Volume'
  ; type=INTEGER,access=rw------,values=2,min=0,max=127,step=1
  : values=10,10

```

Arkq, 2021-08-04:
I know that it has been a while since this issue was created, but I've just raised a PR (#483) which should fix this issue. However, I'm not sure whether it is the right approach to this problem. The problem is that the Bluetooth specification does not define mute state for Bluetooth audio. It has only a volume property. And setting volume to zero does not mean that it is muted (it might be, but also it might be set to e.g. -100dB which is not mute, but very low PCM power output).

I wonder, whether there are some sound cards which have external (e.g. physical) volume control elements. And what happens if control element of such sound card is muted? E.g. it might be that a mute switch is enabled/disabled. And later, what will happen if someone will change the volume of such element. I suspect that the mute stat will be preserved. So, one will turn physical volume gauge, but the e.g. channel will still be muted until unmuted from physical console or via ALSA control element.

The problem with #483 (in my opinion) is that when the PCM is muted, the volume is set to zero. See this scenario:
1. Set volume to 70% -> you can hear music quite laud 
2. Mute PCM -> there is no audio (OK)
3. Use volume up button on headset -> volume i 5% (it was set to 0% when mutting)

But maybe we should have approach as such:
1. volume is changed from Bluetooth device, the volume is also changed in ALSA mixer (it's already done)
2. PCM is muted from ALSA end point, but the mute information is not forwarded to Bluetooth device (i.e. volume is not set to 0) but stays only in BlueALSA (it is a software mute - BlueALSA already implements mute as a software thing)
3. ALSA mixer can change volume and it's synced with BT device
4. Bluetooth device can change its volume and this info is synced with ALSA element (one will see volume change), but mute will stay as it is
5. user can disable mute in ALSA mixer and the volume will be as previously set by the headset

Alternatively, maybe we should change 4. to:
4. Bluetooth device can change its volume and this info is synced with ALSA element (one will see volume change) **and the PCM will be unmutted**

What do you think?

borine, 2021-08-10:
> The problem with #483 (in my opinion) is that when the PCM is muted, the volume is set to zero

I never use the mute switch, so I am not the best person to give advice on this, however given the lack of response so far I will offer an opinion.

FWIW, I absolutely agree that volume mute and volume level are distinct attributes, and that changing one should not affect the other. Your second approach (steps 1 - 5 above) is therefore my view of how to handle mute. I don't like your final alternative step 4, as it still confuses mute with level.

Arkq, 2021-08-10:
> Your second approach (steps 1 - 5 above) is therefore my view of how to handle mute.

Yes, that should be the best approach in my opinion as well. However, it will not address this issue. Anyway, it will be an improvement in comparison with current approach. Also, apparently there are some headsets which do have physical mute button, e.g.: https://github.com/Arkq/bluez-alsa/issues/298

Arkq, 2021-08-10:
I've decided not to update mute state on volume updates. It's not a perfect approach but consistent in itself - implicit mute state update might be confusing.

commit 7d2a440c (2021-08-10): Do not set volume to zero when muting PCM

Controlling audio volume should have two parameters: gain level (dB)
and mute state. Setting gain level to very low value does not mean that
audio will be muted. PCM power will go asymptotically to zero but will
never reach it. For zeroing PCM power there should be a mute switch.

Unfortunately, Bluetooth specification (A2DP and HFP/HSP) does not
define any standard way of muting audio. Audio volume level is
controlled by a property which goes from 0 to MAX (e.g. 15) and by
convention audio is muted when the volume level is set to zero.

On the other  [... cut]
closed 2021-08-10 by Arkq