arkq/bluez-alsa #203: core dump

gearhead, 2019-04-03:
I built this for my raspberry pi. I finally got past the dbus errors and am now trying to get an audio connection. I can start bluealsa and bluetooth, I can pair, but as soon as I connect, it disconnects. In the journal, I get this:

```
Apr 02 19:37:32 livingrune audit[1210]: ANOM_ABEND auid=0 uid=0 gid=0 ses=1 pid=1210 comm="bluealsa" exe="/usr/bin/bluealsa" sig=11 res=1
Apr 02 19:37:32 livingrune kernel: audit: type=1701 audit(1554251852.041:38): auid=0 uid=0 gid=0 ses=1 pid=1210 comm="bluealsa" exe="/usr/bin/bluealsa" sig=11 res=1
Apr 02 19:37:32 livingrune systemd[1]: Created slice system-systemd\x2dcoredump.slice.
Apr 02 19:37:32 livingrune systemd[1]: Started Process Core Dump (PID 1213/UID 0).
Apr 02 19:37:32 livingrune audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-coredump@0-1213-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Apr 02 19:37:32 livingrune kernel: audit: type=1130 audit(1554251852.131:39): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-coredump@0-1213-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Apr 02 19:37:33 livingrune systemd-coredump[1214]: Process 1210 (bluealsa) of user 0 dumped core.
                                                   
                                                   Stack trace of thread 1210:
                                                   #0  0x0000000076914fee strlen (libc.so.6)
                                                   #1  0x00000000768de4f8 printf_positional (libc.so.6)
Apr 02 19:37:33 livingrune systemd[1]: systemd-coredump@0-1213-0.service: Succeeded.

```

gearhead, 2019-04-03:
I rebuilt without a flag and now it does not core dump

closed 2019-04-03 by gearhead

Arkq, 2019-04-03:
Which flag had caused this coredump?

gearhead, 2019-04-03:
I compiled it with aptx and got this result. I removed that flag and now it seems to load and run