arkq/bluez-alsa #204: Bluealsa is not advertised as A2DP sink

gearhead, 2019-04-03:
I think I have followed the instructions and have everything running. I have a bluealsa service running, no errors. I have bluetooth running no errors. It appears it is advertising correctly from bluetoothctl:
```
[bluetooth]# show
Controller 00:02:72:D3:79:8E (public)
	Name: livingrune
	Alias: livingrune
	Class: 0x00480000
	Powered: yes
	Discoverable: yes
	Pairable: yes
	UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
	UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v1D6Bp0246d0532
	Discovering: no
```
I can pair and trust my phone. When it shows up on the phone, though, it is bluetooth with nothing else. No services show up. no headset nothing. This is with bluez 5.50
```
bluez 5.50-6
bluez-alsa-git 1.4.0.r9.g1557602-1
bluez-libs 5.50-6
bluez-utils 5.50-6
```
Am I missing something? I pair as soon as it pairs, it disconnects. It is 'connected yes', then 'services resoved no' then 'connected no'. Happens every time.
```
[NEW] Device BC:FF:EB:39:E0:5E Moto Z2 Play
Request PIN code
[agent] Enter PIN code: 1234
[CHG] Device BC:FF:EB:39:E0:5E Class: 0x005a020c
[CHG] Device BC:FF:EB:39:E0:5E Icon: phone
[CHG] Device BC:FF:EB:39:E0:5E Connected: yes
[CHG] Device BC:FF:EB:39:E0:5E Modalias: bluetooth:v001Dp1200d1436
[CHG] Device BC:FF:EB:39:E0:5E UUIDs: 00001105-0000-1000-8000-00805f9b34fb
[CHG] Device BC:FF:EB:39:E0:5E UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[CHG] Device BC:FF:EB:39:E0:5E UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Device BC:FF:EB:39:E0:5E UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Device BC:FF:EB:39:E0:5E UUIDs: 00001112-0000-1000-8000-00805f9b34fb
[CHG] Device BC:FF:EB:39:E0:5E UUIDs: 00001115-0000-1000-8000-00805f9b34fb
[CHG] Device BC:FF:EB:39:E0:5E UUIDs: 00001116-0000-1000-8000-00805f9b34fb
[CHG] Device BC:FF:EB:39:E0:5E UUIDs: 0000111f-0000-1000-8000-00805f9b34fb
[CHG] Device BC:FF:EB:39:E0:5E UUIDs: 0000112f-0000-1000-8000-00805f9b34fb
[CHG] Device BC:FF:EB:39:E0:5E UUIDs: 00001132-0000-1000-8000-00805f9b34fb
[CHG] Device BC:FF:EB:39:E0:5E UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device BC:FF:EB:39:E0:5E UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Device BC:FF:EB:39:E0:5E UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Device BC:FF:EB:39:E0:5E ServicesResolved: yes
[CHG] Device BC:FF:EB:39:E0:5E Paired: yes
[CHG] Device BC:FF:EB:39:E0:5E ServicesResolved: no
[CHG] Device BC:FF:EB:39:E0:5E Connected: no
[bluetooth]# trust BC:FF:EB:39:E0:5E
[CHG] Device BC:FF:EB:39:E0:5E Trusted: yes
Changing BC:FF:EB:39:E0:5E trust succeeded
```

Arkq, 2019-04-03:
It's really hard to say without bluealsa logs. Please, rebuild with debug
enabled '--enable-debug' and post the logs here. Sometimes I also
experience such a behaviour, it almost always help to remove BT device and
pair it once again.

gearhead, 2019-04-03:
will do

gearhead, 2019-04-03:
Ok, so I built it again and tried again (different hardware, but same software) and get the same result:
```
root@deskrune(rw):~# bluetoothctl
Agent registered
[bluetooth]# power
Missing on/off argument
[bluetooth]# power on
Changing power on succeeded
[bluetooth]# show
Controller 00:1F:81:00:08:30 (public)
        Name: deskrune
        Alias: deskrune
        Class: 0x00480000
        Powered: yes
        Discoverable: no
        Pairable: yes
        UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0532
        Discovering: no
[bluetooth]# discoverable yes
Changing discoverable on succeeded
[CHG] Controller 00:1F:81:00:08:30 Discoverable: yes
[bluetooth]# default-agent
Default agent request successful
[NEW] Device BC:FF:EB:39:E0:5E BC-FF-EB-39-E0-5E
Request PIN code
[agent] Enter PIN code: 1234
[CHG] Device BC:FF:EB:39:E0:5E Class: 0x005a020c
[CHG] Device BC:FF:EB:39:E0:5E Icon: phone
[CHG] Device BC:FF:EB:39:E0:5E Connected: yes
[CHG] Device BC:FF:EB:39:E0:5E Name: Moto Z2 Play
[CHG] Device BC:FF:EB:39:E0:5E Alias: Moto Z2 Play
[CHG] Device BC:FF:EB:39:E0:5E Connected: no
[CHG] Device BC:FF:EB:39:E0:5E Paired: yes
[CHG] Device BC:FF:EB:39:E0:5E Connected: yes
[CHG] Device BC:FF:EB:39:E0:5E Connected: no
[bluetooth]# trust  BC:FF:EB:39:E0:5E
[CHG] Device BC:FF:EB:39:E0:5E Trusted: yes
Changing BC:FF:EB:39:E0:5E trust succeeded
[bluetooth]# exit
root@deskrune(rw):~# bluealsa-aplay BC:FF:EB:39:E0:5E
bluealsa-aplay: ../src/shared/ctl-client.c:108: Connecting to socket: /var/run/bluealsa/hci0
bluealsa-aplay: ../src/shared/ctl-client.c:108: Connecting to socket: /var/run/bluealsa/hci0
bluealsa-aplay: ../src/shared/ctl-client.c:135: Subscribing for events: 111
bluealsa-aplay: aplay.c:711: Starting main loop
```
From the journal:
```
Apr 03 09:32:03 deskrune bluealsa[1060]: /usr/bin/bluealsa: bluez.c:836: Registering endpoint: /hci0/A2DP/APTX/Source/1
Apr 03 09:32:03 deskrune bluealsa[1060]: /usr/bin/bluealsa: ctl.c:515: Starting controller loop: hci0
Apr 03 09:32:03 deskrune bluealsa[1060]: /usr/bin/bluealsa: bluez.c:836: Registering endpoint: /hci0/A2DP/MPEG24/Source/1
Apr 03 09:32:03 deskrune bluealsa[1060]: /usr/bin/bluealsa: bluez.c:836: Registering endpoint: /hci0/A2DP/SBC/Source/1
Apr 03 09:35:47 deskrune bluealsa[1060]: /usr/bin/bluealsa: ctl.c:596: Received new connection: 11
Apr 03 09:35:47 deskrune bluealsa[1060]: /usr/bin/bluealsa: ctl.c:609: New client accepted: 11
Apr 03 09:35:47 deskrune bluealsa[1060]: /usr/bin/bluealsa: ctl.c:634: +-+-
Apr 03 09:35:47 deskrune bluealsa[1060]: /usr/bin/bluealsa: ctl.c:596: Received new connection: 12
Apr 03 09:35:47 deskrune bluealsa[1060]: /usr/bin/bluealsa: ctl.c:609: New client accepted: 12
Apr 03 09:35:47 deskrune bluealsa[1060]: /usr/bin/bluealsa: ctl.c:634: +-+-
Apr 03 09:35:47 deskrune bluealsa[1060]: /usr/bin/bluealsa: ctl.c:634: +-+-
Apr 03 09:36:34 deskrune bluealsa[1060]: /usr/bin/bluealsa: ctl.c:539: Client closed connection: 11
Apr 03 09:36:34 deskrune bluealsa[1060]: /usr/bin/bluealsa: ctl.c:634: +-+-
Apr 03 09:36:34 deskrune bluealsa[1060]: /usr/bin/bluealsa: ctl.c:539: Client closed connection: 12
Apr 03 09:36:34 deskrune bluealsa[1060]: /usr/bin/bluealsa: ctl.c:634: +-+-
```
 The phone never shows connected to a bluetooth audio sink and when I play music, it comes out the phone. 

I am running bluealsa as a service as root and am making all these connections and such as root. Could this be the problem?

Arkq, 2019-04-03:
Bluealsa by default exports source profiles only. In order to connect phone
you need sink profiles (e.g. a2dp-sink). Run 'bluealsa --help' for more
info.

gearhead, 2019-04-03:
That was not clear to me. I did that and now can get a BT audio conn3ection. Thanks for the help. Now to automate things a bit as CLI is a bit clunky.

gearhead, 2019-04-05:
it works as advertised. I was just a bit slow in figuring out how to get it to work. I needed a --profile=a2dp-sink added to show the sink feature at pairing

closed 2019-04-05 by gearhead