arkq/bluez-alsa #206: How to set the correct volume control

gearhead, 2019-04-05:
On my Pis I have an i2s card set up and would like to use hardware volume control. For other programs, the alsa settings are ( for example shairport-sync):
output_device="hw:0"; // alsa_output_device
mixer_control_name="Digital"; // alsa_mixer_control_name

The key is the mixer control name.
Maybe this is already enabled but I have just not found it.

borine, 2019-04-07:
type the shell comand:
```amixer -D bluealsa scontrols```
That should tell you the name of the bluealsa mixer control.

borine, 2019-04-07:
Ah, on second thoughts I guess you are not outputting sound to a bluetooth speaker: you want to control the volume on the i2s device using bluealsa-aplay as an input device - in which case ignore my previous comment!
I think that bluealsa-aplay does not have any volume controls, but I could be wrong. Having re-read your post I realise I am not qualified to comment on it - sorry.

joerg-krause, 2019-04-08:
bluez-alsa does not have hardware volume control, yet.

hifiberry, 2020-01-17:
Is anybody working on this?

borine, 2021-06-04:
see PR #390

hifiberry, 2021-06-04:
I know this, but it seems nothing is going on on this anymore and the original author don't want to spent more time on this. Unfortunately I don't have the skills to work on complex C-programming like this.

friefi, 2021-06-15:
I seem to have the same issues. In my case I would like to use bluealsa together with gmrender-resurrect to implement an audio port for in-room assistance. There are 2 instances of gmrender running on my Pi, one for streaming audio to a DAC (Infineon Merus) and one for streaming audio to any bluetooth speaker device (e.g. headphones) attached to the built-in bluetooth adapter of the Pi. I also want to be able to link a bluetooth audio source (e.g. smartphone) to deliver audio to the DAC. So in the end I'm using bluealsa as sink and source.

First, when I started on a fresh system (Raspian Buster on Pi 3B) to setup bluealsa as source only, I was able to control the volume of audio delivered by the smartphone to the DAC through the volume rockers of the phone as expected. Then I set up bluealsa additionally to delivering audio from gmrender to attached bluetooth headphones, which was working also like a charm. But once this was in place, the other use case didn't work as expected anymore. What ever I do, the volume of audio coming from the smartphone is at a ridiculously high level even if the volume slider of the phone is at the very first level above muted. I tried to decrease the volume capture level of the A2DP source on the Raspberry Pi, but that only limits the absolute maximum volume of the audio going to bluealsa, the volume resolution doesn't change.

The strange thing is that, what ever I do (e.g. uninstalling bluealsa, re-pair the bluetooth devices, purge alsa mixer settings and recreate them), the issue stays the same. So the question is, where or if the (hardware) volume settings from the phone are stored? Are they synchronized somehow when bluealsa starts to capture audio from the phone?

I also tried to use the bluealsa-aplay version from the "aplay" branch, but so far I had no luck. Are the new command line options for mixer controlling provided by this extended version to be intended to fix my problems? If so, could somebody help out to guide me in the right directions with this?

If needed, I can provide my settings for ALSA (/etc/asound.conf) and the used options for bluealsa and bluealsa-aplay.

Thanks!

Arkq, 2021-07-28:
PR #390 has been merged, I'm closing this issue.

closed 2021-07-28 by Arkq