arkq/bluez-alsa #208: buffer under run

gearhead, 2019-04-14:
I am able to play 'high quality audio" (aac) from my phone. The Pi is a Pi3, I am running Arch Linux. I am using the hciattach command and can get paired and can stream audio. I paired a tablet to the same Pi3. It pairs and does not show the 'high quality audio' feature of the sink. When I try to play from this device, the music gets out of sync. It starts playing OK then after a few seconds it gets really fast then starts skipping. I see this behavior with streaming sources (youtube) or music (mp3 or flac) on the tablet. The log on the pi shows:
`bluealsa-aplay[1519]: /usr/bin/bluealsa-aplay: aplay.c:469: An underrun has occurred`
It is strange that aac works, but regular audio does not.

borine, 2019-04-14:
Sorry for the "me too" post, but when I saw this I re-built bluealsa without aac support and now I am getting exactly the same behaviour from bluealsa-aplay as @gearhead on both RPi Zero W and RPi 3B+ - after a few seconds of playback there is a pause, then a brief burst of speeded-up playback, then another pause ..., etc.
Rebuild with --enable-aac and the problems all go away.
I'm using latest master branch of bluez-alsa on latest Raspbian Stretch Lite with alsa 1.1.3 and bluetoothd 5.43

gearhead, 2019-04-14:
mine is built with aac support enabled:
```
 	./configure --prefix=/usr \
	            --sysconfdir=/etc \
                    --enable-ofono \
	            --enable-aac \
                    --enable-debug

```
My version lists as: 1.4.0.r9.g1557602
```

# bluealsa --help
Usage:
  bluealsa [OPTION]...

Options:
  -h, --help		print this help and exit
  -V, --version		print version and exit
  -B, --dbus=NAME	D-Bus service name suffix
  -S, --syslog		send output to syslog
  -i, --device=hciX	HCI device to use
  -p, --profile=NAME	enable BT profile
  --a2dp-force-mono	force monophonic sound
  --a2dp-force-audio-cd	force 44.1 kHz sampling
  --a2dp-keep-alive=SEC	keep A2DP transport alive
  --a2dp-volume		control volume natively
  --aac-afterburner	enable afterburner
  --aac-vbr-mode=NB	set VBR mode to NB

Available BT profiles:
  - a2dp-source	Advanced Audio Source (AAC, SBC)
  - a2dp-sink	Advanced Audio Sink (AAC, SBC)
  - hfp-ofono	Hands-Free handled by oFono (hf & ag)
  - hfp-hf	Hands-Free (v1.7)
  - hfp-ag	Hands-Free Audio Gateway (v1.7)
  - hsp-hs	Headset (v1.2)
  - hsp-ag	Headset Audio Gateway (v1.2)

```

gearhead, 2019-04-16:
This seems to be caused by the Pi onboard bluetooth and the hciattach settings. I now have smooth playback with the B3+ and some devices on the B3, but so far nothing on the Zero. Still working on it. I am running Arch Linux on my Pis and am using a pi-bluetooth user compiled package to get these going. Bluealsa works fine on all my Pis without onboard BT and a BT dongle.

borine, 2019-04-29:
@gearhead - any more progress? I've found that results depend greatly on the audio source used. With a linux box running pulseaudio or bluealsa as the source (I tried both rpi and intel laptop) I get good results on a pi3B+ using bluealsa-aplay as the sink. But using an android phone as the source I get the underruns and speed changes that you see.
For comparison, I removed bluealsa from the pi3B+ sink and installed pulseaudio with pulseaudio-module-bluetooth. Guess what - the results are exactly the same as with bluealsa.

So for me that proves that bluealsa is not to blame, and I suspect that there is little or nothing that bluealsa can do about it. This issue needs to be taken to a rpi forum (or possibly a bluez forum) where more specialist knowledge may be found.

gearhead, 2019-04-30:
I have not had a lot of time recently. I find the same thing, though. I have 2 phones which play fine on the RPi with this. I tried an older Samsung tablet and get choppy playback on the B3, but works fine on the B3+ (with onboard BT). I have had no problems getting a BTUSB device working. The biggest challenge is getting the onboard BT working with this. I find that the deprecated hciattach is needed to get proper polayback and that btattach leads to choppy playback as does the alarm-bluetooth-raspberrypi package. I feel that BT on Linux has been given a back seat and it is a crime that the onboard BT is such a challenge. On the bluealsa side, I would like to see Hardware volume control of the alsa device and aptx playback, but so far so good. I have seen a bluealsa coredump on occasion, but think it has to do with the onboard BT and not bluealsa. If I run into anything more, I'll let you know.

borine, 2019-05-02:
On further reading, I found that the raspberry pi has 2 UART serial devices, and /dev/ttyAMA0 is the "most capable" one which *must* be used for bluetooth audio. Raspbian configures this out of the box, for other systems you would need to check.
My problem with the android phone was caused by my /boot/cmdline.txt where I had `console=/dev/ttyAMA0`
I changed this to `console=serial0,115200`, rebooted, and now my pi3b+ works fine as as a2dp sink with either bluealsa or pulseaudio.
Note that on the pi3b+ the hciattach command is:
```/usr/bin/hciattach /dev/serial1 bcm43xx 3000000 flow - xx:xx:xx:xx:xx:xx```
and that /dev/serial1 is a symlink to /dev/ttyAMA0

For the pi0w, the same change has greatly improved the result, but I am still getting occasional underruns (every 10 - 15 seconds or so); the speed fluctuation has gone. The hciattach command on the pi0w is the same as above.

borine, 2021-06-04:
I think RPi firmware improvements have largely resolved these issues now, and as it is clearly not an issue specifically with bluez-alsa I am closing now.

closed 2021-06-04 by borine