arkq/bluez-alsa #213: a2dp sink delay via AVDTP

morrolinux, 2019-05-20:
Hello,
I am building an ffmpeg-based audio sink system to use a Raspberry Pi as a bluetooth speaker. 

Since the audio takes quite a long time to get past the processing on the Pi, I need to tell the bluetooth "client" device how much delay should be added (eg. for video playback) for it to be in sync. 

This can be done via `AVDTP` and I see in your code you kinda mention it in `ba-transport.h : 133` 

I've looked back and forth at the code and found `a2dp` struct, which has a `delay` attribute, but couln't figure out how/when to read it (was always empty in my prints) and how/when to actually set it for the client to be aware. 
Maybe I'm being naive asking, but is this being implemented?

morrolinux, 2019-05-23:
Please let me know if I'm just delirious or that makes any sense at all. This thing is really important to me.

Arkq, 2019-05-24:
No, it's not supported yet.

closed 2019-10-20 by morrolinux