arkq/bluez-alsa #214: Connected: yes, but no sound

antonbutanaev, 2019-05-21:
Hi! I'm struggling to play sound on embedded linux, using bluez5 5.50 and bluez-alsa 1.4.0. I'm able to connect my headset, but cannot play a2dp sound. I rebuild both bluez5 and bluez-alsa with --enable-debug. Here is what I see:

```
# bluetoothctl -- info C0:7A:A5:01:48:F0
Device C0:7A:A5:01:48:F0 (public)
	Name: SVEN SEB-B270MV 
	Alias: SVEN SEB-B270MV 
	Class: 0x00240404
	Icon: audio-card
	Paired: yes
	Trusted: no
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)

```
/etc/asound.conf
```
pcm.mono {
    type route
	slave {
	    pcm default
	    channels 2
	}
    ttable.0.0 1
    ttable.0.1 1
}

pcm.btheadset {
        type plug
        slave.pcm {
                type bluealsa
		interface "hci0"
                device "C0:7A:A5:01:48:F0"
                profile "a2dp"
        }
        hint {
                show on
                description "Bluetooth Headset"
        }
}

```

and then, when I try to play:

`aplay -D btheadset /my/maybe-next-time.wav`

I'm almost there, got status: SINK_STATE_PLAYING, Created new IO thread: A2DP Source (SBC), Starting IO loop: A2DP Source (SBC), but then happens abruptly src/adapter.c:adapter_service_remove() /org/bluez/hci0 and shutdown begins. No sound. Full log below:

```
# aplay -D btheadset /my/maybe-next-time.wav 
../shared/ctl-client.c:108: Connecting to socket: /var/run/bluealsa/hci0
bluealsa: ctl.c:596: Received new connection: 8
bluealsa: ctl.c:609: New client accepted: 8
bluealsa: ctl.c:634: +-+-
../shared/ctl-client.c:238: Getting transport for C0:7A:A5:01:48:F0 type 0x41
bluealsa: ctl.c:634: +-+-
bluealsa-pcm.c:583: Setting constraints: 3
Playing WAVE '/my/maybe-next-time.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
bluealsa-pcm.c:307: Initializing HW: 3
../shared/ctl-client.c:397: Requesting PCM open for C0:7A:A5:01:48:F0
bluealsa: ctl.c:329: PCM requested for C0:7A:A5:01:48:F0 type 0x41
bluetoothd[708]: profiles/audio/transport.c:media_owner_create() Owner created: sender=:1.1
bluetoothd[708]: profiles/audio/avdtp.c:avdtp_ref() 0x1bc6f80: ref=2
[... 75 lines cut ...]
bluetoothd[708]: profiles/audio/avrcp.c:controller_destroy() 0x1bc3300
bluetoothd[708]: profiles/audio/avrcp.c:target_destroy() 0x1bcab58
bluetoothd[708]: profiles/audio/avctp.c:avctp_set_state() AVCTP Disconnected
bluetoothd[708]: profiles/audio/avdtp.c:avdtp_unref() 0x1bc6f80: ref=0
bluetoothd[708]: profiles/audio/avdtp.c:avdtp_unregister_sep() SEP 0x1bc2cd8 unregistered: type:0 codec:0 seid:1
bluetoothd[708]: profiles/audio/avdtp.c:avdtp_ref() 0x1bc6f80: ref=1
bluetoothd[708]: profiles/audio/avdtp.c:connection_lost() Disconnected from C0:7A:A5:01:48:F0
bluetoothd[708]: profiles/audio/sink.c:sink_set_state() State changed /org/bluez/hci0/dev_C0_7A_A5_01_48_F0: SINK_STATE_PLAYING -> SINK_STATE_DISCONNECTED
bluetoothd[708]: profiles/audio/a2dp.c:channel_remove() chan 0x1bc26e8
bluetoothd[708]: profiles/audio/avdtp.c:avdtp_unref() 0x1bc6f80: ref=0
bluetoothd[708]: profiles/audio/avdtp.c:avdtp_free() 0x1bc6f80
bluetoothd[708]: src/adapter.c:dev_class_changed_callback() Class: 0x080000
bluetoothd[708]: src/adapter.c:dev_class_changed_callback() Class: 0x000000
bluetoothd[708]: src/adapter.c:dev_disconnected() Device C0:7A:A5:01:48:F0 disconnected, reason 2
bluetoothd[708]: src/adapter.c:adapter_remove_connection() 
bluetoothd[708]: plugins/policy.c:disconnect_cb() reason 2
bluetoothd[708]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr C0:7A:A5:01:48:F0 type 0 status 0xe
bluetoothd[708]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
bluetoothd[708]: src/device.c:device_bonding_failed() status 14
bluetoothd[708]: src/adapter.c:resume_discovery() 
bluetoothd[708]: src/adapter.c:remove_temp_devices() /org/bluez/hci0
```
Could you please point where to see to fix this?

Thanks!

sherifomran, 2019-09-23:
I had the same issue, i could solve it by giving the device mac to the aplay, i think aplay has a -d syntax followed by the device address. I don't no the cause of this issue and for sure it would be good to solve.

antonbutanaev, 2019-09-24:
The problem was sbc codec crashing under ARM. https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=862346

closed 2019-09-24 by antonbutanaev