arkq/bluez-alsa #215: Unhanded AAC decoder error in bluealsa causes system crash

snehaltan, 2019-05-23:
I am running bluealsa in a2dp-sink mode streaming from an iphone using aac codec. My system crashed after 4 hours of continous playback. Backtrace showing bluealsa memory consumption of 46MB. See the kernel dump below

```
[12803.156921] [ pid ]   uid  tgid total_vm      rss nr_ptes nr_pmds swapents oom_score_adj name
[12803.170369] [  191]     0   191     2515      141       4       0        0         -1000 udevd
[12803.202339] [  741]   105   741      730       55       4       0        0             0 dbus-daemon
[12803.235165] [  801]     0   801     1174      113       4       0        0             0 bluetoothd
[12803.555683] [ 1026]     0  1026      448       18       3       0        0             0 hciattach
[12803.568641] [ 1030]     0  1030    56543    46015     102       0        0             0 bluealsa
[12803.577716] [ 1044]     0  1044     8506       93       9       0        0             0 bluealsa-aplay

```
So I decided to turn the bluealsa debug on to dig further. Below is part of the log and it continues until there is no more memory to allocate and causing the entire system to crash. I am looking at blualsa source code and it seems `io_thread_a2dp_sink_aac` reacts to an invalid bt frame and gets stuck in this memory allocation loop. I don't know much about fdk-aac myself, so I thought it would be a good idea to submit an issue here while I am trying to understand the exact failure mechanism.

```
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 1024 -> 2048
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 2048 -> 3072
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 3072 -> 4096
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 4096 -> 5120
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 5120 -> 6144
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 6144 -> 7168
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 7168 -> 8192
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 8192 -> 9216
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 9216 -> 10240
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 10240 -> 11264
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 11264 -> 12288
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 12288 -> 13312
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 13312 -> 14336
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 14336 -> 15360
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 15360 -> 16384
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 16384 -> 17408
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 17408 -> 18432
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 18432 -> 19456
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 19456 -> 20480
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 20480 -> 21504
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 21504 -> 22528
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 22528 -> 23552
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 23552 -> 24576
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 24576 -> 25600
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 25600 -> 26624
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 26624 -> 27648
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 27648 -> 28672
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 28672 -> 29696
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 29696 -> 30720
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 30720 -> 31744
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 31744 -> 32768
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: AAC decode frame error: Unsupported gain control data
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 32768 -> 33792
bluealsa: ../../git/src/utils.c:644: Unknown error code: 0x5
bluealsa: AAC buffer fill error: Unknown error
bluealsa: ../../git/src/utils.c:644: Unknown error code: 0x5
bluealsa: AAC buffer fill error: Unknown error
bluealsa: ../../git/src/io.c:739: Resizing LATM buffer: 33792 -> 34816
bluealsa: ../../git/src/utils.c:644: Unknown error code: 0x5
bluealsa: AAC buffer fill error: Unknown error
```

Arkq, 2019-05-23:
Which version of fdk-aac you are using? And which iOS version?

Does this resize log appears from time to time, or it burst out suddenly which results in OOM?

snehaltan, 2019-05-23:
Hi Arkq,
Sorry I missed the important details. I am running bluezalsa tag 1.4.0. iOs 12.1. It happens once every 3-4 hours. Its results in OOM in about 4-5mins after it gets into resizing loop. If I happen to pause the playback in that time bluez alsa will kill the aac decoding thread and memory is released. I can then resume the playback again and everything works normal for another 3-4 hours. If I dont pause then OOM will occur and system needs to be restarted.

Arkq, 2019-05-23:
> works normal for another 3-4 hours

Hmmm... it looks like some counter overflow. I will look into it. To be honest, I've never done aging tests with bluealsa :) But this will have to wait until (I think) next week. Right now I'm in the middle of a very big change in bluealsa. In the meantime you can check whether the timing is always e.g. 3.5 hours. This will definitely mean overflow, and the question will be whether this overflow is in the fdk-aac or bluealsa :)

snehaltan, 2019-05-23:
Thanks for the hints. I will do more testing and debugging over next few days. Hopefully I will be able to create a pull request early next week.

snehaltan, 2019-05-24:
I am now testing the change above by letting run bluealsa over the weekend. I will update the result next week.

snehaltan, 2019-06-20:
Fix seems to work fine for me. But it would be worthwhile if you @Arkq  can have a look at the fix as I don't have full understanding of LATM buffer management.

commit 4d73f961 (2019-07-26): Rewind LATM buffer on aacDecode error

If LATM buffer is not rewinded on unsuccessful decoding, same BT frame
data will be retried for decoding. If the frame is unable to decode LATM
buffer will be allocated more memory to store new BT frames and will
eventually cause OOM. Since retry on same frame is not going to cause
AAC decode to success, it is best to just drop the frame and carry on.

Fixes #215
closed 2019-07-26 by Arkq