arkq/bluez-alsa #220: Unsupported AT message: CMD: command:+BLDN

xuewenxu, 2019-06-05:
I have tried to run 'arecord capture.wav'  to record  voice that from headset,  the logs like these:
bluealsa: ../../src/ctl.c:594: Received new connection: 17
bluealsa: ../../src/ctl.c:607: New client accepted: 17
bluealsa: ../../src/ctl.c:632: +-+-
bluealsa: ../../src/ctl.c:632: +-+-
bluealsa: ../../src/ctl.c:327: PCM requested for 1C:52:16:17:A8:F4 type 0x82
bluealsa: ../../src/ctl.c:632: +-+-
bluealsa: ../../src/ba-transport.c:565: State transition: 2 -> 2
bluealsa: ../../src/ctl.c:632: +-+-
bluealsa: ../../src/ba-transport.c:809: New SCO link: 18 (MTU: R:48 W:48)

and then click headset to start speaking,  the logs like these:
bluealsa: ../../src/at.c:161: AT message: CMD: command:+BLDN, value:(null)
bluealsa: Unsupported AT message: CMD: command:+BLDN, value:(null)
bluealsa: ../../src/rfcomm.c:102: Sending AT message: RESP: command:(null), value:ERROR

No voices were capture in capture.wav.

Arkq, 2019-06-05:
Internal HFP support might not work correctly. If possible, please use '-p
hfp-ofono'.

xuewenxu, 2019-06-05:
Hi, 
    @Arkq  Run bluealsa use '-p hfp-ofono' and then use bluetoothctl to connect headset, it shows 'Failed to connect: org.bluez.Error.Failed'

Arkq, 2019-11-05:
+BLDN is for redialing last number, and is not ment to be supported by bluealsa. Use `bluealsa-rfcomm` if you want to catch it. As for SCO recording please use current master with hsp-ag or hfp-ag. If it does not work, it is most likely to be a problem with your BT chip setup (e.g. known broadcom chip issue).

closed 2019-11-05 by Arkq