arkq/bluez-alsa #222: Cannot play to bluetooth speaker With source build install on Raspberry Pi 3B

out0fmemory, 2019-06-09:
Hello,

  I build bule-alsa from source(master branch),but on rpi 3b, it cannot play to bluetooth speaker.

my asound config is blew
```
pcm.!default {
        type plug
        slave.pcm {
                type bluealsa
                device "30:21:54:94:C6:33"
                profile "a2dp"
        }
}

pcm.!jack {
	type hw
	card 0
}

ctl.!default {
	type hw
	card 0
}
```

when i play a mp3 file use mpg123, play seems ok, but the blue speaker is no any audio out ,the alsa log is blew, no error occur
```
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes
../../../src/asound/bluealsa-pcm.c:589: /org/bluealsa/hci0/dev_30_21_54_94_C6_33/a2dp: Setting constraints
../../../src/asound/bluealsa-pcm.c:309: /org/bluealsa/hci0/dev_30_21_54_94_C6_33/a2dp: Closing
../../../src/asound/bluealsa-pcm.c:589: /org/bluealsa/hci0/dev_30_21_54_94_C6_33/a2dp: Setting constraints
../../../src/asound/bluealsa-pcm.c:309: /org/bluealsa/hci0/dev_30_21_54_94_C6_33/a2dp: Closing

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: 4021963472.mp3 ...
../../../src/asound/bluealsa-pcm.c:589: /org/bluealsa/hci0/dev_30_21_54_94_C6_33/a2dp: Setting constraints
../../../src/asound/bluealsa-pcm.c:319: /org/bluealsa/hci0/dev_30_21_54_94_C6_33/a2dp: Initializing HW
../../../src/asound/bluealsa-pcm.c:346: /org/bluealsa/hci0/dev_30_21_54_94_C6_33/a2dp: FIFO buffer size: 4096
../../../src/asound/bluealsa-pcm.c:352: /org/bluealsa/hci0/dev_30_21_54_94_C6_33/a2dp: Selected HW buffer: 3 periods x 12800 bytes <= 38404 bytes
../../../src/asound/bluealsa-pcm.c:367: /org/bluealsa/hci0/dev_30_21_54_94_C6_33/a2dp: Initializing SW
../../../src/asound/bluealsa-pcm.c:367: /org/bluealsa/hci0/dev_30_21_54_94_C6_33/a2dp: Initializing SW
../../../src/asound/bluealsa-pcm.c:367: /org/bluealsa/hci0/dev_30_21_54_94_C6_33/a2dp: Initializing SW
../../../src/asound/bluealsa-pcm.c:389: /org/bluealsa/hci0/dev_30_21_54_94_C6_33/a2dp: Prepared
../../../src/asound/bluealsa-pcm.c:367: /org/bluealsa/hci0/dev_30_21_54_94_C6_33/a2dp: Initializing SW
../../../src/asound/bluealsa-pcm.c:280: /org/bluealsa/hci0/dev_30_21_54_94_C6_33/a2dp: Stopping
../../../src/asound/bluealsa-pcm.c:359: /org/bluealsa/hci0/dev_30_21_54_94_C6_33/a2dp: Freeing HW
../../../src/asound/bluealsa-pcm.c:309: /org/bluealsa/hci0/dev_30_21_54_94_C6_33/a2dp: Closing

MPEG 1.0 L III cbr128 44100 j-s

Title:                                                                  Artist:
Comment:                                                                Album:
Year:                                                                   Genre:  Unknown
../../../src/asound/bluealsa-pcm.c:589: /org/bluealsa/hci0/dev_30_21_54_94_C6_33/a2dp: Setting constraints
../../../src/asound/bluealsa-pcm.c:319: /org/bluealsa/hci0/dev_30_21_54_94_C6_33/a2dp: Initializing HW
../../../src/asound/bluealsa-pcm.c:346: /org/bluealsa/hci0/dev_30_21_54_94_C6_33/a2dp: FIFO buffer size: 4096
../../../src/asound/bluealsa-pcm.c:352: /org/bluealsa/hci0/dev_30_21_54_94_C6_33/a2dp: Selected HW buffer: 3 periods x 12800 bytes <= 38404 bytes
../../../src/asound/bluealsa-pcm.c:367: /org/bluealsa/hci0/dev_30_21_54_94_C6_33/a2dp: Initializing SW
../../../src/asound/bluealsa-pcm.c:367: /org/bluealsa/hci0/dev_30_21_54_94_C6_33/a2dp: Initializing SW
../../../src/asound/bluealsa-pcm.c:367: /org/bluealsa/hci0/dev_30_21_54_94_C6_33/a2dp: Initializing SW
../../../src/asound/bluealsa-pcm.c:389: /org/bluealsa/hci0/dev_30_21_54_94_C6_33/a2dp: Prepared
../../../src/asound/bluealsa-pcm.c:367: /org/bluealsa/hci0/dev_30_21_54_94_C6_33/a2dp: Initializing SW
../../../src/asound/bluealsa-pcm.c:240: /org/bluealsa/hci0/dev_30_21_54_94_C6_33/a2dp: Starting
../../../src/asound/bluealsa-pcm.c:134: /org/bluealsa/hci0/dev_30_21_54_94_C6_33/a2dp: Starting IO loop: 8
 ```

out0fmemory, 2019-06-09:
syslog is blew
```
un 10 00:49:42 raspberrypi bluealsa[21857]: /usr/bin/bluealsa: ../../src/bluealsa-dbus.c:133: Called: org.bluealsa.Manager1.GetPCMs()
Jun 10 00:49:42 raspberrypi bluealsa[21857]: /usr/bin/bluealsa: ../../src/bluealsa-dbus.c:133: Called: org.bluealsa.Manager1.GetPCMs()
Jun 10 00:49:42 raspberrypi bluealsa[21857]: /usr/bin/bluealsa: ../../src/bluealsa-dbus.c:133: Called: org.bluealsa.Manager1.GetPCMs()
Jun 10 00:49:42 raspberrypi bluealsa[21857]: /usr/bin/bluealsa: ../../src/bluealsa-dbus.c:316: Called: org.bluealsa.PCM1.Open()
Jun 10 00:49:43 raspberrypi bluealsa[21857]: /usr/bin/bluealsa: ../../src/ba-transport.c:697: New transport: 24 (MTU: R:672 W:672)
Jun 10 00:49:43 raspberrypi bluealsa[21857]: /usr/bin/bluealsa: ../../src/bluez.c:1263: Signal: org.freedesktop.DBus.Properties.PropertiesChanged()
Jun 10 00:49:43 raspberrypi bluealsa[21857]: /usr/bin/bluealsa: ../../src/bluez.c:1297: Signal: PropertiesChanged: org.bluez.MediaTransport1: State
Jun 10 00:49:43 raspberrypi bluealsa[21857]: /usr/bin/bluealsa: ../../src/ba-transport.c:576: State transition: 0 -> 2
Jun 10 00:49:43 raspberrypi bluealsa[21857]: /usr/bin/bluealsa: ../../src/io.c:2068: Created new IO thread: A2DP Source (SBC)
Jun 10 00:49:43 raspberrypi bluealsa[21857]: /usr/bin/bluealsa: ../../src/io.c:435: Starting IO loop: A2DP Source (SBC)
Jun 10 00:49:43 raspberrypi bluealsa[21857]: /usr/bin/bluealsa: ../../src/io.c:108: PCM read buffer flushed: 0
Jun 10 00:49:43 raspberrypi bluealsa[21857]: /usr/bin/bluealsa: ../../src/io.c:93: PCM has been closed: 20
Jun 10 00:49:43 raspberrypi bluealsa[21857]: /usr/bin/bluealsa: ../../src/ba-transport.c:857: Closing PCM: 20
Jun 10 00:49:43 raspberrypi bluealsa[21857]: /usr/bin/bluealsa: ../../src/ba-transport.c:857: Closing PCM: 20
Jun 10 00:49:43 raspberrypi bluealsa[21857]: /usr/bin/bluealsa: ../../src/io.c:489: Keep-alive polling: 0
Jun 10 00:49:43 raspberrypi bluealsa[21857]: /usr/bin/bluealsa: ../../src/io.c:489: Keep-alive polling: 0
Jun 10 00:49:43 raspberrypi bluealsa[21857]: /usr/bin/bluealsa: ../../src/ba-transport.c:724: Releasing transport: A2DP Source (SBC)
Jun 10 00:49:43 raspberrypi bluealsa[21857]: /usr/bin/bluealsa: ../../src/bluealsa-dbus.c:133: Called: org.bluealsa.Manager1.GetPCMs()
Jun 10 00:49:43 raspberrypi bluealsa[21857]: /usr/bin/bluealsa: ../../src/bluealsa-dbus.c:316: Called: org.bluealsa.PCM1.Open()
Jun 10 00:49:43 raspberrypi bluealsa[21857]: /usr/bin/bluealsa: ../../src/ba-transport.c:665: Reusing transport: 24
Jun 10 00:49:43 raspberrypi bluealsa[21857]: /usr/bin/bluealsa: ../../src/ba-transport.c:576: State transition: 2 -> 2
Jun 10 00:49:43 raspberrypi bluealsa[21857]: /usr/bin/bluealsa: ../../src/ba-transport.c:753: Closing BT: 24
Jun 10 00:49:43 raspberrypi bluealsa[21857]: /usr/bin/bluealsa: ../../src/ba-transport.c:903: Exiting IO thread: A2DP Source (SBC)
Jun 10 00:49:43 raspberrypi bluealsa[21857]: /usr/bin/bluealsa: ../../src/bluez.c:1263: Signal: org.freedesktop.DBus.Properties.PropertiesChanged()
Jun 10 00:49:43 raspberrypi bluealsa[21857]: /usr/bin/bluealsa: ../../src/bluez.c:1297: Signal: PropertiesChanged: org.bluez.MediaTransport1: State
Jun 10 00:49:43 raspberrypi bluealsa[21857]: /usr/bin/bluealsa: ../../src/ba-transport.c:576: State transition: 2 -> 0
```

borine, 2019-06-19:
There is not nearly enough information in your post for anyone to help.
Which OS distribution are you using?
If that distribution provides a bluealsa package, have you installed and tested that first?
Which commit was the master branch at when you built it?
What parameters did you pass to configure when building?
What parameters did you pass to bluealsa when starting it?
Does mpg123 terminate cleanly when finished playing, or does it hang or crash?

I've tried mpg123 on Raspbian Stretch Lite 2019-04-08 with bluez-alsa commit 31085f9 using a RPi3B+, on-board bluetooth adapter, wifi enabled, and all works well.

Arkq, 2019-10-06:
@out0fmemory please try the newest master. There are two top commits which are related to mpg123 playback. Maybe they will resolve your issue.

borine, 2021-06-04:
mpg123 is working well with bluealsa for over a year now, so hopefully the issue is now resolved. I am closing this issue as there has been no activity for some. Please feel free to re-open it if the problem persists with latest bluealsa code.

closed 2021-06-04 by borine