arkq/bluez-alsa #223: Using Bluetooth Microphone and 3.5 Speaker

hankp46, 2019-06-10:
I'm trying to setup my audio to receive input from a bluetooth speaker and play output throught the raspberry pi 3.5 jack. I setup my .asoundrc as such:
```
pcm.!default {
type asym
capture.pcm "mic"
playback.pcm "speaker"
}
pcm.speaker {
type plug slave { pcm "hw:0,0" }
}
pcm.mic {
type plug
slave.pcm {
type bluealsa device "FC:58:FA:0C:74:0C"
profile "sco"
}
}

ctl.!default {
type bluealsa
}
```
I can play wav files through the speaker but cannot get arecord to record anything:
```
arecord  -f S16_LE   capture.wav
Recording WAVE 'capture.wav' : Signed 16 bit Little Endian, Rate 8000 Hz, Mono
```
capure.wav is empty

I have tested the bluetooth microphone on other devices and it is operating correctly.

borine, 2019-06-17:
What happens if you ignore the .asoundrc settings and expicitly point arecord at the bluealsa device?
```
arecord -f S16_LE -D bluealsa:SRV=org.bluealsa,DEV=FC:58:FA:0C:74:0C,PROFILE=sco capture.wav
```

Abu-Abdullah, 2019-09-19:
the same output, file is empty and the size is fixed of 44 bytes

borine, 2019-09-20:
@Abu-Abdullah what command line are you using to start bluealsa on the pi? Try:
`sudo bluealsa -p hfp-ag -p a2dp-source`
Are you able to capture sco audio now?

Abu-Abdullah, 2019-09-20:
same result
bluealsa version on Pi is 1.3.1

im using this command
```
arecord -f S16_LE --dump-hw-params -D bluealsa:HCI=hci0,DEV=11:75:58:0D:84:4A,PROFILE=sco mic.wav
```

borine, 2019-09-20:
I think that bluealsa version is quite old. Maybe you will need to upgrade before this can work. The commands I've described here work for me, and @hankp46 says in #225 that it is working there too, but with very poor sound quality.
There have been a number of issues with bluetooth on the raspberry pi, so it's possible that you are hitting some problem with bluetooth config in general, not bluealsa specific. Check on the forums for your distribution to see if anyone there can help.

qq297110281, 2020-03-04:
i got  the same problem here， did you solve it now? @Abu-Abdullah

borine, 2021-06-22:
This issue has been inactive for some time now. Given improvements to bluealsa as well as the firmware and drivers of the RPi bluetooth stack, it may even be fixed now. Therefore I am closing this issue.

If you experience problems capturing SCO streams on a Raspberry Pi with the latest versions of all software and firmware, please open a new issue with debug logs from the latest bluez-alsa code.

closed 2021-06-22 by borine