arkq/bluez-alsa #224: Compile error - macros dependency

amilino, 2019-06-14:
Hi, I am trying to compile code on my Ubuntu machine but get an error
`checking pkg-config m4 macros... no`

I checked all 3 packages and "macros" package is missing. I have on idea how to install macros package on Ubuntu. Any help would be appreciated.

TheOldPresbyope, 2019-06-17:
Google is your friend. The system is looking for the file containing m4 macros for pkg-config. On my Linux Mint system (derived from Ubuntu) it's /usr/share/aclocal/pkg.m4 and was installed as part of the pkg-config package.

amilino, 2019-06-17:
I will repeat my self one more time. pkg-config and m4 packages are installed. also what you are referring to pkg.m4 file is also on correct location. Not sure why this check is checking for macros package???

Arkq, 2019-10-07:
It seems that it is required to run `autoreconf -i` after `apt install pkg-config`. I've just created simple installation guide and during the process of step by step dependencies installation I've stumpled upon the same issue. See this: https://github.com/Arkq/bluez-alsa/wiki/Installation-from-source

I'm closing this issue, however, it will still occur after `autoreconf -i` please reopen.

closed 2019-10-07 by Arkq