arkq/bluez-alsa #225: Connecting headset - Playback Working but Not Recording Raspian Stretch

hankp46, 2019-06-14:
Running a fresh copy of Raspian Stretch. Connected my Bluetooth Headset via Dashboard controls. Can playback wav files.But when trying to record with headset microphone get following errors:
```
pi@raspberrypi:~ $ arecord  onvers.wav
ALSA lib bluealsa-pcm.c:680:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA transport: No such device
arecord: main:788: audio open error: No such device
```
Then tried following:
```
pi@raspberrypi:~ $ arecord  -D bluealsa onvers.wav
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
arecord: main:788: audio open error: No such file or directory
```
My asoundrc file is as follows;
```
pcm.!default {
	type plug
	slave.pcm {
		type bluealsa
		device "FC:58:FA:0C:74:0C"
		interface "hci0"
		profile "a2dp"
	}
}

ctl.!default {
	type bluealsa
}
```Amixer shows:
```
pi@raspberrypi:~ $ amixer -D bluealsa
Simple mixer control 'ONN In-Ear - A2DP',0
  Capabilities: pvolume pswitch
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 127
  Mono:
  Front Left: Playback 127 [100%] [on]
  Front Right: Playback 127 [100%] [on]
Simple mixer control 'ONN In-Ear - SCO',0
  Capabilities: pvolume pvolume-joined cvolume cvolume-joined pswitch pswitch-joined cswitch cswitch-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: Playback 0 - 15 Capture 0 - 15
  Mono: Playback 15 [100%] [on] Capture 15 [100%] [on]
Simple mixer control 'ONN In-Ear | Battery',0
  Capabilities: pvolume pvolume-joined
  Playback channels: Mono
  Limits: Playback 0 - 100
  Mono: Playback 100 [100%]
```
has anybody gotten a headset to work on a ERaspberry Pi?

borine, 2019-06-17:
The text  of your asoundrc and amixer output has been corrupted by the markdown. But, assuming it is all there, then you seem to be missing a bluealsa SCO default pcm capture device definition - there is only a A2DP playback device shown. Maybe that is why arecord reports "no such device" ?

I don't have a SCO device, but I can verify that A2DP capture works well for me on a PI3B+ (and reasonably well with a PIZeroW) with Raspbian Stretch Lite and bluez-alsa master (commit 31085f9)

hankp46, 2019-06-17:
I attempted the command in your email:
```
arecord -f S16_LE -D bluealsa:SRV=org.bluealsa,DEV=FC:58:FA:0C:74:0C,PROFILE=sco capture.wav
ALSA lib conf.c:4858:(parse_args) Unknown parameter SRV
ALSA lib conf.c:4991:(snd_config_expand) Parse arguments error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa:SRV=org.bluealsa,DEV=FC:58:FA:0C:74:0C,PROFILE=sco
arecord: main:788: audio open error: No such file or directory
```
You also mention "A2DP capture works well for me on a PI3B+" Did you capture from a microphone or did you mean playback? If you did capture what does your .asoundrc look like?

borine, 2019-06-17:
For the benefit of future readers of this post, see #223 for context to "the command in your email".

 "Unknown parameter SRV" - this means you are not using the latest bluealsa sources. Older releases used a different method to communicate between ALSA and the bluealsa daemon, and had a parameter called "HCI" instead. So try
```
arecord -f S16_LE -D bluealsa:HCI=hci0,DEV=FC:58:FA:0C:74:0C,PROFILE=sco capture.wav
```
I use A2DP capture to receive music from a mobile phone and play it through an IQAudio DAC to my stereo system. I think that headset microphones use SCO, not A2DP, although I am no expert on bluetooth profiles. As I said, I have no SCO device. It is clear from your amixer output that your headset supports A2DP profile for playback only  ("ONN In-Ear - A2DP") and SCO profile ("ONN In-Ear - SCO") for both playback and capture.

I do not use .asoundrc, instead I always tell my applications explicitly which device to use.

Anyway, hopefully debug output from the above arecord command will help you to find the source of this issue.

borine, 2019-06-18:
A thought just occurred to me - bluealsa by default does not enable the input profiles, you have to explicitly specify each one you want on the command line when bluealsa is launched. Type
```
bluealsa --help
```
for a list of the available profiles. I'm guessing you need hfp-hf and/or hsp-hf.

If you're running the default Raspbian bluealsa package, then it is probably started from a systemd service file. To test my theory, do
```
sudo systemctl stop bluealsa
# the bluetooth headset is now disconnected
sudo bluealsa -p a2dp-source -p a2dp-sink -p hfp-hf -p hfp-ag -p hsp-hs -p hsp-ag &
# you will need to reconnect the bluetooth headset here
arecord -f S16_LE -D bluealsa:HCI=hci0,DEV=FC:58:FA:0C:74:0C,PROFILE=sco capture.wav
```
Hopefully now arecord will at least be able to open the device.

If this works, you can remove each of the parameters in turn and retry to find the minimum set required for your headset.

hankp46, 2019-06-20:
I finally got my bluetooth headset working, both recording and playback. Thank You Borine!!!
The project I'm working on requires that I use the bluetooth mic but play through the 3.5 speaker jack. I also got this working using the following config:
```
pcm.!default {
type asym
capture.pcm "mic"
playback.pcm "speaker"
}

pcm.speaker {
		type plug 
		slave.pcm "hw:0,0"
}

pcm.mic {
type plug
slave.pcm {
		type bluealsa
		device "FC:58:FA:0C:74:0C"
		profile "sco"
                interface "hci0"
}
}

ctl.!default {
type bluealsa
}
```
The only problem is that the quality of the recording is lacking. I attempted to add "rate 44100" and 
"format  S16_LE " to the mic but got following errors:
```
ALSA lib bluealsa-pcm.c:641:(_snd_pcm_bluealsa_open) Unknown field rate
arecord: main:788: audio open error: Invalid argument

ALSA lib bluealsa-pcm.c:641:(_snd_pcm_bluealsa_open) Unknown field format
arecord: main:788: audio open error: Invalid argument
```
Are there equivalent fields for bluealsa or are you stuck with what the defaults?

borine, 2019-06-21:
AFAIK the bluetooth audio profiles do not allow the application to control the ADC parameters of the device. So your headset will use whatever rate/format/channel settings have been set at the factory. The SCO audio transport places quite severe restrictions on this (it was designed for voice applications, not music).
You can see the values supported by the device if you add `--dump-hw-params` to the arecord command:
```
arecord -f S16_LE --dump-hw-params -D bluealsa:HCI=hci0,DEV=FC:58:FA:0C:74:0C,PROFILE=sco capture.wav
```
SCO is always format S16_LE, mono, but may have a higher sample rate than the 8000Hz that arecord uses by default. So perhaps you need to pass a `-r` parameter to make sure that arecord is not accidentally resampling the PCM stream. The best quality will be achieved if you tell arecord the actual rate deiivered by the device (which may be 8000Hz, or it may be higher, but will never be as high as 44100kHz when using the hsp or hfp bluetooth profiles.

If you really want to just direct the microphone input to a loudspeaker connected to the Pi 3.5mm jack, then I recommend you use the bluealsa-aplay application instead. That application will handle all the details of detecting the input PCM parameters and passing the stream to the desired alsa playback device.

Using your .asoundrc configuration above for the 3.5mm jack, I would expect something like:
```
bluealsa-aplay --hci=hci0 --profile-sco FC:58:FA:0C:74:0C
```
to "just work".

borine, 2019-08-19:
I just came across a post from last year on the raspberry pi linux repository suggesting that the broadcom bluetooth chip onboard the rpi uses a "non-standard" packet size for SCO traffic, and this was causing very poor sound quality on HSP/HFP profile through pulseaudio on the pi. This was fixed in an update to the pulseaudio code. The specific comment of interest in that post is:
https://github.com/raspberrypi/linux/issues/2229#issuecomment-391561840
@Arkq : I note that in  src/ba-transport.c:833 bluealsa forces the SCO mtu size to be 48 bytes; according to the above post (and the pulseaudio patch in the linked comment) the raspberry pi onboard BT requires an mtu of 64 bytes. I note that pulse makes the same change in the ofono transport as well as its native SCO backend.
I wonder if that is the reason for poor HSP / HFP experienced by @hankp46 on the raspberry pi? Unfortunately I don't have a headset with which to test this.
@hankp46 - if you are still playing with this, could you try building bluealsa with t->mtu_read and t->mtu_write set to 64 instead of 48 in src/ba-transport.c lines 833 and 834?

borine, 2021-06-06:
This issue has been inactive for some time now. Given improvements to bluealsa as well as the firmware and drivers of the RPi bluetooth stack, some of the comments here are now outdated and no longer relevant. Therefore I am closing this issue.

If you experience problems capturing SCO streams on a Raspberry Pi with the latest versions of all software and firmware, please open a new issue with debug logs from the latest bluez-alsa code.

closed 2021-06-06 by borine