arkq/bluez-alsa #228: 10-20% dropped frames on Youtube and Chrome

jaccc, 2019-06-27:
there are 10%-20% dropped frames according to "stats for nerds" on Youtube, Chrome and Gentoo testing

pcm.!default {
type plug
slave.pcm bluealsa
}

defaults.bluealsa {
    interface "hci0"
    device "xxxxx"         
    profile "a2dp"
    rate 48000
    delay 10000
}

jaccc, 2019-06-28:
Number of dropped frames is close to 0 with bluetooth/pulse and usb/alsa. It is possible that different codec (aptX) is used with bluetooth/pulse.

borine, 2020-08-24:
Today I opened a second tab in chromium, and ... `Audio renderer error. Please restart your computer.` D'oh !!
I forgot chromium runs each tab in a separate process.

So the dropped frames issue reported here is fixed, but there's still a mountain to climb before bluealsa plays nicely with chromium-based browsers :disappointed:

borine, 2021-10-04:
The issue of dropped frames in Chrome was fixed a year ago (bluez--alsa release 3.0.0, September 2020), Another issue that affected chromium/chrome among others was the way the BlueALSA plugin reported device disconnection to applications. This is fixed in the latest BlueALSA code, and will be in the next release. When used with a single tab in chromium, BlueALSA now gives excellent results. Use of multiple tabs requires chromium/chrome to open the same PCM multiple times, and that is not supported by BlueALSA PCMs **by design**, it is not a bug (although clearly it would be a desirable enhancement). Given that the actual fault reported here is fixed, I am closing this issue.

closed 2021-10-04 by borine