arkq/bluez-alsa #233: latest git version D-bus error

Pro-pra, 2019-07-16:
dbus sessions launched, bluetoothd work fine.
```
bluealsa: bluez.c:801: Registering endpoint: /org/bluez/hci0/A2DP/APTX/Source/1
bluealsa: bluez.c:801: Registering endpoint: /org/bluez/hci0/A2DP/MPEG24/Source/1
bluealsa: bluez.c:801: Registering endpoint: /org/bluez/hci0/A2DP/SBC/Source/1
bluealsa: ba-adapter.c:134: Freeing adapter: hci0
bluealsa: bluez.c:1065: Registering profile: /org/bluez/HSP/AudioGateway
bluealsa: bluez.c:1065: Registering profile: /org/bluez/HFP/AudioGateway
bluealsa: main.c:328: Acquiring D-Bus service name: org.bluealsa
bluealsa: main.c:333: Starting main dispatching loop
bluealsa: Couldn't acquire D-Bus name: org.bluealsa
bluealsa: main.c:337: Exiting main loop
```

darjusp, 2019-07-18:
got this one too  on Arch when launching 'bluealsa':
bluealsa: Couldn't acquire D-Bus name: org.bluealsa

Acquired package from AUR https://aur.archlinux.org/packages/bluez-alsa-git/

is there a way how it could be avoided or fixed?

Arkq, 2019-07-18:
You can run dbus-monitor and see whether some "obvious" reason of not
acquiring dbus name is not the issue here. Dbus provides some extra message
unfortunately I don't know how to access it from the gdbus API.

darjusp, 2019-07-18:
sorry my mistake, it seems I launched it twice. when "killall bluealsa" and then "bluealsa &" comes without any errors. So does it mean that bluealsa works properly, and most probably doesn't cause failure to connect to bluetooth devices, which I am trying to solve now?

Arkq, 2019-07-18:
> So does it mean that bluealsa works properly

It's really hard to tell, because there can ba a lot of obstacles preventing connection with the BT device. But the best way to tell is to compile bluez-alsa with debug enabled and see the logs. If there is no logs (when the connection attempt is made) it might mean, that the connection is not forwarded to the bluealsa by the bluez daemon. Then, you have to look for the issue in the bluez itself. Also, make sure that profile which you try to utilize is enabled in the bluealsa (see `bluealsa --help`).

Pro-pra, 2019-07-19:
added dbus-monitor log, see start message.

This is right dbus-monitor log for old worked version bluealsa:

```
signal sender=org.freedesktop.DBus -> dest=(null destination) serial=235 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameOwnerChanged
   string ":1.119"
   string ""
   string ":1.119"
signal sender=:1.4 -> dest=(null destination) serial=544 path=/org/bluez/hci0; interface=org.freedesktop.DBus.Properties; member=PropertiesChanged
   string "org.bluez.Adapter1"
   array [
      dict entry(
         string "UUIDs"
         variant             array [
               string "0000110a-0000-1000-8000-00805f9b34fb"
               string "00001801-0000-1000-8000-00805f9b34fb"
               string "0000110e-0000-1000-8000-00805f9b34fb"
               string "00001200-0000-1000-8000-00805f9b34fb"
               string "0000110c-0000-1000-8000-00805f9b34fb"
               string "00001800-0000-1000-8000-00805f9b34fb"
            ]
      )
   ]
   array [
   ]
signal sender=:1.4 -> dest=(null destination) serial=548 path=/org/bluez/hci0; interface=org.freedesktop.DBus.Properties; member=PropertiesChanged
   string "org.bluez.Adapter1"
   array [
      dict entry(
         string "UUIDs"
         variant             array [
               string "00001112-0000-1000-8000-00805f9b34fb"
               string "0000110a-0000-1000-8000-00805f9b34fb"
               string "00001801-0000-1000-8000-00805f9b34fb"
               string "0000110e-0000-1000-8000-00805f9b34fb"
               string "00001200-0000-1000-8000-00805f9b34fb"
               string "0000110c-0000-1000-8000-00805f9b34fb"
               string "00001800-0000-1000-8000-00805f9b34fb"
            ]
      )
   ]
   array [
   ]
signal sender=:1.4 -> dest=(null destination) serial=550 path=/org/bluez/hci0; interface=org.freedesktop.DBus.Properties; member=PropertiesChanged
   string "org.bluez.Adapter1"
   array [
      dict entry(
         string "UUIDs"
         variant             array [
               string "00001112-0000-1000-8000-00805f9b34fb"
               string "00001801-0000-1000-8000-00805f9b34fb"
               string "0000110e-0000-1000-8000-00805f9b34fb"
               string "00001800-0000-1000-8000-00805f9b34fb"
               string "00001200-0000-1000-8000-00805f9b34fb"
               string "0000110c-0000-1000-8000-00805f9b34fb"
               string "0000110a-0000-1000-8000-00805f9b34fb"
               string "0000111f-0000-1000-8000-00805f9b34fb"
            ]
      )
   ]
   array [
   ]

```

darjusp, 2019-07-19:
> It's really hard to tell, because there can ba a lot of obstacles preventing connection with the BT device. But the best way to tell is to compile bluez-alsa with debug enabled and see the logs. If [...]

Arkadsiusz, thank you a lot. You delivered me to the right place - help section about profile. Maybe someone will face such problem in the future. So problem was not using profile calling: 'bluealsa &' . Adding profile allowed to connect 'bluealsa -p a2dp-sink &'. now i have another issue 'bluealsa-aplay: Couldn't open PCM: No such file or directory' but i won't be spamming here. Thanks again, now i can move forward with my project

Pro-pra, 2019-07-23:
Yes, fixed for me, missing /etc/dbis-1/system.d/bluealsa.conf in my system.

closed 2019-07-23 by Pro-pra