arkq/bluez-alsa #234: Cannot play sound after successful connection

karlkar, 2019-07-29:
Hello.
I'm running manually built version of bluealsa on my Raspberry Pi with Raspbian Buster. I've removed all pulseaudio packages and want to connect to my soundbar and play some music. I have 2 BT interfaces and I want to use hci0 to stream music to the soundbar (this one is a USB dongle).

I've built bluealsa with only one flag --enable-debug and also I've set in .profile file LIBASOUND_THREAD_SAFE=0

For now I am running bluealsa from the command line.
My /etc/asound.conf looks like this:

```
pcm.!default {
    type plug
    slave.pcm "soundbar"
}

pcm.soundbar {
    type plug
    slave.pcm {
        type bluealsa
        interface "hci0"
        device "84:A4:66:44:B8:09"
        profile "a2dp"
    }
}

pcm.sluchawki {
    type plug
    slave.pcm {
        type bluealsa
        interface "hci0"
        device "20:15:01:21:B5:0F"
        profile "a2dp"
    }
}

pcm.s9plus {
    type plug
    slave.pcm {
        type bluealsa
        interface "hci0"
        device "34:2D:0D:F8:E7:FA"
        profile "a2dp"
    }
}

pcm.kruger {
    type plug
    slave.pcm {
        type bluealsa
        interface "hci0"
        device "39:4B:88:E5:CD:09"
        profile "a2dp"
    }
}

pcm.pseye {
    type plug
    slave {
        pcm "hw:1,0"
    }
}

pcm.dmic_hw {
    type hw
    card 1
    channels 1
    format S16_LE
}

pcm.dmic_sv {
    type softvol
    slave.pcm dmic_hw
    control {
        name "Boost Capture Volume"
        card 1
    }
    min_dB -20.0
    max_dB 30.0
}

ctl.!default {
    type bluealsa
    interface "hci0"
}
```

That's intro to my issue ;)
So I'm connecting to my soundbar using bluetoothctl and connection is successful. In bluealsa logs I can see:
```
sudo bluealsa
bluealsa: ../../src/bluez.c:801: Registering endpoint: /org/bluez/hci0/A2DP/SBC/Source/1
bluealsa: ../../src/ba-adapter.c:134: Freeing adapter: hci0
bluealsa: ../../src/bluez.c:801: Registering endpoint: /org/bluez/hci1/A2DP/SBC/Source/1
bluealsa: ../../src/ba-adapter.c:134: Freeing adapter: hci1
bluealsa: ../../src/bluez.c:1065: Registering profile: /org/bluez/HSP/AudioGateway
bluealsa: ../../src/bluez.c:1065: Registering profile: /org/bluez/HFP/AudioGateway
bluealsa: ../../src/main.c:328: Acquiring D-Bus service name: org.bluealsa
bluealsa: ../../src/main.c:333: Starting main dispatching loop
bluealsa: ../../src/bluez.c:721: Called: org.bluez.MediaEndpoint1.SelectConfiguration()
bluealsa: ../../src/bluez.c:1192: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: ../../src/bluez.c:721: Called: org.bluez.MediaEndpoint1.SetConfiguration()
bluealsa: ../../src/bluez.c:636: A2DP Source (SBC) configured for device 84:A4:66:44:B8:09
bluealsa: ../../src/bluez.c:638: Configuration: channels: 2, sampling: 48000
bluealsa: ../../src/ba-transport.c:592: State transition: 0 -> 0
bluealsa: ../../src/bluez.c:801: Registering endpoint: /org/bluez/hci0/A2DP/SBC/Source/2
```

But when I'll try to launch any sound using my device:
```
aplay -D soundbar /usr/share/sounds/alsa/Front_Center.wav
ALSA lib ../../../src/asound/bluealsa-pcm.c:682:(_snd_pcm_bluealsa_open) Unknown field interface
aplay: main:828: audio open error: Invalid argument
```

How can it be fixed? Any idea?

Arkq, 2019-07-29:
> ALSA lib ../../../src/asound/bluealsa-pcm.c:682:(_snd_pcm_bluealsa_open) Unknown field interface
> aplay: main:828: audio open error: Invalid argument

You are using master branch with configuration file designed for bluez-alsa <= v1.4.0 :) There are a lot of changes since v1.4.0, and few of them are not backward compatible (next version of bluez-alsa will be v2.0.0). Please look at the [20-bluealsa.conf](https://github.com/Arkq/bluez-alsa/blob/4d73f961eb0b3fbe16d5b9e01513727dfe07c432/src/asound/20-bluealsa.conf) in the master branch. You have to specify bluealsa D-Bus service name, not a HCI device. Default configuration should work just fine. Also, make sure that D-Bus has configuration file installed, otherwise your connection with the D-Bus service will fail - see #233 .

karlkar, 2019-07-29:
Ok, you're right - after fixing the config (in fact just removing the interface "hci0" and changing it to service "org.bluealsa" it started to work a bit better ;)
Just a bit, because I had to attempt to connect for few times before connection has finally been established. Error shown was:
```
[bluetooth]# connect 84:A4:66:44:B8:09
Attempting to connect to 84:A4:66:44:B8:09
[CHG] Device 84:A4:66:44:B8:09 Connected: yes
Failed to connect: org.bluez.Error.Failed
[CHG] Device 84:A4:66:44:B8:09 Connected: no
[CHG] Device 84:A4:66:44:B8:09 Connected: yes
[CHG] Device 84:A4:66:44:B8:09 Connected: no
```

The other thing is that bluealsa process that has to be running. Back in the days you have provided a .service file which was working. Right now I've tried to use it and I cannot enable it - there's an error that this process isn't intended to be run in such a way. What is the proper way to launch bluealsa process so it awaits on connections?

Arkq, 2019-07-29:
> Back in the days you have provided a .service file which was working.

This file was provided by some Linux distribution. There is no init/systemd integration in bluez-alsa repository. There is too much init systems out there to provide configuration for all of them, so I have decided to provide none :D. Every Linux distribution should provide proper init setup when packaging bluez-alsa.

Regarding bluealsa startup. Right now, things should be much more simple then in v1.4.0. Just start bluealsa during system sturtup (after D-Bus is started and running), and everything should be just fine. Bluealsa shall detect BlueZ/oFono service startup and register with it automatically.

karlkar, 2019-07-30:
Awesome! It really seems to work better than previous version!

One last thing - I am not sure if it's fault of bluealsa or mopidy.
I'm playing some streams using mopidy and all of them work fine. However there's one stream that uses mpeg4 aac encoding. So I've installed gstreamer-1.0-plugin-bad and now the stream plays, but sound is strange. I don't know if that's because I've compiled bluealsa without support for aac or some issue in gstreamer plugin. Do you know how can I find out? Can bluealsa be the issue here?

Arkq, 2019-07-31:
> I don't know if that's because I've compiled bluealsa without support for aac or some issue in gstreamer plugin. Do you know how can I find out? Can bluealsa be the issue here?

It would be rather strange that bluealsa causes this problem. The AAC support in bluealsa means that the BT data can be encoded/decoded with AAC codec. However, the input audio still has to be in the raw PCM format. So, even though you've got audio encoded with AAC, it has to be decoded to PCM and then bluealsa will encode it again (sic!) with AAC. In theory it is possible not to reencode audio, but I do not know if anyone is doing it - except some embedded solutions.

karlkar, 2019-08-02:
Thanks for explanation!

closed 2019-08-02 by karlkar