arkq/bluez-alsa #235: after connection: aplay error "Couldn't get BlueALSA PCM: Rejected send message"

clwgg, 2019-08-03:
Hi,
I am struggling to set up BlueALSA with my Bose QC35 II headphones on my Thinkpad T480s running Manjaro Linux. I have compiled the latest github version of BlueALSA (1.4.0) with `--enable-debug`, and connected the headphones using `bluetoothctl`. You find the logs below. Also, I have set up the device in `~/.asoundrc` (also below). Once I do all that, if i run:

```
$ LIBASOUND_THREAD_SAFE=0 aplay /usr/lib/libreoffice/share/gallery/sounds/nature2.wav
```
I get the error:
```
ALSA lib bluealsa-pcm.c:718:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: Rejected send message, 1 matched rules; type="method_call", sender=":1.6102" (uid=1000 pid=16254 comm="aplay /usr/lib/libreoffice/share/gallery/sounds/na") interface="org.bluealsa.Manager1" member="GetPCMs" error name="(unset)" requested_reply="0" destination="org.bluealsa" (uid=0 pid=16013 comm="./src/bluealsa ")
aplay: main:828: audio open error: No such device
```

Some version numbers:
```
extra/bluez 5.50-6 [installed]
extra/alsa-firmware 1.0.29-2 [installed]
extra/alsa-lib 1.1.9-1 [installed]
Kernel: 4.19.62-1-MANJARO
```

Any help would be greatly appreciated!

Some logs:
```
$ sudo ./src/bluealsa 

./src/bluealsa: ../../src/bluez.c:801: Registering endpoint: /org/bluez/hci0/A2DP/MPEG24/Source/1
./src/bluealsa: ../../src/bluez.c:801: Registering endpoint: /org/bluez/hci0/A2DP/SBC/Source/1
./src/bluealsa: ../../src/ba-adapter.c:134: Freeing adapter: hci0
./src/bluealsa: ../../src/bluez.c:1065: Registering profile: /org/bluez/HSP/AudioGateway
./src/bluealsa: ../../src/bluez.c:1065: Registering profile: /org/bluez/HFP/AudioGateway
./src/bluealsa: ../../src/main.c:328: Acquiring D-Bus service name: org.bluealsa
./src/bluealsa: ../../src/main.c:333: Starting main dispatching loop
./src/bluealsa: ../../src/bluez.c:1192: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
./src/bluealsa: ../../src/bluez.c:1192: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
./src/bluealsa: ../../src/bluez.c:1192: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
./src/bluealsa: ../../src/bluez.c:1192: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
./src/bluealsa: ../../src/bluez.c:1010: Called: org.bluez.Profile1.NewConnection()
./src/bluealsa: ../../src/bluez.c:929: RFCOMM: HFP Audio Gateway configured for device <<MAC:ADDRESS>>
./src/bluealsa: ../../src/ba-transport.c:592: State transition: 0 -> 2
./src/bluealsa: ../../src/io.c:2052: Created new IO thread: RFCOMM: HFP Audio Gateway
./src/bluealsa: ../../src/rfcomm.c:597: Starting loop: RFCOMM: HFP Audio Gateway
./src/bluealsa: ../../src/ba-transport.c:592: State transition: 0 -> 2
./src/bluealsa: ../../src/io.c:2052: Created new IO thread: HFP Audio Gateway
./src/bluealsa: ../../src/io.c:1565: Starting IO loop: HFP Audio Gateway
./src/bluealsa: ../../src/at.c:161: AT message: SET: command:+BRSF, value:703
./src/bluealsa: ../../src/rfcomm.c:104: Sending AT message: RESP: command:+BRSF, value:480
./src/bluealsa: ../../src/rfcomm.c:104: Sending AT message: RESP: command:(null), value:OK
./src/bluealsa: ../../src/rfcomm.c:123: HFP state transition: 0 -> 2
./src/bluealsa: ../../src/at.c:161: AT message: TEST: command:+CIND, value:(null)
./src/bluealsa: ../../src/rfcomm.c:104: Sending AT message: RESP: command:+CIND, value:("call",(0,1)),("callsetup",(0-3)),("service",(0-1)),("signal",(0-5)),("roam",(0-1)),("battchg",(0-5)),("callheld",(0-2))
./src/bluealsa: ../../src/rfcomm.c:104: Sending AT message: RESP: command:(null), value:OK
./src/bluealsa: ../../src/rfcomm.c:123: HFP state transition: 2 -> 5
./src/bluealsa: ../../src/at.c:161: AT message: GET: command:+CIND, value:(null)
./src/bluealsa: ../../src/rfcomm.c:104: Sending AT message: RESP: command:+CIND, value:0,0,0,0,0,0,0
./src/bluealsa: ../../src/rfcomm.c:104: Sending AT message: RESP: command:(null), value:OK
./src/bluealsa: ../../src/rfcomm.c:123: HFP state transition: 5 -> 7
./src/bluealsa: ../../src/at.c:161: AT message: SET: command:+CMER, value:3, 0, 0, 1
./src/bluealsa: ../../src/rfcomm.c:104: Sending AT message: RESP: command:(null), value:OK
./src/bluealsa: ../../src/rfcomm.c:123: HFP state transition: 7 -> 8
./src/bluealsa: ../../src/rfcomm.c:123: HFP state transition: 8 -> 9
./src/bluealsa: ../../src/rfcomm.c:104: Sending AT message: RESP: command:+BCS, value:1
./src/bluealsa: ../../src/at.c:161: AT message: SET: command:+VGS, value:10
./src/bluealsa: ../../src/rfcomm.c:104: Sending AT message: RESP: command:(null), value:OK
./src/bluealsa: ../../src/rfcomm.c:104: Sending AT message: RESP: command:+BCS, value:1
./src/bluealsa: ../../src/at.c:161: AT message: GET: command:+BTRH, value:(null)
./src/bluealsa: ../../src/rfcomm.c:104: Sending AT message: RESP: command:(null), value:OK
./src/bluealsa: ../../src/rfcomm.c:104: Sending AT message: RESP: command:+BCS, value:1
./src/bluealsa: ../../src/at.c:161: AT message: SET: command:+XAPL, value:009E-4020-0318,3
./src/bluealsa: ../../src/rfcomm.c:104: Sending AT message: RESP: command:(null), value:+XAPL=BlueALSA,0
[... 51 lines cut ...]
[CHG] Device <<MAC:ADDRESS>> Name: Bose QC35 II
[CHG] Device <<MAC:ADDRESS>> Alias: Bose QC35 II
Connection successful

```

```
$ LIBASOUND_THREAD_SAFE=0 aplay /usr/lib/libreoffice/share/gallery/sounds/nature2.wav
ALSA lib bluealsa-pcm.c:718:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: Rejected send message, 1 matched rules; type="method_call", sender=":1.6102" (uid=1000 pid=16254 comm="aplay /usr/lib/libreoffice/share/gallery/sounds/na") interface="org.bluealsa.Manager1" member="GetPCMs" error name="(unset)" requested_reply="0" destination="org.bluealsa" (uid=0 pid=16013 comm="./src/bluealsa ")
aplay: main:828: audio open error: No such device

$ cat .asoundrc 
pcm.!default {
      type plug
      slave.pcm {
              type bluealsa
              device "<<MAC:ADDRESS>>"
              profile "a2dp"
      }
      hint {
              show on
              description "BT Headset"
      }
}
```

clwgg, 2019-08-16:
@Arkq do you have any pointers on how to debug this, or an idea if the problem is with the pairing itself rather than within BlueALSA? I am a little confused by the `rfcomm` activity in the bluealsa log, though I don't have any experience on what that means. Any pointers would be appreciated, I'd love to avoid installing pulse.

Arkq, 2019-08-17:
Firstly look at these issues: https://github.com/Arkq/bluez-alsa/issues/233 https://github.com/Arkq/bluez-alsa/issues/234#issuecomment-516092455 and make sure that everything is correctly configured (D-Bus and .asoundrc).