arkq/bluez-alsa #238: [wiki] draft page documenting systemd usage

borine, 2019-08-16:
I've drafted some brief guidelines on using bluealsa with systemd. This may be useful in the wiki here, so please feel free to adapt as appropriate. You will find the document here: https://github.com/borine/bluez-alsa-wiki/blob/master/Systemd-integration.md

Arkq, 2019-08-16:
Great idea. But I think that you can write this guide in the bluez-alsa
wiki by yourself. You should have the access for editing (it is public
wiki), if not please let me know. Just create new page and then add it to
the main page.

Also regarding the "on demand" functionality. What is the exact use case?
Because if client wants to use Bluetooth PCM, such PCM has to exists, and
it won't exist if BT device is not connected, and BT device will not be
connected without running bluealsa daemon :) There is a one case where that
might work - oFono HFP profile :p

borine, 2019-08-16:
oops - ok, so that was the half-baked edition; I've put a more complete solution directly onto the wiki here as requested.Hopefully this one makes more sense.

Arkq, 2019-08-17:
Great! Many thanks for this wiki page :) If anything pops up in your mind, feel free to add it to the wiki.

closed 2019-08-17 by Arkq