arkq/bluez-alsa #240: aplay stay silent with files - sample rate?

yogo1212, 2019-08-24:
version 1.4.0

this produces no sound:
```
aplay Heart.wav
Playing WAVE 'Heart.wav' : Unsigned 8 bit, Rate 22050 Hz, Mono
```

this does:
```
play 01\ Virgin\ Pullets.mp3 
play WARN alsa: can't encode 0-bit Unknown or not applicable

01 Virgin Pullets.mp3:

 File Size: 1.08M     Bit Rate: 128k
  Encoding: MPEG audio    
  Channels: 2 @ 16-bit   Track: 1/13
Samplerate: 44100Hz      Album: English, Welsh, Scottish & Irish Fiddle Tunes
Replaygain: off         Artist: Robin Williamson
  Duration: 00:01:07.47  Title: Virgin Pullets
```

using mpg123 and aplay, i got various other results - white noise, halved tempo, and such.
```
mpg123 -s 01\ Virgin\ Pullets.mp3 --stereo --enc s16 -rate 44100  | aplay -f S16_LE -c2 -r44100 -D btheadset -
```

is it possible to get an error message when a format isn't supported?
or does that happen inside the client device?

killerkalamari, 2019-09-05:
I have the same silence problem playing via portaudio on sample rates other than 24000/48000 (with portaudio spitting out errors). So far, I've been able to get it to play via a custom .asoundrc that resamples the audio, fixes channels, etc (see #243), however, in my case it truncates the audio. Still, maybe it'd work for you, as I suspect the truncation is due to an interaction between portaudio and bluez-alsa.

borine, 2019-09-08:
Could you post the contents of your .asoundrc file (or whichever file contains the definition of "btheadset"?
The warning "play WARN alsa: can't encode 0-bit Unknown or not applicable" is from the SOX library - it is not caused by bluealsa.

Also, do you know if the alsa installation you are using was compiled with its "thread safety" featues enabled? If it is you will need to set the environment variable "LIBASOUND_THREAD_SAFE=0" when running aplay, play, or mpg123 for outputting to bluealsa.

In a default installation of bluealsa, the alsa device called "bluealsa" is a plug device wrapping the actual bluealsa interface. The plug *should* perform any format transformations required (number of channels, sample rate, etc) so that any PCM stream plays correctly.

yogo1212, 2019-09-09:
hello and thanks for your response :-)

```
pcm.btheadset {
	type plug
	slave {
		pcm {
			type bluealsa
			device 00:02:3C:5D:3D:D0
			profile "a2dp"
		}
	}
	hint {
		show on
		description "BT Headset"
	}
}
ctl.btheadset {
	type bluealsa
}
```

My alsa installation is what comes with Arch Linux - but i'll check what it is and try the env variable.

So, it should be possible to make it run.
I'll give it another go, then

borine, 2019-09-10:
I've just tried mpg123 with bluealsa and I can't get it to work at all. I'll look into it further over the next few days and report what I find.

yogo1212, 2019-09-10:
i remember having found parameters that worked with at least one file.
if we could meet in a shady area, we could commit copirat-infringemocide ;-)

borine, 2019-09-11:
mpg123 is weird - it opens the pcm 4 times in total, and with 2 of these goes on to initialise the (virtual) hardware. This causes the device to start 2 PCM transports, one of which is then closed before playback begins.
Unfortunately, bluealsa currently cannot cope with this behaviour. I suspect because of a race condition in the daemon, closing one of the transports leaves the other in an inconsistent state in which it never reads audio data from the alsa client. The client (mpg123) is then blocked in a write() call because the FIFO is full. 
There is also some xrun wierdness going on when mpg123 first starts the pcm, which is a different issue which I think may impact other applications too.
I will try to prepare 2 PRs which attempt to address these two problems, and see whether @Arkq thinks they are worth considering for inclusion.

borine, 2019-09-15:
Looking at your .asoundrc, but not knowing what defaults you have set, I think the following commands should work for you. To get sox play to work reliably with bluealsa you will need to build bluealsa from source as it requires commit 2cd6e46 which was made yesterday. mpg123 still has some issues, but I think files sampled at 44.1kHz with 2 channels and s16le format (eg your Virgin Pullets sample) should work ok if your headset uses  48kHz (see issue #247 for reasons why).

To get debug info from bluealsa you need to add the configure flag --enable-debug when you build it.

So, assuming you have up-to-date bluealsa build:

```
LIBASOUND_THREAD_SAFE=0 aplay -D btheadset Heart.wav

LIBASOUND_THREAD_SAFE=0 AUDIODEV=btheadset play 01\ Virgin\ Pullets.mp3

LIBASOUND_THREAD_SAFE=0 mpg123 -o alsa -a btheadset 01\ Virgin\ Pullets.mp3
```
If you are able build bluealsa and test this, please report the results back here, I would like to see independent confirmation that the change in the above commit is working as expected.

Arkq, 2021-01-18:
I think that this problem was resolved with @borine patches (available in bluez-alsa-3.0.0). I'm closing this issue, then.

closed 2021-01-18 by Arkq