arkq/bluez-alsa #241: I miss   void *io_thread_a2dp_sink_aptx(void *arg)

gpeter12, 2019-08-28:
Well guys I really really miss a void *io_thread_a2dp_sink_aptx(void *arg) function from io.c. I tried to work on that my phone recognize my RPi Zero W as an aptx-HD capable sink device in the last two days. Than I noticed in the source code that bluealsa isn't capable such thing. It can only receive sound through sbc and aac (as I can see). I'm wondering now: how hard can it be to fix that problem. Is it solvable? Is it a "mission impossible" kind of problem? Am I completely wrong about that and this feature is already implemented?
Thanks for the help!

Arkq, 2019-09-03:
> Is it solvable? Or is it a "mission impossible" kind of problem?

Of course it is solvable. However, right now there is no official library from Qialcomm. There is reverse engineered support in ffmpeg, however, I'm not 100% convinced that the HD codec works correctly (I might be wrong, though). I've made some test with the library found on Android and ffmpeg and they produced different output... (or my test was incorrect). Anyway, I've got no real device with aptxHD support, so it is hard for me to test developed code. Hence, there is no aptxHD support. It will me in the future for sure :)

powerpun, 2019-09-12:
how I could help on this, bought fiio btr3, claimed support aptx(hd)/ldac/hwa...

powerpun, 2019-09-13:
@gpeter12 I got the same problem using common pulseaudio in RPi3 also with my own platform. I had to say bluezalsa provide a lot better quality than pulse while playing via A2DP mode(but pulseaudio default is SBC while bluealsa was using AAC) I hear the click during playing, no clue.

t123yh, 2019-10-01:
Please have a try on #250 .

gpeter12, 2019-10-07:
@t123yh Oh man It's working!!! Thank you very very much!
It has only one issue: I hear a crackling noise. It's a couple milliseconds long and It happens 1-2 times in 5 minutes. I guess it's caused by buffer underruns or damaged bluetooth packets.
I tested your code on a RPi 3B with Asus Zenfone 5Z. I will test it soon on another device (RPi Zero W) with a USB DAC.

Arkq, 2021-01-16:
I'm closing this issue, because since commit b11cff0c2fd436e64c98b77943bb20a2d3d0ea77 apt-X / apt-X HD decoding support is available in the master branch.

closed 2021-01-16 by Arkq