arkq/bluez-alsa #243: End of audio being truncated with 22kHz sample rate on portaudio

killerkalamari, 2019-09-04:
When I play 32-bit integer 22kHz audio, via portaudio and a custom .asoundrc, the end of the sound is truncated. It seems to be okay for 44.1kHz and 48kHz. 

The reason for the selection of 32-bit integer samples (rather than 16-bit integer) was to get resampling to work with portaudio (otherwise I'd get an error). Likewise the rate plugin was needed (and then plug to get around an error playing mono sounds, and then route to handle mono sounds in the wrong speaker), so my .asoundrc got a little complicated:

```
pcm.bluetooth {
	type route
	ttable.0.0 0.5
	ttable.0.1 0.5
	ttable.1.0 0.5
	ttable.1.1 0.5
	slave.pcm {
		type rate
		converter "speexrate"
		slave {
			rate 48000
			pcm {
				type plug
				slave.pcm {
					type bluealsa
					device "6B:F4:54:06:AC:D1"
					profile "a2dp"
				}
			}
		}
	}
	hint {
		show on
		description "Bluetooth (bluez-alsa)"
	}
}
```
It seems possible that the myriad plugins I have to use to get this to work might be contributing to the problem.

I saw #67 however in my case, aplay works fine, because it's not being routed through portaudio. Based on that, I tried increasing ba-transport.c usleep(200000) to usleep(1000000), but it didn't seem to make a difference.

Please note that playing to a non-bluez-alsa device works fine in portaudio and I don't need a custom .asoundrc or sample size for those.

Any ideas on this would be very much appreciated!

borine, 2019-09-15:
commit 2cd6e46 fixes an issue bluealsa had with repeatedly opening/closing/reopening a pcm device. As portaudio uses this technique to test the capabilities of a device before selecting it, it is possible that this commit may also improve the behaviour of bluealsa with portaudio applications.
I'm not sure how much of your complex configuration is just to overcome issues, but it would be great if you could test the latest bluealsa code with a default configuration and with the config above to compare results with the bluealsa code version you have used so far.

killerkalamari, 2019-09-16:
Thanks! I've gone ahead and given that commit a try, but I'm still unable to play audio with a non-crazy .asoundrc. Here are some details if you'd like to try to recreate:

PortAudio example program (modified by me to try out different sample rates and formats, but it should do everything the original does and more, although I'm not sure I got mono playback correct):
https://pastebin.com/CHtV2gWG

Just in case, here's the original PortAudio example (NOTE: works because it's using 32-bit floats... why that matters remains a mystery to me. To run this one you'd need to modify my OP .asoundrc to include !default):
https://github.com/EddieRingle/portaudio/blob/master/examples/paex_sine.c

Compile:
```
gcc -W -Wall -o paex_sine_ext paex_sine_ext.c -lm -lportaudio
```

.asoundrc reverted to what I hope is standard:
```
pcm.bluetooth {
	type plug
	slave.pcm {
		type bluealsa
		device "E9:08:EF:2D:40:33"
		profile "a2dp"
	}
	hint {
		show on
		description "Bluetooth (bluez-alsa)"
	}
}
```

Run (NOTE: 2 happens to be the bluetooth device for me, it might be different for you. i16 has always failed for me with a non-48000 sample rate, but i32 or f32 should work with my OP .asoundrc):
```
pa/paex_sine_ext 2 i16 stereo 5 44100 64
```

paex_sine_ext output (NOTE: I've added a bit of debug code to PortAudio to look into the framesAvail error, trivial patch at https://pastebin.com/U2ZDWVBh, unfortunately it didn't help me understand what went wrong):
```
PortAudio Test: output sine wave. f = 8, stereo, NS = 5, SR = 44100, BufSize = 64
before paHostApiInitializers[0].
ALSA version (build): 1.1.8
ALSA version (runtime): 1.1.8
BuildDeviceList: Ignoring ALSA plugin device [cards] of type [unknown]
BuildDeviceList: Found plugin [default] of type [unknown]
BuildDeviceList: Found plugin [sysdefault] of type [unknown]
BuildDeviceList: Found plugin [front] of type [unknown]
BuildDeviceList: Found plugin [rear] of type [unknown]
BuildDeviceList: Found plugin [center_lfe] of type [unknown]
BuildDeviceList: Found plugin [side] of type [unknown]
BuildDeviceList: Found plugin [surround21] of type [unknown]
BuildDeviceList: Found plugin [surround40] of type [unknown]
BuildDeviceList: Found plugin [surround41] of type [unknown]
BuildDeviceList: Found plugin [surround50] of type [unknown]
BuildDeviceList: Found plugin [surround51] of type [unknown]
BuildDeviceList: Found plugin [surround71] of type [unknown]
BuildDeviceList: Found plugin [iec958] of type [unknown]
BuildDeviceList: Found plugin [spdif] of type [unknown]
BuildDeviceList: Found plugin [hdmi] of type [unknown]
BuildDeviceList: Found plugin [dmix] of type [unknown]
BuildDeviceList: Ignoring ALSA plugin device [dsnoop] of type [unknown]
BuildDeviceList: Found plugin [modem] of type [unknown]
BuildDeviceList: Found plugin [phoneline] of type [unknown]
BuildDeviceList: Ignoring ALSA plugin device [hw] of type [hw]
BuildDeviceList: Ignoring ALSA plugin device [plughw] of type [plug]
BuildDeviceList: Ignoring ALSA plugin device [plug] of type [plug]
BuildDeviceList: Ignoring ALSA plugin device [shm] of type [shm]
BuildDeviceList: Ignoring ALSA plugin device [tee] of type [file]
BuildDeviceList: Ignoring ALSA plugin device [file] of type [file]
BuildDeviceList: Ignoring ALSA plugin device [null] of type [null]
BuildDeviceList: Found plugin [bluealsa] of type [plug]
BuildDeviceList: Found plugin [bluetooth] of type [plug]
BuildDeviceList: Filling device info for 21 devices
[... 184 lines cut ...]

bluealsa output during run:
```
/usr/bin/bluealsa: bluealsa-dbus.c:328: Called: org.bluealsa.PCM1.Open()
/usr/bin/bluealsa: ba-transport.c:722: New transport: 20 (MTU: R:672 W:895)
/usr/bin/bluealsa: bluez.c:1330: Signal: org.freedesktop.DBus.Properties.PropertiesChanged()
/usr/bin/bluealsa: bluez.c:1364: Signal: PropertiesChanged: org.bluez.MediaTransport1: State
/usr/bin/bluealsa: ba-transport.c:601: State transition: 0 -> 2
/usr/bin/bluealsa: io.c:2576: Created new IO thread: A2DP Source (SBC)
/usr/bin/bluealsa: io.c:469: Starting IO loop: A2DP Source (SBC)
/usr/bin/bluealsa: ba-transport.c:601: State transition: 2 -> 2
/usr/bin/bluealsa: io.c:126: PCM read buffer flushed: 1512
/usr/bin/bluealsa: io.c:126: PCM read buffer flushed: 0
/usr/bin/bluealsa: io.c:111: PCM has been closed: 16
/usr/bin/bluealsa: ba-transport.c:882: Closing PCM: 16
/usr/bin/bluealsa: ba-transport.c:882: Closing PCM: 16
/usr/bin/bluealsa: io.c:520: Keep-alive polling: 0
/usr/bin/bluealsa: io.c:520: Keep-alive polling: 0
/usr/bin/bluealsa: ba-transport.c:749: Releasing transport: A2DP Source (SBC)
/usr/bin/bluealsa: ba-transport.c:778: Closing BT: 20
/usr/bin/bluealsa: ba-transport.c:928: Exiting IO thread: A2DP Source (SBC)
/usr/bin/bluealsa: bluez.c:1330: Signal: org.freedesktop.DBus.Properties.PropertiesChanged()
/usr/bin/bluealsa: bluez.c:1364: Signal: PropertiesChanged: org.bluez.MediaTransport1: State
/usr/bin/bluealsa: ba-transport.c:601: State transition: 2 -> 0
```

borine, 2019-09-17:
I don't get any sound at all when using your paex_sine_ext program.

I have some sample mp3 files with sample rates of 22.05kHz, 44.1kHz, and 48kHz, and sample formats of s16le, s32le, f32le. I tested them all with mpg123-portaudio (which is the only portaudio application I have available right now) and they all appear to work fine with a default install of bluealsa. So I'm not convinced this is a bluealsa issue.

The thing is, bluealsa does not do any pcm format conversions - the a2dp link will always be stereo s16le (defined by the a2dp spec) and the rate is nearly always 48kHz (although could possibly be 44.1kHz, but the application does not get to choose, its fixed when the bluetooth connection is made). The bluealsa plugin will only accept this one format  when the application sets the hardware params. As a convenience, a default bluealsa install also defines an alsa pcm device called "bluealsa" which uses the alsa "plug" plugin to do any necessary conversions. If you have a file which is not converted correctly when playing to this device, then does that not suggest it is the converter which is failing rather than the bluealsa code?

killerkalamari, 2019-09-19:
Thanks for taking a look! Is there any way to use the bluealsa pcm device and its conversions via .asoundrc? The reason I need that is to be able to specify the bluetooth address. Should my .asoundrc be able to use the bluealsa device resampling abilities, or is it flawed in some way?

borine, 2019-09-20:
I'm no expert in alsa configuration, but your .asoundrc looks ok but rather over-complicated as far as I can tell. You wrap the bluealsa plugin within a "plug" plugin which does the automatic channels/format/rate conversion for you; but you then wrap that in "speexrate" and "route" which seems rather redundant? Have a look at the section "Automatic Conversion Plugin" in https://www.alsa-project.org/alsa-doc/alsa-lib/pcm_plugins.html - that shows how to use the "plug" plugin to do the required conversions.

My concern is that this complex setup all resulted from a perceived failure of port audio to handle 16-bit samples at 22kHz. Yet I can see no reason why port audio should fail there. Is the failure only when using bluealsa, or is it with all devices? If all devices, then I would concentrate on solving it with your on-board card before adding the extra complexity of third-party plugins such as bluealsa.

killerkalamari, 2019-09-24:
My apologies, I was referring to the reverted .asoundrc from this comment, was it correct?
https://github.com/Arkq/bluez-alsa/issues/243#issuecomment-531912301 

Audio plays fine with the internal non-bluealsa devices via PortAudio. 

FWIW, I refactored my application to play directly to ALSA and it plays 16-bit audio fine now, at all sample rates. Go ahead and close if you want, although I do believe there is something weird happening between PortAudio and bluealsa, perhaps in the PortAudio code? Hard to say.

borine, 2019-09-24:
@killerkalamari - if your application works fine with alsa hw devices and not bluealsa, then clearly I was wrong - this is a bluealsa issue.

I'm not the bluealsa developer, just someone interested in understanding issues with it for my own benefit. I'm afraid that at present, I can't see where or how bluealsa can cause the "framesAvail" error you are seeing. It may need someone with knowledge of portaudio internals to explain that.

So I think this issue should be left open because, as you say, there is possibly some unexpected interaction between bluealsa and portaudio that might well impact other applications.

borine, 2021-06-22:
This issue has been idle for over 18 months now, Given the significant changes in bluez-alsa in that time, and the fact that I am unable to reproduce the issue, I am closing.

If you experience problems using the latest bluez-alsa code with portaudio please open a new issue, including error/debug logs from bluealsa if possible.

closed 2021-06-22 by borine