arkq/bluez-alsa #244: Is it possible to record from and play on a bluetooth device?

crocket, 2019-09-07:
I want to record from and play on a bluetooth device simultaneously.
Is that possible?

Arkq, 2019-09-14:
Yes, it should be possible (if you're talking about A2DP). Just connect both devices and use e.g. `bluealsa-aplay` to play audio from iPhone. However, it is not possible right now to stream more than one stream into bluetooth speaker (there is no mixer), incoming audio has to be played on wired speaker.

closed 2019-09-14 by Arkq

crocket, 2019-09-14:
Is it possible to use ALSA dmix to combine two streams into one and send it to bluetooth speaker?

Arkq, 2019-09-15:
No, unfortunately it is not possible. These plug-ins require hardware sound dard with timer which is used for synchronization. See #48.

crocket, 2019-09-15:
What if ALSA dmix outputs to SPDIF output which is connected to SPDIF input which then is read from bluealsa?

Arkq, 2019-09-15:
The problem right now is that bluealsa can handle only one PCM stream (input and output) and the usage of dmix is not possible. So everything, which allows to externally mix  PCM streams and forward them to bluealsa might do the trick. You have to try it and see the results, though.

crocket, 2019-09-15:
I already use SPDIF trick to route ALSA dmix to JACK although I am going to remove JACK and use ALSA only. I guess it can also work for bluealsa as long as bluealsa can read from SPDIF input.

Arkq, 2019-09-15:
> I guess it can also work for bluealsa as long as bluealsa can read from SPDIF input.

If it's standard ALSA interface (PCM), then you can use arecord|aplay, otherwise you will have to create some forwarding (or modify bluealsa to support reading from SPDIF). You might use bluealsa D-Bus API (omit ALSA library which might add additional overhead and delay). See bluealsa-aplay for reference implementation.

crocket, 2019-09-15:
You can also use ALSA loopback device, but when I used ALSA loopback, it caused a lot of pops.

https://alsa.opensrc.org/Jack_and_Loopback_device_as_Alsa-to-Jack_bridge