arkq/bluez-alsa #245: bug reported

powerpun, 2019-09-14:
bluealsa: AAC decode frame error: Transport sync error

android phone connected to bluetooth headphone and bluealsa, when incoming phone call :
bluealsa: AAC decode frame error: Transport sync error

bluealsa seg fault.

Arkq, 2019-09-14:
Could you attach tcpdump of BT connection with such crash? Remember to
start tcpdump before BT connection establishment.

powerpun, 2019-09-14:
I'm busy in other codes recently, will do it ASAP.

powerpun, 2019-09-14:
@Arkq please guide how I should use tcpdump to dump data for you, or I might dump the wrong information then that's a waste of time.

Arkq, 2019-09-14:
Use something like this: `tcpdump -i bluetooth0 -w /tmp/xxx.dump`. And remember to run this command and THEN connect BT device, otherwise it might be impossible to properly decode BT packets. Also, I hope that the crash happens very quickly, otherwise this dump file might grow very huge...

Arkq, 2019-09-14:
> BTW, is it any way to make LDAC/APTX sink work?

For LDAC, it is currently impossible since there is no decoder library. APTX is semi possible, but it is not implemented (in bluez-alsa) - I've got no device for testing so it is rather hard for me to code it properly.

powerpun, 2019-09-16:
[dump.gz](https://github.com/Arkq/bluez-alsa/files/3616647/dump.gz)
test condition: connect the phone to one bt headphone and try to connect to bluealsa while bluealsa in background, bluealsa-aplay 0:0:0:0:0:0, 
when music playing in one bluetooth headphone, on connection to bluealsa it must die.
if only bluealsa in background, it will not die, start aplay then it plays.

dumpfile record all above operations...

powerpun, 2019-09-16:
and, if pulseaudio could be a sink for aptx, then could we try to make it work ?
or I could send you a aptxhd module ?

powerpun, 2019-09-17:
one more things, I wonder if I could send a command to BT player for pause/play/stop ?

Arkq, 2019-09-17:
> one more things, I wonder if I could send a command to BT player for
pause/play/stop ?

Yes, you should use AVRCP. It is not wrapped by bluez-alsa, though. So, you
have to code it by yourself, or find some ready to use tool.

macmpi, 2019-09-20:
@Arkq any recommendation for such AVRCP ready-to-use tool (Debian for instance)?

Arkq, 2019-09-23:
@macmpi sorry, but I don't know any... You might try to use BlueZ D-Bus API with e.g. `gdbus`:
```gdbus call --system -d org.bluez -o /org/bluez/hci0/dev_1C_48_F9_9D_81_5C/fd2 -m org.freedesktop.DBus.Properties.Set org.bluez.MediaTransport1 Volume '<uint16 30>'```
but use "Player" API.

Arkq, 2020-03-15:
This SIGSEGV is caused by the incorrect AAC stream send by your smartphone (Huawei Mate 20 Pro... maybe you should switch to iPhone :) ). Please, look at the "Data:" section on the bottom (selected row). It should start with `47fc`, but for some reasons, when you start playback (Seq=1) it starts with `0147fc`. The rest of the frames is correct, so it should play well, but this very first frame causes fdk-aac library to SIGSEGV. There is nothing I can do about this - it shall be fixed in fdk-aac library.

[image]

closed 2020-03-15 by Arkq