arkq/bluez-alsa #249: BT devices being turned on causes corruption of ALSA mixer

spl237, 2019-09-17:
This may well be an error on my part rather than an actual bug, but I cannot figure out how to stop it happening, and it is completely breaking my system, so I'd really appreciate some advice!

I'm using a Raspberry Pi running Raspbian. The custom volume plugin which is part of lxpanel uses standard ALSA configuration in .asoundrc to select Bluealsa as a device:

pcm.!default {
   type asym
   playback.pcm {
      type plug
      slave.pcm "output"
   }
   capture.pcm {
      type plug
      slave.pcm "input"
   }
}

pcm.output {
   type "bluealsa"
   device "xx:xx:xx:xx:xx:xx" (where xx... is address of output Bluetooth device)
   profile "a2dp"
}

ctl.!default {
   type bluealsa
}

The volume plugin associates a mixer with the Bluealsa device using the standard ALSA calls (snd_mixer_open, snd_mixer_attach, snd_mixer_selem_register, snd_mixer_load etc.)

This all works as expected.

The problem arises when I have multiple Bluetooth audio devices paired with the Pi. If the mixer is associated with one particular device, and a second device which is already paired with the Pi is turned on, this creates an event within ALSA / Bluealsa which corrupts the mixer; the mixer type changes from 0 to a random number, and the mixer is invalid thereafter.

I've done some debugging. What I think is happening is that bluealsa_dbus_msg_filter is being called when the new device turns on - as it is a previously-paired device, a connection is automatically created, and a DBus message is presumably created by BlueZ which tells Bluealsa that a device has connected.

The signal which arrives at this function seems to be PCMAdded on the interface manager, which results in the code at the label remove_add being called; as the comment suggests, this then removes all existing elements and creates a new list of elements - this is what I believe is rendering the mixer invalid, as presumably the element with which ALSA is currently associated is removed and re-added at a different address as part of this process; hence the mixer at the previous address is no longer valid.

Given this removal and re-creation of elements is presumably deliberate, there needs to be (and presumably already is...) some mechanism whereby my code can tell that the existing mixer is no longer valid and can locate the new mixer. By trial and error, I have established that it is not safe to access the memory or call any of the functions on the mixer once the elements have changed - so how can I find out that the mixer is no longer valid and create a new one? Is there some event at the ALSA level that I can see?

Alternatively, is there some way to prevent Bluealsa from removing all existing elements from the mixer, and just to add any new ones?

spl237, 2019-09-18:
Further investigation suggests that I need to monitor the PCMAdded and PCMRemoved signals on the org.bluealsa.Manager1 DBus object, and reassociate the mixer when either occurs - that does seem to (mostly) fix the problem.

Arkq, 2019-09-18:
Hmm, it's hard to tell who is to blame. I'm testing mixer plugin with `amixer` and `alsamixer`. The thingy with removing PCMs and adding them back is to "sort them".

Consider two BT devices (e.g. headsets) "Sony MX" and "JBL". If one runs `alsamixer`, then connects Sony first, then JBL, the order of PCMs in the `alsamixer` will be determined by the connection order. This will be inconsistent when running `alsamixer` when both devices are connected (sorted by name - that's the only natural sorting for BT devices IMHO). I had no idea how to allow proper PCM ordering other than remove and add action.

But maybe the `alsamixer` scenario is too simple, and I have to consider other possibilities :D However, I think that sice `alsamixer` is able to detect remove&add action, it is possible to detect it via ALSA ctl API...

spl237, 2019-09-19:
I have to admit that I was confused by paired devices which connect immediately appearing in alsamixer as valid control options when a bluealsa device is the default, and I assume that is what is causing the problem.

I would expect to see only the specific Bluetooth device which is selected as the default ALSA Bluetooth device (by having the address defined in .asoundrc) as the one which Bluealsa uses. I can see that it is potentially useful to have the other devices appear - as otherwise switching between them can be difficult - but it seems counter-intuitive to me that after I have hard-coded the address of the device I want ALSA to use in .asoundrc, devices with different addresses also appear in the mixer.

I'm not sure that this auto-connection and auto-association of other devices is useful in every circumstance (and I think it is what is causing the problem I am seeing) - perhaps it could be made optional with a parameter in the config file or similar?

In terms of detecting add and remove of devices, I've never found a way to get ALSA to tell me this - it would be extremely useful, because when a USB audio device is removed, it leaves things in a high power drain state - but I'm not convinced it is possible via ALSA itself. Monitoring the org.bluealsa signals I mentioned above on DBus does seem to work for Bluetooth devices - there is also the option of monitoring BlueZ for object added / removed signals, but that then generates messages for non-audio devices as well, which is non-ideal.

spl237, 2019-09-19:
I think there's another problem with this...

If I have two paired BT audio output devices, and I have one of them powered on and selected as the output device in .asoundrc. I then have a mixer associated with that, and I can control the volume on that device with the mixer volume control, as expected.

If I then turn on the second BT device, it connects. I get a PCMAdded signal from the monitor, so I delete and recreate the mixer, but it should still be associated with the device selected in .asoundrc, as that hasn't changed. But the volume control no longer controls the volume on my original device. If I look in alsamixer, it appears that what has happened is that the mixer associated with bluealsa is now attached to the new device, but that audio output is still being routed to the original device. So I have a working volume control, but it is no longer controlling the output that it was, and is now in fact controlling an output which is not being used.

This automatic connection of devices seems to be causing a lot of problems - could the code not (at least optionally) only connect BlueALSA to the device selected in .asoundrc? I think that would make things a lot more robust in the environment where multiple devices are being used. (Admittedly most people will only use a single device, but multiple devices definitely causes problems in the current implementation!)

spl237, 2019-09-19:
Now I think about it...

Should the device on which volume is to be controlled be the one selected as the default ctl device in asoundrc?

From what I have seen, the ctl.!default section of .asoundrc only takes a single line:

ctl.!default {
   type bluealsa
}

Is there some way of specifying a device address in here, like:

ctl.!default {
   type bluealsa
   device "xx:xx:xx:xx:xx:xx"
}

Not sure if that is what the ctl section is for?

spl237, 2019-09-19:
I've been doing some digging about in the bluealsa-ctl.c file.

One possible solution to this would be to allow a device "xx..." line in the ctl section of .asoundrc; this could be parsed and read into a new data field added to the bluealsa_ctl structure.

At present, the bluealsa_find_elem function searches the list of ctl_elems and returns the first element which has a matching id (and as the list is presumably sorted alphabetically, this explains the behaviour above). What I think this could do is to compare the name field in the bt_dev entry of each item in the list with the address stored in the bluealsa_ctl structure, and only return an element which matches both the id and the address. (If no address is provided in asoundrc, this can continue to match on just id, as now.)

That seems to be a relatively small change, and I think it would be useful - with multiple devices, getting what is in effect a random volume control element is undesired behaviour.

I'm going to have a play with this tomorrow - I don't think it is a huge change to the code. If I get something working, I'll submit a PR if that would be useful?

spl237, 2019-09-20:
No, the approach I suggested in my previous post doesn't work - it looks as if the find_elem function matches on the element numid before it gets to comparing other information, and I can't work out how the numids are assigned, so I can't see how to filter the calls properly.

Arkq, 2019-09-21:
> In terms of detecting add and remove of devices, I've never found a way to get ALSA to tell me this - it would be extremely useful, because when a USB audio device is removed, it leaves things in a high power drain state - but I'm not convinced it is possible via ALSA itself.

By theory ALSA library should tell you if the device has been disconnected - almost every API function shall return ENODEV error, however, it is not working properly. The problem is (IMHO), that nobody is testing this feature because it's rather hard to test. When creating ALSA plug-ins for bluealsa, I've tried to follow the behavior of USB sound card, because it is possible to unplug it (just like BT device). More or less this behavior is mimicked, but in order to fully support it, some changes in the ALSA library are required... I've proposed some patches, but they have not been reviewed. So, yes, ALSA library does signal user that the device has been disconnected. The problem is that no one is checking for such error code, hence most applications go into infinite loop :D .

> If I then turn on the second BT device, it connects. I get a PCMAdded signal from the monitor, so I delete and recreate the mixer, but it should still be associated with the device selected in .asoundrc, as that hasn't changed. But the volume control sometimes no longer controls the volume on my original device. If I look in alsamixer, it appears that what has happened is that the mixer associated with bluealsa is now attached to the new device, but that audio output is still being routed to the original device. So I have a working volume control, but it is no longer controlling the output that it was, and is now in fact controlling an output which is not being used.

That is the problem of UI. With non-pluggable sound cards the case is simple - static configuration is good enough. When we talk about pluggable sound cards there is a problem how the interface should look like. With USB sound card the problem is less severe, than with BT. The USB card is most the time connected to the PC, but BT device is **VERY** mobile. You can "plug" and "unplug" headset by walking from leaving-room to the kitchen and vice versa. So the user interface should be more complex. Static configuration is not enough - IMHO. The problem with ALSA is, that it does not allow dynamic reconfiguration (it's one of reasons why PulseAudio has emerged). But the question is whether is should DO such automation or just allow it (e.g. as a software plugin).

Bluealsa is designed to mimic hardware (from the ALSA point of view). Any additional features required for user friendly interface is beyond bluealsa scope (I might be wrong, though...). So in general bluealsa should be regarded as a sound card (BT sound card) which provides PCMs. And these PCMs are *dynamic* - this concept is somehow new, so it might be a problem.

The above is my general idea about bluealsa. I'm not sure whether is has been constructed in a right way, but it is more or less what I've tried to accomplish.

> It looks as if the device whose volume control takes effect is based on alphabetical order of device names - I have two devices, "Antzz Soundbuds" and "Soundcore Spirit", and it seems that turning on "Soundcore" does not move the volume control from "Antzz", but turning "Antzz" on moves it from "Soundcore" - does that seem likely?

Yes, effectively after connecting new BT device, IDs no longer point to previous controls... Maybe that is what should be reconsidered and fixed. Because I see, that this might cause troubles.

> The automatic connection of devices to bluealsa seems to be causing problems - could the code perhaps (at least optionally) only connect bluealsa to the device selected in .asoundrc? I think that would make things a lot more robust in the environment where multiple devices are being used. (Admittedly most people will only use a single device, but multiple devices definitely causes problems in the current implementation!)

If someone is using only one BT device, the problem does not occur (if I'm correct). We should focus on multi-BT device environment. I think that editing some configuration file every time I want to switch BT device might be rather painful... It is a repetitive task, so it should be perfect for some software (like I've mentioned before, some additional ALSA plugin, which allows dynamic reconfiguration).

> No, the approach I suggested in my previous post doesn't work - it looks as if the find_elem function matches on the element numid before it gets to comparing other information, and I can't work out how the numids are assigned, so I can't see how to filter the calls properly.

Anyway, implementing such "static" configuration should be possible, maybe with a different approach, which also should be rather simple. The question is whether it is a right way of fixing the problem this issue is about.

spl237, 2019-09-21:
> If someone is using only one BT device, the problem does not occur (if I'm correct).

Yes, if only one device is paired, the problem cannot occur, and even if multiple devices are paired as long as only one connects, the problem does not occur.

> I think that editing some configuration file every time I want to switch BT device might be rather painful...

Agreed - but then that is how ALSA works; you either have to provide devices on command lines or put them in .asoundrc. In Raspbian, I have written a plugin for the main taskbar which pops up a menu of ALSA devices and allows you to click on one; this then makes the relevant updates to .asoundrc. I do understand your point about the more dynamic nature of BT devices, and I can see that this does make this more of an issue.

> Anyway, implementing such "static" configuration should be possible, maybe with a different approach, which also should be rather simple.

I think all that is needed is the ability to (optionally) define a device in the ctl.!default section of .asoundrc if type bluealsa is in there. If this line is found, then the only controls which are offered by bluealsa should be those associated with the device at that address; if the line is not in there, then bluealsa can continue the dynamic behaviour as at present. The problem at present is that you can define a static pcm device for bluealsa, but not a matching ctl device; as long as there is a mechanism for a static ctl device, I think (when combined with my monitoring of the PCMAdded and PCMRemoved events) that fixes 99% of the problems I am seeing.

I did spend a day hacking about with the code in bluealsa-ctl.c to see if I could find a way to make a static ctl address work, but I didn't manage to get anything working; I really don't know enough about ALSA!

Many thanks for all your work on this, btw - it is very much appreciated; Raspbian is so much better for the existence of your plugin that it seems wrong to post issues!

borine, 2019-11-16:
I came across this thread while looking into another issue. Please forgive me if I have misunderstood the  requirement.

If your application is using the libasound interface to interact with bluealsa, then I think you should be able to achieve your requirements (mostly) without D-Bus,

First, I would not use USB devices as a close analogy for bluealsa, especially where ctl elements are concerned. For the ctl interface, the bluealsa daemon is the analog of the USB adapter - when you obtain an alsa snd_mixer_t* handle on blealsa you effectively have a handle on the daemon, *not* on an individual BT device. So connecting/disconnecting BT devices is not at all like plugging unplugging USB adapters as far as the mixer is concerned.

> this creates an event within ALSA / Bluealsa which corrupts the mixer; the mixer type changes from 0 to a random number, and the mixer is invalid thereafter

I think that is wrong. The "mixer" (ie snd_mixer_t*) remains valid - it is just the "control" (snd_mixer_elem_t*) which becomes invalid.

> so how can I find out that the mixer is no longer valid and create a new one? Is there some event at the ALSA level that I can see?

In your elem callback function, you need to check for the SND_CTL_EVENT_MASK_REMOVE event and use that as notification that the *element* (control) is now invalid aand must no longer be de-rerefenced.

You also need a mixer callback function that uses the SND_CTL_EVENT_MASK_ADD event to get the updated element handle when bluealsa has re-assigned it. See
https://github.com/alsa-project/alsa-utils/blob/c0591c540d44f3e00c2092742e09de0891ad4703/alsamixer/mixer_widget.c#L61
and
https://github.com/alsa-project/alsa-utils/blob/c0591c540d44f3e00c2092742e09de0891ad4703/alsamixer/mixer_widget.c#L76
for how alsamixer does this (you will additionally have to check that the passed elem is the correct one for your default pcm device)

The case that is not handled gracefully is if the bluealsa daemon is terminated while your application is running (that is the issue I am looking at atm). The alsa ctl functions do not return ENODEV, so it looks as if D-Bus signals may be the only option there; I'm not sure yet, more reading and testing is required.

borine, 2019-11-17:
An update on handling bluealsa daemon termination:
It seems that alsamixer makes a call to snd_ctl_get_power_state() which does return ENODEV if a hardware card has been unplugged. Unfortunately in the alsa ctl external plugin API, that call always returns the value 0, and there is no way for the plugin to override that. So basically there is no consistent way for the bluealsa ctl plugin to inform the application that the daemon has stopped - it is impossible for bluealsa to repiicate the behaviour of hardware devices here.

borine, 2021-06-24:
I'm closing this because the basic assumption underlying this analysis is that it is possible to use .asoundrc to associate a specific control element with a specific PCM device. That assumption is false. The ALSA config API does not support any such association. The nearest you can get is the "softvol" plugin, but that is totally unworkable for temporary, removable PCM devices such as bluetooth devices. The reality is that a static configuration as used by ALSA is incapable of expressing fully the dynamic nature of bluetooth device connections. For desktop environments a dynamic layer above ALSA is required to achieve the kind of flexiblity  desired here; and I believe that the Raspberry Pi team have already accepted this by adopting PulseAudio for their desktop earlier this year. BlueALSA sits *behind* the ALSA API - use it in environments where applications need direct access to the ALSA layer.

closed 2021-06-24 by borine