arkq/bluez-alsa #250: buffer overflow

digitxo, 2019-09-21:
alsa_out -d bluealsa:HCI=hci0,DEV=XX:XX:XX:XX:XX:XX,PROFILE=a2dp

using the above commadn i get the meessage

*** buffer overflow detected ***: alsa_out terminated

help! I'm trying to set my output to a bluetooth device, and my input to a line-in soundcard USB

bquandt, 2020-03-19:
I did get this to work.   My notes are still disordered and barely understandable, but thought I'd post for someone else struggling (hey clean up my notes, much appreciate it)

step 2) get and compile sourcebluealasa 495  ../configure --enable-aac  --enable-ofono --enable-msbc
  496  ../configure --enable-aac  
  498  ../configure --enable-aac  --enable-debug
tried 495, no go, saw SBC as choices guessed that was the issue, scrapped the two profiles opted only aacafter install, sudo vi /etc/dbux-1/system.d/bluealsa.conf  (copy own line from root to audio groupadd my user to audio grouprun bluealsa as user (might work as root, but tried first, ended up here)step 

3) start qjackctl as usual (mine defaults to 44100)
step 

4) start up our jack output itemalsa_out -j blueout -d bluealsa -r 44100 -p 10000
alsa_in -j cloop -d cloopalsa_out -j ploop -d ploop

step 1) Added loop back items so I can grab others and throw them later to a jack in/out.   alos added bluetooth items (see attached)

step5) run bluealsa-aplay <bluetoothaddr> so we have an input, do same for it against alsa_in when I figure that out.... this is TBD, but pretty sure I can make things work now I got a gist of things working on playback (yup got sound, yup from chrome youtube... etc into qjackctl and playing with routing, now back to having fun!

[asoundrc.txt](https://github.com/Arkq/bluez-alsa/files/4353334/asoundrc.txt)

borine, 2020-04-01:
This reply is probably 6 months too late, but anyway ...

I've been playing around with Jack for the last couple of days, and I've found a couple of bugs in the alsa_out utility that relate to this issue. 

The *** buffer overflow *** abort is caused by the undocumented fact that alsa_out copies the given alsa device name (from the -d argument) into a buffer that can hold only 19 characters. Since bluealsa device names are typically much longer than that, this buffer overflows and BOOM!

You can work around this by putting an alias in your ~/.asoundrc - for example:
```
pcm.mybtdevice "bluealsa:DEV=XX:XX:XX:XX:XX:XX,PROFILE=a2dp"
```

Next you will find that there is a SEGV failure with the default period size, and constant xruns for period sizes <= 4096. So I use `-p 4097` as a working value that gives lowest latency.

So now if you try:
```
alsa_out -d  mybtdevice -p 4097
```

hopefully you will hear music :)

capoei, 2020-07-18:
I can only alsa_out bluealsa with "-p 10000"
the output messages are very strange tough.
how can I have about the same delay between the jack sound card and the bluealsa device?

Arkq, 2021-01-16:
Not bluez-alsa issue. Closing.

PS.
Thanks to @borine for finding the cause and resolving it!

closed 2021-01-16 by Arkq