arkq/bluez-alsa #254: What is --enable-aac use case

roizcorp, 2019-10-04:
Hi,

I'm trying to understand what would be useful for by using aac?
(sorry if the answer is obvious)

Arkq, 2019-10-04:
Better audio quality with iPhone :) and headsets dedicated for iPhone (with
AAC support but not aptx nor ldac).

closed 2019-10-04 by Arkq

roizcorp, 2019-10-04:
so why it is mentioned in the default build guide? (I'm using android)
any other"--enable-XXX" option you would recommend?

Arkq, 2019-10-04:
Because it's the only extra codec which works out of the box with bluealsa and provides encoder and decoder. It works with iPhone and some android phones so it covers almost all cases for sink. Also, it was the first codec available with this project, so it is explicitly mentioned in the build section.

roizcorp, 2019-10-04:
cool, I already built fdk-aac for it...so I will include it in, sounds great BTW
can it make the phone as the speaker? (considering the other side is the source for sound?)