arkq/bluez-alsa #256: error: ALSA lib ../../../src/asound/bluealsa-pcm.c:718:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: PCM not found

sherifomran, 2019-10-05:
Hello Every body,

I have 2 devices being connected
```

pi@raspberrypi:~ $ bluetoothctl devices 
Device FC:58:FA:7C:33:42 SBL 3 C3
Device 00:0B:D5:F5:F8:B9 JABRA DRIVE

pi@raspberrypi:~ $ bluetoothctl info 00:0B:D5:F5:F8:B9
Device 00:0B:D5:F5:F8:B9 (public)
	Name: JABRA DRIVE
	Alias: JABRA DRIVE
	Class: 0x00240404
	Icon: audio-card
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)

pi@raspberrypi:~ $ bluetoothctl info FC:58:FA:7C:33:42
Device FC:58:FA:7C:33:42 (public)
	Name: SBL 3 C3
	Alias: SBL 3 C3
	Class: 0x00260404
	Icon: audio-card
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)

Now playing on one device, it works fine

pi@raspberrypi:~ $ aplay -D bluealsa:DEV=FC:58:FA:7C:33:42,PROFILE=a2dp file_example_WAV_1MG.wav 
Playing WAVE 'file_example_WAV_1MG.wav' : Signed 16 bit Little Endian, Rate 8000 Hz, Stereo

playing on the second device produces not found pcm

pi@raspberrypi:~ $ aplay -D bluealsa:DEV=00:0B:D5:F5:F8:B9,PROFILE=a2dp file_example_WAV_1MG.wav 
ALSA lib ../../../src/asound/bluealsa-pcm.c:718:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: PCM not found
aplay: main:828: audio open error: No such device
```

borine, 2019-10-05:
The Jabra Drive does not show support for A2DP; the line:
`UUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)`
does not appear in the bluetoothctl info output.
So there is no A2DP connection to the device, which is what the bluealsa error message is telling you.
You could try 
`aplay -D bluealsa:DEV=00:0B:D5:F5:F8:B9,PROFILE=sco ...`
But even if that works, you would only get voice-quality audio.

sherifomran, 2019-10-05:
solution found using profile sco

closed 2019-10-05 by sherifomran