arkq/bluez-alsa #258: No sound when playing via SCO

killerkalamari, 2019-10-09:
When I play via SCO and hsp-ag on my headset I hear a pop, but no actual audio playback. Best I can tell, I'm not getting an error either.

How I'm playing: `aplay -v -D bluealsa:DEV=E9:08:EF:2D:40:33,PROFILE=sco test_8k.wav`, and output:
```
bluealsa-pcm.c:589: /org/bluealsa/hci0/dev_E9_08_EF_2D_40_33/sco: Setting constraints
Playing WAVE 'test_8k.wav' : Signed 16 bit Little Endian, Rate 8000 Hz, Mono
bluealsa-pcm.c:319: /org/bluealsa/hci0/dev_E9_08_EF_2D_40_33/sco: Initializing HW
bluealsa-pcm.c:346: /org/bluealsa/hci0/dev_E9_08_EF_2D_40_33/sco: FIFO buffer size: 4096
bluealsa-pcm.c:352: /org/bluealsa/hci0/dev_E9_08_EF_2D_40_33/sco: Selected HW buffer: 4 periods x 2000 bytes == 8000 bytes
bluealsa-pcm.c:367: /org/bluealsa/hci0/dev_E9_08_EF_2D_40_33/sco: Initializing SW
bluealsa-pcm.c:367: /org/bluealsa/hci0/dev_E9_08_EF_2D_40_33/sco: Initializing SW
bluealsa-pcm.c:389: /org/bluealsa/hci0/dev_E9_08_EF_2D_40_33/sco: Prepared
bluealsa-pcm.c:367: /org/bluealsa/hci0/dev_E9_08_EF_2D_40_33/sco: Initializing SW
Plug PCM: BlueALSA PCM: /org/bluealsa/hci0/dev_E9_08_EF_2D_40_33/sco
BlueALSA BlueZ device: /org/bluez/hci0/dev_E9_08_EF_2D_40_33
BlueALSA Bluetooth codec: 0x1
bluealsa-pcm.c:240: /org/bluealsa/hci0/dev_E9_08_EF_2D_40_33/sco: Starting
bluealsa-pcm.c:134: /org/bluealsa/hci0/dev_E9_08_EF_2D_40_33/sco: Starting IO loop: 8
bluealsa-pcm.c:280: /org/bluealsa/hci0/dev_E9_08_EF_2D_40_33/sco: Stopping
bluealsa-pcm.c:104: /org/bluealsa/hci0/dev_E9_08_EF_2D_40_33/sco: IO thread cleanup
bluealsa-pcm.c:280: /org/bluealsa/hci0/dev_E9_08_EF_2D_40_33/sco: Stopping
bluealsa-pcm.c:359: /org/bluealsa/hci0/dev_E9_08_EF_2D_40_33/sco: Freeing HW
bluealsa-pcm.c:309: /org/bluealsa/hci0/dev_E9_08_EF_2D_40_33/sco: Closing
```

How I'm running bluealsa: `sudo /usr/bin/bluealsa -p hsp-ag`, and initial output:
```
/usr/bin/bluealsa: ba-adapter.c:141: Freeing adapter: hci0
/usr/bin/bluealsa: bluez.c:1068: Creating hands-free profile object: /org/bluez/HSP/AudioGateway
/usr/bin/bluealsa: bluez.c:1098: Registering hands-free profile: /org/bluez/HSP/AudioGateway
/usr/bin/bluealsa: main.c:353: Acquiring D-Bus service name: org.bluealsa
/usr/bin/bluealsa: main.c:358: Starting main dispatching loop
```

Output when BT headset is connected:
```
/usr/bin/bluealsa: bluez.c:1036: Called: org.bluez.Profile1.NewConnection()
/usr/bin/bluealsa: bluez.c:960: RFCOMM: HSP Audio Gateway configured for device E9:08:EF:2D:40:33
/usr/bin/bluealsa: ba-transport.c:601: State transition: 0 -> 2
/usr/bin/bluealsa: io.c:2576: Created new IO thread: RFCOMM: HSP Audio Gateway
/usr/bin/bluealsa: ba-transport.c:601: State transition: 0 -> 2
/usr/bin/bluealsa: rfcomm.c:597: Starting loop: RFCOMM: HSP Audio Gateway
/usr/bin/bluealsa: io.c:2576: Created new IO thread: HSP Audio Gateway
/usr/bin/bluealsa: io.c:2074: Starting IO loop: HSP Audio Gateway
/usr/bin/bluealsa: at.c:161: AT message: SET: command:+VGS, value:15
/usr/bin/bluealsa: rfcomm.c:105: Sending AT message: RESP: command:(null), value:OK
/usr/bin/bluealsa: at.c:161: AT message: SET: command:+CKPD, value:200
/usr/bin/bluealsa: Unsupported AT message: SET: command:+CKPD, value:200
/usr/bin/bluealsa: rfcomm.c:105: Sending AT message: RESP: command:(null), value:ERROR
/usr/bin/bluealsa: at.c:161: AT message: SET: command:+XAPL, value:0000-0000-0100,7
/usr/bin/bluealsa: rfcomm.c:105: Sending AT message: RESP: command:(null), value:+XAPL=BlueALSA,0
/usr/bin/bluealsa: at.c:161: AT message: SET: command:+IPHONEACCEV, value:2,1,8,2,0
/usr/bin/bluealsa: rfcomm.c:105: Sending AT message: RESP: command:(null), value:OK
```

Output during playback:
```
/usr/bin/bluealsa: bluealsa-dbus.c:134: Called: org.bluealsa.Manager1.GetPCMs()
/usr/bin/bluealsa: bluealsa-dbus.c:328: Called: org.bluealsa.PCM1.Open()
/usr/bin/bluealsa: ba-transport.c:601: State transition: 2 -> 2
/usr/bin/bluealsa: ba-transport.c:848: New SCO link: 17 (MTU: R:48 W:48)
/usr/bin/bluealsa: ba-transport.c:678: PCM drained
/usr/bin/bluealsa: ba-transport.c:882: Closing PCM: 8
/usr/bin/bluealsa: io.c:126: PCM read buffer flushed: -1
/usr/bin/bluealsa: io.c:126: PCM read buffer flushed: -1
/usr/bin/bluealsa: ba-transport.c:858: Closing SCO: 17
```
The only configure arg that I used is --enable-debug, as I read in other tickets that mSBC might not be working properly for audio output yet. Also, I think I read that mSBC only applies to HFP?

This is on aarch64. It's an embedded platform, but not a Raspberry Pi. Pulseaudio is not installed. I hear the playback fine if I don't restrict to hsp-ag profile and if I play to A2DP, so that works. If I don't restrict to hsp-ag profile and play to SCO, I again hear nothing.

Due to the lack of errors, I'm assuming more information is needed in order to troubleshoot this. Do you have any recommendations on which source files I should be looking to add debugging code to?

borine, 2019-10-10:
It is possible that SCO traffic is not being accepted on the HCI link between the host and controller. I know this has been an issue with the raspberry pi, so I guess similar issues might affect other boards too. For example, with the broadcom chip used on the rpi, SCO traffic is by default routed through the "PCM" pins, not the UART pins used by HCI. So the host may be sending SCO data through the HCI (so `btmon --sco` shows the data packets going out from the host), but the controller is not picking them up to pass to the radio stack. For the rpi, this can be remedied by sending a vendor custom HCI command to switch SCO routing to HCI. To find if any similar kind of issue is affecting your board you would need to do a web search for SCO issues with the bluetooth chip model number

killerkalamari, 2019-10-10:
Thank you so much for mentioning btmon --sco. Here's the output from that.

Before connecting:
```
Bluetooth monitor ver 5.50
= New Index: 70:2C:1F:32:15:85 (Primary,UART,hci0)                                                                                                                                                 [hci0] 0.284101
= Open Index: 70:2C:1F:32:15:85                                                                                                                                                                    [hci0] 0.284104
= Index Info: 70:2C:1F:32:15:85 (Broadcom Corporation) 
```

Connection:
```
< HCI Command: Create Connection (0x01|0x0005) plen 13                                                                                                                                        #1 [hci0] 162.904171
        Address: E9:08:EF:2D:40:33 (OUI E9-08-EF)
        Packet type: 0xcc18
          DM1 may be used
          DH1 may be used
          DM3 may be used
          DH3 may be used
          DM5 may be used
          DH5 may be used
        Page scan repetition mode: R2 (0x02)
        Page scan mode: Mandatory (0x00)
        Clock offset: 0x0000
        Role switch: Allow slave (0x01)
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                                     #2 [hci0] 162.904873
      Create Connection (0x01|0x0005) ncmd 1
        Status: Success (0x00)
> HCI Event: Connect Complete (0x03) plen 11                                                                                                                                                  #3 [hci0] 165.998347
        Status: Success (0x00)
        Handle: 12
        Address: E9:08:EF:2D:40:33 (OUI E9-08-EF)
        Link type: ACL (0x01)
        Encryption: Disabled (0x00)
< HCI Command: Read Remote Supported Features (0x01|0x001b) plen 2                                                                                                                            #4 [hci0] 165.998696
        Handle: 12
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                                     #5 [hci0] 165.999598
      Read Remote Supported Features (0x01|0x001b) ncmd 1
        Status: Success (0x00)
> HCI Event: Read Remote Supported Features (0x0b) plen 11                                                                                                                                    #6 [hci0] 165.999774
        Status: Success (0x00)
        Handle: 12
        Features: 0xff 0xff 0x8f 0xfe 0x9b 0xff 0x59 0x87
          3 slot packets
          5 slot packets
          Encryption
          Slot offset
          Timing accuracy
          Role switch
          Hold mode
          Sniff mode
          Park state
          Power control requests
          Channel quality driven data rate (CQDDR)
          SCO link
          HV2 packets
          HV3 packets
          u-law log synchronous data
          A-law log synchronous data
          CVSD synchronous data
          Paging parameter negotiation
          Power control
          Transparent synchronous data
          Broadcast Encryption
          Enhanced Data Rate ACL 2 Mbps mode
          Enhanced Data Rate ACL 3 Mbps mode
          Enhanced inquiry scan
          Interlaced inquiry scan
          Interlaced page scan
          RSSI with inquiry results
[... 948 lines cut ...]
        Source CID: 321
< ACL Data TX: Handle 12 flags 0x00 dlen 12                                                                                                                                                 #811 [hci0] 208.793837
      L2CAP: Disconnection Response (0x07) ident 3 len 4
        Destination CID: 65
        Source CID: 321
> ACL Data RX: Handle 12 flags 0x02 dlen 12                                                                                                                                                 #812 [hci0] 208.794995
      L2CAP: Disconnection Response (0x07) ident 8 len 4
        Destination CID: 321
        Source CID: 65
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                                                      #813 [hci0] 208.972450
        Num handles: 1
        Handle: 12
        Count: 1
< HCI Command: Disconnect (0x01|0x0006) plen 3                                                                                                                                              #814 [hci0] 210.791597
        Handle: 12
        Reason: Remote User Terminated Connection (0x13)
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                                   #815 [hci0] 210.792226
      Disconnect (0x01|0x0006) ncmd 1
        Status: Success (0x00)
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                                                              #816 [hci0] 210.943570
        Status: Success (0x00)
        Handle: 12
        Reason: Connection Terminated By Local Host (0x16)
@ Device Disconnected: E9:08:EF:2D:40:33 (0) reason 2
```

borine, 2019-10-11:
That trace appears to show that bluealsa is correctly sending SCO data to the HCI. Can't tell if that data is "music" or just a "pop", but if there are lots of SCO data packets, and the content is not all zeros, then I think it's reasonable to assume bluealsa is doing its job correctly here.

There may be many reasons why the headset does not render the sound, but that's possibly out of scope here, and certainly beyond my knowledge.

You should check that the controller is actually transmitting the data it is receiving on the HCI [SCO and A2DP are handled very differently by the controller hardware] - tcpdump is your friend there  - and beyond that you might seek advice for your specific board/controller/headset.

killerkalamari, 2019-10-11:
I trimmed the audio. There are some zeros at the beginning, but most looks like audio. Thanks for your help. I'll look into a controller problem.

killerkalamari, 2019-10-14:
After some digging, I found out we are using the CYW43455 (also called BCM43455) chip, which is the same as is being used in the Pi 3 Model B+ and Pi 4 (confusingly, the original Pi 3 model used a CYW43438). Found the fix you were referring to, which is for the CYW43438 (or BCM43438). Tried their fix even though it was the wrong chipset, and it actually worked to enable audio on the CYW43455 in both hsp-ag and hfp-ag modes, although I am getting popping and audio truncation (might be a separate issue at this point). 

Someone in this thread has also already asked Cygnus for CYW43455 help with the Pi 4, as I guess the workaround isn't helping there: https://github.com/raspberrypi/linux/issues/2229#issuecomment-359233111

This ticket is probably resolved, but I'd like to try out the fix on a Pi 3 (original), and a Pi 4, and see how things work there with bluealsa under Raspbian (vs this embedded chipset and OpenWRT installation) and see if there is truncation.

borine, 2019-10-15:
You should also try the mtu fix for bluealsa, to see if that helps with the sound quality. See #255, specifically my comment here: https://github.com/Arkq/bluez-alsa/issues/225#issuecomment-522459636.
The line numbers to change are now 866 and 867 in src/ba-transport.c in the latest code. You could also try an mtu of 60 instead of 64 to see which, if any, gives best results.

I have spent a lot of time and effort researching this MTU issue with linux bluetooth, but have made little progress in finding a generic solution. However, specifically with bluealsa with the BCM43455, hardcoding the MTU to 64 works for me (but breaks if i then use a USB bt adapter).

killerkalamari, 2019-10-17:
Was able to recreate the truncated audio on Pi 3 with the hcitool cmd fix and bluez-alsa. So, I decided to try the fix with pulseaudio on the Pi 3. Well, with pulse, the first part of the sound was missing, and the audio quality was atrocious. If bluealsa used any of that code there has obviously been a lot of repair work done. I'll go ahead and create a new bug for the truncation issue.

killerkalamari, 2019-10-17:
Fix for the no sound via SCO issue on the CYW43438 and CYW43455 can be found here (thanks to borine for the tip!): https://github.com/raspberrypi/linux/issues/2229#issuecomment-359233111

Sample run (the output was identical for both chips):
```
$ sudo hcitool cmd 0x3F 0x01C 0x01 0x02 0x00 0x01 0x01
< HCI Command: ogf 0x3f, ocf 0x001c, plen 5
  01 02 00 01 01 
> HCI Event: 0x0e plen 4
  01 1C FC 00 
```

Closing.

closed 2019-10-17 by killerkalamari

killerkalamari, 2019-10-24:
The hcitool call from https://github.com/Arkq/bluez-alsa/issues/258#issuecomment-543280391 is no longer needed since commit 335f48c18a6dc87c4046f34591a14bd6abf65235. Thanks!