arkq/bluez-alsa #261: play failure at some times with rpi

sherifomran, 2019-10-20:
I installed the bluealsa and it was working fine but at some time i get the following
```

pi@raspberrypi:~ $ mpg123 --verbose -o bluealsa:DEV=00:0B:D5:F5:F8:B9,PROFILE=sco -f 200 ~/AudioMan/sounds/Azan.mp3
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.25.10; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes
Decoder: generic
Trying output module: bluealsa, device: DEV=00:0B:D5:F5:F8:B9,PROFILE=sco
[src/libout123/module.c:133] error: Failed to open module bluealsa.
[src/libout123/libout123.c:455] error: Found no driver out of [bluealsa] working with device DEV=00:0B:D5:F5:F8:B9,PROFILE=sco.
main: [src/mpg123.c:309] error: out123 error 3: failure loading driver module

```

borine, 2019-10-21:
I think the mpg123 command line is wrong; try:
```
mpg123 -o alsa -a bluealsa:DEV=00:0B:D5:F5:F8:B9,PROFILE=sco -f 200 ~/AudioMan/sounds/Azan.mp3
```

sherifomran, 2019-10-21:
It is not a mistake in the command because it used to work fine, but at some time, it can not open the socket connection

`pi@raspberrypi:~ $ mpg123 -o bluealsa:DEV=00:0B:D5:F5:F8:B9,PROFILE=sco -f 200 ~/AudioMan/sounds/Azan.mp3
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.25.10; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes
[src/libout123/module.c:133] error: Failed to open module bluealsa.
[src/libout123/libout123.c:455] error: Found no driver out of [bluealsa] working with device DEV=00:0B:D5:F5:F8:B9,PROFILE=sco.
main: [src/mpg123.c:309] error: out123 error 3: failure loading driver module
pi@raspberrypi:~ $ bluetoothctl
Agent registered
[SBL 3 C3]# info
Device FC:58:FA:7C:33:42 (public)
	Name: SBL 3 C3
	Alias: SBL 3 C3
	Class: 0x00260404
	Icon: audio-card
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
[SBL 3 C3]# devices
Device FC:58:FA:7C:33:42 SBL 3 C3
Device 00:0B:D5:F5:F8:B9 JABRA DRIVE
[SBL 3 C3]# info 00:0B:D5:F5:F8:B9 
Device 00:0B:D5:F5:F8:B9 (public)
	Name: JABRA DRIVE
	Alias: JABRA DRIVE
	Class: 0x00240404
	Icon: audio-card
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
[SBL 3 C3]# exit
pi@raspberrypi:~ $ mpg123 -o alsa bluealsa:DEV=00:0B:D5:F5:F8:B9,PROFILE=sco -f 200 ~/AudioMan/sounds/Azan.mp3
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.25.10; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes
ALSA lib ../../../src/asound/bluealsa-pcm.c:682:(_snd_pcm_bluealsa_open) Unknown field interface
[src/libout123/modules/alsa.c:181] error: cannot open device default
[src/libout123/libout123.c:455] error: Found no driver out of [alsa] working with device <default>.
main: [src/mpg123.c:309] error: out123 error 3: failure loading driver module
pi@raspberrypi:~ $ mpg123 -o alsa bluealsa:DEV=00:0B:D5:F5:F8:B9,PROFILE=a2dp -f 200 ~/AudioMan/sounds/Azan.mp3
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.25.10; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes
ALSA lib ../../../src/asound/bluealsa-pcm.c:682:(_snd_pcm_bluealsa_open) Unknown field interface
[src/libout123/modules/alsa.c:181] error: cannot open device default
[src/libout123/libout123.c:455] error: Found no driver out of [alsa] working with device <default>.
main: [src/mpg123.c:309] error: out123 error 3: failure loading driver module
pi@raspberrypi:~ $ 
`

sherifomran, 2019-10-21:
src/libout123/module.c:133 I get failed to open module bluealsa
src/libout123/libout123.c:455 error: found no driver out of [bluealsa] working with device bluealsa:DEV=00:0B:D5:F5:F8:B9,PROFILE=sco
main: [src/mpg123.c:309] error: out123 error 3 failure loading driver module

are we using pulse audio?
how can i play mp3 using bluealsa, the only method i know is mpg123 with the device ID, otherwise i need to adjust jack or pulse audio, please correct me if i am mistaken, whereas aplay does not play mp3 files but wave files only.

borine, 2019-10-21:
with mpg123 you need to specify the device with ` -a ` option (or with environment variable).
the ` -o ` option specifies a mpg123 module to use.

`mpg123 -o bluealsa:DEV=...` will always fail because mpg123 does not have a module called bluealsa.

`mp123 -o alsa bluealsa:DEV=...` fails because it tries to use the alsa "default" device to play the file "bluealsa:DEV=...".

`mpg123 -o alsa -a bluealsa:DEV=...` should at least have a chance of working 

`man mpg123` explains all this.

sherifomran, 2019-11-25:
i have another problem, when i use mpg123 with any syntax, the terminal freezes up and i don't hear any sound, but when i use aplay it works only with wav files. 
is there a way to play mp3 using aplay ? any body has an idea where the problem could be?
it was working fine, may be i installed a wrong packet for mpg123

borine, 2019-11-26:
> i have another problem, when i use mpg123 with any syntax, the terminal freezes up and i don't hear any sound

if mpg123 locks up the terminal, that might be the libasound thread locking bug (see "2. ALSA thread-safe API (alsa-lib >= 1.1.2)" on the bluealsa home page https://github.com/Arkq/bluez-alsa.
Try:
```
LIBASOUND_THREAD_SAFE=0 mpg123 -o alsa -a bluealsa:DEV=00:0B:D5:F5:F8:B9,PROFILE=sco ~/AudioMan/sounds/Azan.mp3
```

> is there a way to play mp3 using aplay ?

aplay cannot play mp3 files.

borine, 2021-06-04:
mpg123 is working well with bluealsa for over a year now, so hopefully the issue is now resolved. I am closing this issue as there has been no activity for some. Please feel free to re-open it if the problem persists with latest bluealsa code.

closed 2021-06-04 by borine