arkq/bluez-alsa #262: Embedded Linux - Playback working but not Recording (Yocto - cross-compiled bluez-alsa v.2.0.0)

daniele86rm, 2019-10-24:
On my customized embedded device, based on Cortex A9 processor, I have the same problem of other users.
After paired and connected bluetooth audio device with "bluetoothctl" utility, audio playback works properly but recording no. When I run the following command to capture audio from bluetooth device:
`arecord -D bluealsa:DEV=00:18:09:8A:09:35,PROFILE=sco capture.wav`
i obtain an empty .wav file with 44 bytes size.
I have tried to start bluealsa with different profiles, as described in others posts:
1) bluealsa -p a2dp-source -p a2dp-sink -p hfp-hf -p hfp-ag -p hsp-hs -p hsp-ag &
2) bluealsa -p hfp-ag -p a2dp-source &
but I have always the same problem.

I'm using latest release of bluez-alsa, version 2.0.0 released 9 days ago, cross-compiled with the following Yocto recipe:

```
SUMMARY = "Bluetooth Audio ALSA Backend"
HOMEPAGE = "https://github.com/Arkq/bluez-alsa"
SECTION = "devel"

LICENSE = "MIT"
LIC_FILES_CHKSUM = "file://LICENSE.txt;md5=3d7d6ac7e2dbd2505652dceb3acdf1fe"

DEPENDS = "alsa-lib bluez5 glib-2.0 sbc"

SRCREV = "1c86e6dc2709f028a2f0a9eb19942b93bfc55e62"
SRC_URI = "git://github.com/Arkq/bluez-alsa.git;branch=master;protocol=https \
           file://bluez-alsa.service"

FILESEXTRAPATHS_append := "${THISDIR}/files:"
S = "${WORKDIR}/git"
inherit systemd pkgconfig autotools

do_install () {
    autotools_do_install
    install -d ${D}${systemd_unitdir}/system
    install -m 0644 ${WORKDIR}/bluez-alsa.service ${D}${systemd_unitdir}/system
}

FILES_${PN} += "${libdir}/alsa-lib/lib*.so ${datadir}/alsa"
FILES_${PN}-dev += "${libdir}/alsa-lib/*.la"
FILES_${PN}-staticdev += "${libdir}/alsa-lib/lib*.a"
FILES_${PN}-dbg += "${libdir}/alsa-lib/.debug/*.so"
SYSTEMD_SERVICE_${PN} = "bluez-alsa.service"
```

My bluetooth module, integrated on device motherboard, is "Sterling-LWB Module - 450-0159 - Base SiP Module".

Any suggestions? Thanks in advance.

Best Regards
Daniele

borine, 2019-10-24:
I note that your bluetooth adapter 450-0159 uses the Cypress/Broadcom  CYW4343W / BCM4343W BT/WLAN chip. Looking at the datasheet for that chip here:
https://www.cypress.com/file/298081/download
I see that it routes SCO audio traffic to its PCM interface, not the UART HCI interface, in the same way as the cypress chips used in the raspberry pi. For bluealsa to see any SCO/eSCO traffic it must be routed through the HCI (UART) interface. For the RPi chips, Cypress released a custom vendor HCI command that re-routes that traffic - maybe the same or a similar command is available for the CYW4343W. If not, then I suspect you are out of luck with bluealsa and pulseaudio.
See #258 for a discussion with links about this issue on another embedded board.

daniele86rm, 2019-10-24:
Thanks for quick response. I have launched `hcitool cmd 0x3F 0x01C 0x01 0x02 0x00 0x01 0x01` and now recording working properly, with low quality. I will try to change mtu from 64 to 60, as describe in others posts.

Arkq, 2019-10-24:
Since it seems that a lot of ppl has this issue, I've upstreamed the fix for Broadcom chips. @daniele86rm could you compile master branch (not tag v2.0.0) and verify whether the latest commit fixes the issue for you?

@borine many thanks for actively explaining why SCO is not working properly with Broadcom :)

daniele86rm, 2019-10-24:
Many thanks. Also the low quality problem is fixed or I should modify mtu value? Tomorrow I will test the last commit.

Arkq, 2019-10-24:
> Also the low quality problem is fixed or I should modify mtu value?

No, this commit does not change any MTU value. You have to experiment by yourself. It only tries to setup SCO routing for Broadcom chips.

daniele86rm, 2019-10-25:
Hi Arkq, I have tested bluez-alsa version under master branch and microphone working fine, now audio is caputured from bluetooth device, many thanks.
I don't known why there is always  noise on recordings, I don't known if there is an hardware problem or alsa configuration problem (for example audio file format), any suggestions? You can listen for example https://www.dropbox.com/s/5owa5uak8ywqdux/testBtMicrophone.wav?dl=0 recording, started with `arecord -D bluealsa:DEV=00:18:09:8A:09:35,PROFILE=sco testBtMicrophone.wav` command.

daniele86rm, 2019-10-25:
There is also a loss of samples https://www.dropbox.com/s/e8dozmhp2p6l663/2KHz_MicBt.wav?dl=0.

daniele86rm, 2019-10-29:
Any suggestion on how to remove noise/distortion when there is silence?

Arkq, 2019-10-29:
The key question is, who is producing such noise. If that's from the mic itself, then you out of luck. Only thing I can think of is software noise cancellation. As for missing samples, you might try changing MTU value as @borine has explained in #258

closed 2019-11-10 by daniele86rm