arkq/bluez-alsa #264: HFP plays silence, bluez-alsa to bluez-alsa

killerkalamari, 2019-11-05:
I've only successfully played HFP audio through a TaoTronics Bluetooth to 3.5mm adapter. Other devices (sound bar, multiple vehicles, etc), the audio looks like it's playing, no errors, etc, but no sound is heard. I'm confident that the SCO routing issues are no longer a factor for this chipset, because it worked with the TaoTronics even with "bluealsa -p hfp-ag", so other profiles should have been disabled. Because I wasn't sure if I was just unlucky with my playback devices, I decided to connect bluez-alsa on my embedded device to bluez-alsa on a Raspberry Pi 3. First, I got a2dp audio working and it played on the remote device. However, SCO audio did not play. I forced connection via HFP profile, and audio still did not play.

Note that I am not using ofono nor phonesim, and bluealsa is compiled without --enable-ofono.

Due to lack of error messages, I'm having trouble investigating this issue. I'm happy to provide more information or do additional troubleshooting or even code changes.

Arkq, 2019-11-05:
Are you using the latest master? Because it might work ONLY with latest commits. I haven't tested such scenario, though.

killerkalamari, 2019-11-05:
No, I missed those! Thanks! I'll give it a try.

killerkalamari, 2019-11-05:
On master I can still play to the TaoTronics, but nothing else. One change in behavior: Playing SCO from bluealsa to bluealsa doesn't play, but now aplay also no longer terminates. Here's what I see on both ends (added some blank lines to separate setup/connect/"playback"):

HFP-AG side:
```
$ sudo bluealsa -p hfp-ag
bluealsa: ba-adapter.c:151: Freeing adapter: hci0
bluealsa: bluez.c:1105: Creating hands-free profile object: /org/bluez/HFP/AudioGateway
bluealsa: bluez.c:1135: Registering hands-free profile: /org/bluez/HFP/AudioGateway
bluealsa: main.c:353: Acquiring D-Bus service name: org.bluealsa
bluealsa: main.c:358: Starting main dispatching loop

bluealsa: bluez.c:1073: Called: org.bluez.Profile1.NewConnection()
bluealsa: sco.c:154: Created SCO dispatcher [ba-sco-dispatch]: hci0
bluealsa: sco.c:69: Starting SCO dispatcher loop: hci0
bluealsa: bluez.c:997: RFCOMM: HFP Audio Gateway configured for device B8:27:EB:87:AD:65
bluealsa: ba-transport.c:635: State transition: 0 -> 2
bluealsa: rfcomm.c:593: Starting loop: RFCOMM: HFP Audio Gateway
bluealsa: ba-transport.c:982: Created new thread [ba-rfcomm]: RFCOMM: HFP Audio Gateway
bluealsa: ba-transport.c:635: State transition: 0 -> 2
bluealsa: ba-transport.c:982: Created new thread [ba-sco]: HFP Audio Gateway (CVSD)
bluealsa: sco.c:194: Starting SCO loop: HFP Audio Gateway (CVSD)
bluealsa: at.c:161: AT message: SET: command:+BRSF, value:628
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:+BRSF, value:2528
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
bluealsa: rfcomm.c:124: HFP state transition: 0 -> 2
bluealsa: at.c:161: AT message: TEST: command:+CIND, value:(null)
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:+CIND, value:("call",(0,1)),("callsetup",(0-3)),("service",(0-1)),("signal",(0-5)),("roam",(0-1)),("battchg",(0-5)),("callheld",(0-2))
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
bluealsa: rfcomm.c:124: HFP state transition: 2 -> 5
bluealsa: at.c:161: AT message: GET: command:+CIND, value:(null)
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:+CIND, value:0,0,0,0,0,0,0
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
bluealsa: rfcomm.c:124: HFP state transition: 5 -> 7
bluealsa: at.c:161: AT message: SET: command:+CMER, value:3,0,0,1,0
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
bluealsa: rfcomm.c:124: HFP state transition: 7 -> 8
bluealsa: rfcomm.c:124: HFP state transition: 8 -> 9
bluealsa: rfcomm.c:124: HFP state transition: 9 -> 13

bluealsa: bluealsa-dbus.c:139: Called: org.bluealsa.Manager1.GetPCMs()
bluealsa: bluealsa-dbus.c:333: Called: org.bluealsa.PCM1.Open()
bluealsa: ba-transport.c:875: Checking Broadcom internal SCO routing
bluealsa: ba-transport.c:881: Current SCO interface setup: 1 2 0 1 1
bluealsa: ba-transport.c:915: New SCO link: B8:27:EB:87:AD:65: 15 (MTU: R:64 W:64)
bluealsa: ba-transport.c:635: State transition: 2 -> 2
bluealsa: ba-transport.c:863: Reusing SCO: 15
bluealsa: ba-transport.c:931: Closing SCO: 15
```

aplay (doesn't exit until Ctrl+C pressed):
```
$ aplay -v -D bluealsa:DEV=B8:27:EB:87:AD:65,PROFILE=sco test.wav 
$ aplay -v -D bluealsa:DEV=B8:27:EB:87:AD:65,PROFILE=sco test.wav 
bluealsa-pcm.c:625: /org/bluealsa/hci0/dev_B8_27_EB_87_AD_65/sco: Setting constraints
Playing WAVE 'test.wav' : Signed 16 bit Little Endian, Rate 22050 Hz, Mono
bluealsa-pcm.c:344: /org/bluealsa/hci0/dev_B8_27_EB_87_AD_65/sco: Initializing HW
bluealsa-pcm.c:371: /org/bluealsa/hci0/dev_B8_27_EB_87_AD_65/sco: FIFO buffer size: 4096
bluealsa-pcm.c:377: /org/bluealsa/hci0/dev_B8_27_EB_87_AD_65/sco: Selected HW buffer: 4 periods x 2000 bytes == 8000 bytes
bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_B8_27_EB_87_AD_65/sco: Initializing SW
bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_B8_27_EB_87_AD_65/sco: Initializing SW
bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_B8_27_EB_87_AD_65/sco: Initializing SW
bluealsa-pcm.c:414: /org/bluealsa/hci0/dev_B8_27_EB_87_AD_65/sco: Prepared
bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_B8_27_EB_87_AD_65/sco: Initializing SW
Plug PCM: Rate conversion PCM (8000, sformat=S16_LE)
Converter: libspeex (builtin)
Protocol version: 10002
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
[... 61 lines cut ...]
bluealsa: ../../src/rfcomm.c:124: HFP state transition: 6 -> 7
bluealsa: ../../src/rfcomm.c:106: Sending AT message: SET: command:+CMER, value:3,0,0,1,0
bluealsa: ../../src/at.c:161: AT message: RESP: command:, value:OK
bluealsa: ../../src/rfcomm.c:124: HFP state transition: 7 -> 8
bluealsa: ../../src/rfcomm.c:124: HFP state transition: 8 -> 9
bluealsa: ../../src/rfcomm.c:124: HFP state transition: 9 -> 13

bluealsa: ../../src/sco.c:95: New incoming SCO link: 70:2C:1F:32:15:85: 8
bluealsa: ../../src/ba-transport.c:931: Closing SCO: 8
```

bluealsa-aplay:
```
$ bluealsa-aplay --profile-sco -v 70:2C:1F:32:15:85
Selected configuration:
  BlueALSA service: org.bluealsa
  PCM device: default
  PCM buffer time: 500000 us
  PCM period time: 100000 us
  Bluetooth device(s): 70:2C:1F:32:15:85
  Profile: SCO
bluealsa-aplay: ../../utils/aplay.c:801: Starting main loop
bluealsa-aplay: ../../utils/aplay.c:525: Creating PCM worker 70:2C:1F:32:15:85
bluealsa-aplay: ../../utils/aplay.c:356: Starting PCM loop
```

killerkalamari, 2019-11-07:
This is what it looks like when I play to the TaoTronics Wireless Receiver TT-BR08 (where I can hear the sound):

```
$ sudo bluealsa -p hfp-ag
bluealsa: ba-adapter.c:151: Freeing adapter: hci0
bluealsa: bluez.c:1105: Creating hands-free profile object: /org/bluez/HFP/AudioGateway
bluealsa: bluez.c:1135: Registering hands-free profile: /org/bluez/HFP/AudioGateway
bluealsa: main.c:353: Acquiring D-Bus service name: org.bluealsa
bluealsa: main.c:358: Starting main dispatching loop

bluealsa: bluez.c:1073: Called: org.bluez.Profile1.NewConnection()
bluealsa: sco.c:154: Created SCO dispatcher [ba-sco-dispatch]: hci0
bluealsa: sco.c:69: Starting SCO dispatcher loop: hci0
bluealsa: bluez.c:997: RFCOMM: HFP Audio Gateway configured for device E9:08:EF:2D:40:33
bluealsa: ba-transport.c:635: State transition: 0 -> 2
bluealsa: rfcomm.c:593: Starting loop: RFCOMM: HFP Audio Gateway
bluealsa: ba-transport.c:982: Created new thread [ba-rfcomm]: RFCOMM: HFP Audio Gateway
bluealsa: ba-transport.c:635: State transition: 0 -> 2
bluealsa: ba-transport.c:982: Created new thread [ba-sco]: HFP Audio Gateway (CVSD)
bluealsa: sco.c:194: Starting SCO loop: HFP Audio Gateway (CVSD)
bluealsa: at.c:161: AT message: SET: command:+BRSF, value:191
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:+BRSF, value:2528
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
bluealsa: rfcomm.c:124: HFP state transition: 0 -> 2
bluealsa: at.c:161: AT message: TEST: command:+CIND, value:(null)
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:+CIND, value:("call",(0,1)),("callsetup",(0-3)),("service",(0-1)),("signal",(0-5)),("roam",(0-1)),("battchg",(0-5)),("callheld",(0-2))
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
bluealsa: rfcomm.c:124: HFP state transition: 2 -> 5
bluealsa: at.c:161: AT message: GET: command:+CIND, value:(null)
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:+CIND, value:0,0,0,0,0,0,0
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
bluealsa: rfcomm.c:124: HFP state transition: 5 -> 7
bluealsa: at.c:161: AT message: SET: command:+CMER, value:3, 0, 0, 1
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
bluealsa: rfcomm.c:124: HFP state transition: 7 -> 8
bluealsa: rfcomm.c:124: HFP state transition: 8 -> 9
bluealsa: rfcomm.c:124: HFP state transition: 9 -> 13
bluealsa: at.c:161: AT message: SET: command:+VGS, value:15
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
bluealsa: at.c:161: AT message: GET: command:+BTRH, value:(null)
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
bluealsa: at.c:161: AT message: SET: command:+XAPL, value:0000-0000-0100,7
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:+XAPL=BlueALSA,0
bluealsa: at.c:161: AT message: SET: command:+IPHONEACCEV, value:2,1,8,2,0
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
bluealsa: at.c:161: AT message: CMD: command:+CLCC, value:(null)
bluealsa: Unsupported AT message: CMD: command:+CLCC, value:(null)
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:ERROR
bluealsa: at.c:161: AT message: SET: command:+BIA, value:1,1,0,0,0,0,1
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
bluealsa: at.c:161: AT message: SET: command:+CSRSF, value:0,0,0,1,0,0,0
bluealsa: Unsupported AT message: SET: command:+CSRSF, value:0,0,0,1,0,0,0
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:ERROR

bluealsa: bluealsa-dbus.c:139: Called: org.bluealsa.Manager1.GetPCMs()
bluealsa: bluealsa-dbus.c:333: Called: org.bluealsa.PCM1.Open()
bluealsa: ba-transport.c:875: Checking Broadcom internal SCO routing
bluealsa: ba-transport.c:881: Current SCO interface setup: 1 2 0 1 1
bluealsa: ba-transport.c:635: State transition: 2 -> 2
bluealsa: ba-transport.c:915: New SCO link: E9:08:EF:2D:40:33: 15 (MTU: R:64 W:64)
bluealsa: ba-transport.c:863: Reusing SCO: 15
bluealsa: ba-transport.c:863: Reusing SCO: 15
bluealsa: ba-transport.c:718: PCM drained
bluealsa: a2dp.c:122: PCM read buffer flushed: 0
bluealsa: a2dp.c:122: PCM read buffer flushed: 0
bluealsa: ba-transport.c:955: Closing PCM: 8
bluealsa: ba-transport.c:931: Closing SCO: 15
```

```
[... 36 lines cut ...]
  period_time  : 125000
  tstamp_mode  : NONE
  tstamp_type  : GETTIMEOFDAY
  period_step  : 1
  avail_min    : 2756
  period_event : 0
  start_threshold  : 11025
  stop_threshold   : 11025
  silence_threshold: 0
  silence_size : 0
  boundary     : 6206523236469964800
Slave: BlueALSA PCM: /org/bluealsa/hci0/dev_E9_08_EF_2D_40_33/sco
BlueALSA BlueZ device: /org/bluez/hci0/dev_E9_08_EF_2D_40_33
BlueALSA Bluetooth codec: 0x1
bluealsa-pcm.c:264: /org/bluealsa/hci0/dev_E9_08_EF_2D_40_33/sco: Starting
bluealsa-pcm.c:137: /org/bluealsa/hci0/dev_E9_08_EF_2D_40_33/sco: Starting IO loop: 8
bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_E9_08_EF_2D_40_33/sco: Initializing SW
bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_E9_08_EF_2D_40_33/sco: Initializing SW
bluealsa-pcm.c:169: /org/bluealsa/hci0/dev_E9_08_EF_2D_40_33/sco: IO thread paused: 4
bluealsa-pcm.c:305: /org/bluealsa/hci0/dev_E9_08_EF_2D_40_33/sco: Stopping
bluealsa-pcm.c:105: /org/bluealsa/hci0/dev_E9_08_EF_2D_40_33/sco: IO thread cleanup
bluealsa-pcm.c:305: /org/bluealsa/hci0/dev_E9_08_EF_2D_40_33/sco: Stopping
bluealsa-pcm.c:384: /org/bluealsa/hci0/dev_E9_08_EF_2D_40_33/sco: Freeing HW
bluealsa-pcm.c:334: /org/bluealsa/hci0/dev_E9_08_EF_2D_40_33/sco: Closing
```

killerkalamari, 2019-11-07:
Another log, this time from a soundbar where I don't hear anything. I'm able to play to this soundbar from my Android phone, where I manually uncheck "Media audio" in the Bluetooth settings for the device, which results in it saying "Connected (no media)". If I play a sound on my phone, it plays over my phone speaker. Then I run the app BTmono and turn that On. If I play a high quality sound on my phone, it then plays to the soundbar (in the poor quality you'd expect from HFP). When I use the same process in my car, it shows up as a phone, so that indicates to me that it is in HFP mode rather than HSP mode. Additionally, I cannot seem to connect to the soundbar in HSP mode, nor is it listed in the capabilities. All that leads me to believe that the Android test is valid. 

Anyhow, here are the logs from the bluealsa playback:

```
$ sudo bluealsa -p hfp-ag
bluealsa: ba-adapter.c:151: Freeing adapter: hci0
bluealsa: bluez.c:1105: Creating hands-free profile object: /org/bluez/HFP/AudioGateway
bluealsa: bluez.c:1135: Registering hands-free profile: /org/bluez/HFP/AudioGateway
bluealsa: main.c:353: Acquiring D-Bus service name: org.bluealsa
bluealsa: main.c:358: Starting main dispatching loop

bluealsa: bluez.c:1073: Called: org.bluez.Profile1.NewConnection()
bluealsa: sco.c:154: Created SCO dispatcher [ba-sco-dispatch]: hci0
bluealsa: sco.c:69: Starting SCO dispatcher loop: hci0
bluealsa: bluez.c:997: RFCOMM: HFP Audio Gateway configured for device 6B:F4:54:06:AC:D1
bluealsa: ba-transport.c:635: State transition: 0 -> 2
bluealsa: rfcomm.c:593: Starting loop: RFCOMM: HFP Audio Gateway
bluealsa: ba-transport.c:982: Created new thread [ba-rfcomm]: RFCOMM: HFP Audio Gateway
bluealsa: ba-transport.c:635: State transition: 0 -> 2
bluealsa: ba-transport.c:982: Created new thread [ba-sco]: HFP Audio Gateway (CVSD)
bluealsa: sco.c:194: Starting SCO loop: HFP Audio Gateway (CVSD)
bluealsa: at.c:161: AT message: SET: command:+BRSF, value:031
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:+BRSF, value:2528
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
bluealsa: rfcomm.c:124: HFP state transition: 0 -> 2
bluealsa: at.c:161: AT message: TEST: command:+CIND, value:(null)
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:+CIND, value:("call",(0,1)),("callsetup",(0-3)),("service",(0-1)),("signal",(0-5)),("roam",(0-1)),("battchg",(0-5)),("callheld",(0-2))
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
bluealsa: rfcomm.c:124: HFP state transition: 2 -> 5
bluealsa: at.c:161: AT message: GET: command:+CIND, value:(null)
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:+CIND, value:0,0,0,0,0,0,0
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
bluealsa: rfcomm.c:124: HFP state transition: 5 -> 7
bluealsa: at.c:161: AT message: SET: command:+CMER, value:3,0,0,1
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
bluealsa: rfcomm.c:124: HFP state transition: 7 -> 8
bluealsa: rfcomm.c:124: HFP state transition: 8 -> 9
bluealsa: rfcomm.c:124: HFP state transition: 9 -> 13
bluealsa: at.c:161: AT message: TEST: command:+CHLD, value:(null)
bluealsa: Unsupported AT message: TEST: command:+CHLD, value:(null)
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:ERROR
bluealsa: at.c:161: AT message: SET: command:+CMEE, value:1
bluealsa: Unsupported AT message: SET: command:+CMEE, value:1
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:ERROR
bluealsa: at.c:161: AT message: SET: command:+CLIP, value:1
bluealsa: Unsupported AT message: SET: command:+CLIP, value:1
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:ERROR
bluealsa: at.c:161: AT message: SET: command:+CCWA, value:1
bluealsa: Unsupported AT message: SET: command:+CCWA, value:1
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:ERROR
bluealsa: at.c:161: AT message: SET: command:+NREC, value:0
bluealsa: Unsupported AT message: SET: command:+NREC, value:0
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:ERROR
bluealsa: at.c:161: AT message: SET: command:+VGS, value:15
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
bluealsa: at.c:161: AT message: SET: command:+VGM, value:15
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
bluealsa: at.c:161: AT message: SET: command:+XAPL, value:ABCD-1234-0100,10
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:+XAPL=BlueALSA,0

bluealsa: bluealsa-dbus.c:139: Called: org.bluealsa.Manager1.GetPCMs()
bluealsa: bluealsa-dbus.c:333: Called: org.bluealsa.PCM1.Open()
bluealsa: ba-transport.c:875: Checking Broadcom internal SCO routing
bluealsa: ba-transport.c:881: Current SCO interface setup: 1 2 0 1 1
bluealsa: ba-transport.c:915: New SCO link: 6B:F4:54:06:AC:D1: 15 (MTU: R:64 W:64)
bluealsa: ba-transport.c:635: State transition: 2 -> 2
bluealsa: ba-transport.c:863: Reusing SCO: 15
bluealsa: ba-transport.c:863: Reusing SCO: 15
bluealsa: ba-transport.c:718: PCM drained
[... 36 lines cut ...]
  period_time  : 125000
  tstamp_mode  : NONE
  tstamp_type  : GETTIMEOFDAY
  period_step  : 1
  avail_min    : 2756
  period_event : 0
  start_threshold  : 11025
  stop_threshold   : 11025
  silence_threshold: 0
  silence_size : 0
  boundary     : 6206523236469964800
Slave: BlueALSA PCM: /org/bluealsa/hci0/dev_6B_F4_54_06_AC_D1/sco
[... cut ...]
bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_6B_F4_54_06_AC_D1/sco: Initializing SW
bluealsa-pcm.c:169: /org/bluealsa/hci0/dev_6B_F4_54_06_AC_D1/sco: IO thread paused: 4
bluealsa-pcm.c:305: /org/bluealsa/hci0/dev_6B_F4_54_06_AC_D1/sco: Stopping
bluealsa-pcm.c:105: /org/bluealsa/hci0/dev_6B_F4_54_06_AC_D1/sco: IO thread cleanup
bluealsa-pcm.c:305: /org/bluealsa/hci0/dev_6B_F4_54_06_AC_D1/sco: Stopping
bluealsa-pcm.c:384: /org/bluealsa/hci0/dev_6B_F4_54_06_AC_D1/sco: Freeing HW
bluealsa-pcm.c:334: /org/bluealsa/hci0/dev_6B_F4_54_06_AC_D1/sco: Closing
```

killerkalamari, 2019-11-07:
Android phone with BTmono connected to bluealsa on my Raspberry Pi 3 also results in me being able to hear sound from the Pi 3.5mm jack:

```
$ sudo bluealsa -p hfp-hf
bluealsa: ../../src/ba-adapter.c:151: Freeing adapter: hci0
bluealsa: ../../src/bluez.c:1105: Creating hands-free profile object: /org/bluez/HFP/HandsFree
bluealsa: ../../src/bluez.c:1135: Registering hands-free profile: /org/bluez/HFP/HandsFree
bluealsa: ../../src/main.c:353: Acquiring D-Bus service name: org.bluealsa
bluealsa: ../../src/main.c:358: Starting main dispatching loop

bluealsa: ../../src/bluealsa-dbus.c:139: Called: org.bluealsa.Manager1.GetPCMs()

bluealsa: ../../src/bluez.c:1073: Called: org.bluez.Profile1.NewConnection()
bluealsa: ../../src/sco.c:154: Created SCO dispatcher [ba-sco-dispatch]: hci0
bluealsa: ../../src/sco.c:69: Starting SCO dispatcher loop: hci0
bluealsa: ../../src/bluez.c:997: RFCOMM: HFP Hands-Free configured for device 2C:FD:A1:21:0D:B4
bluealsa: ../../src/ba-transport.c:635: State transition: 0 -> 2
bluealsa: ../../src/ba-transport.c:982: Created new thread [ba-rfcomm]: RFCOMM: HFP Hands-Free
bluealsa: ../../src/ba-transport.c:635: State transition: 0 -> 2
bluealsa: ../../src/ba-transport.c:982: Created new thread [ba-sco]: HFP Hands-Free (CVSD)
bluealsa: ../../src/sco.c:194: Starting SCO loop: HFP Hands-Free (CVSD)
bluealsa: ../../src/rfcomm.c:593: Starting loop: RFCOMM: HFP Hands-Free
bluealsa: ../../src/rfcomm.c:106: Sending AT message: SET: command:+BRSF, value:628
bluealsa: ../../src/bluealsa-dbus.c:333: Called: org.bluealsa.PCM1.Open()
bluealsa: ../../src/at.c:161: AT message: RESP: command:+BRSF, value: 2919
bluealsa: ../../src/rfcomm.c:124: HFP state transition: 0 -> 1
bluealsa: ../../src/at.c:161: AT message: RESP: command:, value:OK
bluealsa: ../../src/rfcomm.c:124: HFP state transition: 1 -> 2
bluealsa: ../../src/rfcomm.c:106: Sending AT message: SET: command:+BAC, value:1
bluealsa: ../../src/at.c:161: AT message: RESP: command:, value:OK
bluealsa: ../../src/rfcomm.c:124: HFP state transition: 2 -> 3
bluealsa: ../../src/rfcomm.c:106: Sending AT message: TEST: command:+CIND, value:(null)
bluealsa: ../../src/at.c:161: AT message: RESP: command:+CIND, value: ("call",(0,1)),("callsetup",(0-3)),("service",(0-1)),("signal",(0-5)),("roam",(0,1)),("battchg",(0-5)),("callheld",(0-2))
bluealsa: ../../src/rfcomm.c:124: HFP state transition: 3 -> 4
bluealsa: ../../src/at.c:161: AT message: RESP: command:, value:OK
bluealsa: ../../src/rfcomm.c:124: HFP state transition: 4 -> 5
bluealsa: ../../src/rfcomm.c:106: Sending AT message: GET: command:+CIND, value:(null)
bluealsa: ../../src/at.c:161: AT message: RESP: command:+CIND, value: 0,0,1,5,0,4,0
bluealsa: ../../src/rfcomm.c:124: HFP state transition: 5 -> 6
bluealsa: ../../src/at.c:161: AT message: RESP: command:, value:OK
bluealsa: ../../src/rfcomm.c:124: HFP state transition: 6 -> 7
bluealsa: ../../src/rfcomm.c:106: Sending AT message: SET: command:+CMER, value:3,0,0,1,0
bluealsa: ../../src/at.c:161: AT message: RESP: command:, value:OK
bluealsa: ../../src/rfcomm.c:124: HFP state transition: 7 -> 8
bluealsa: ../../src/rfcomm.c:124: HFP state transition: 8 -> 9

bluealsa: ../../src/at.c:161: AT message: RESP: command:+CIEV, value: 2,2
bluealsa: ../../src/at.c:161: AT message: RESP: command:+CIEV, value: 2,3
bluealsa: ../../src/at.c:161: AT message: RESP: command:+CIEV, value: 1,1
bluealsa: ../../src/sco.c:95: New incoming SCO link: 2C:FD:A1:21:0D:B4: 8
bluealsa: ../../src/ba-transport.c:863: Reusing SCO: 8
bluealsa: ../../src/at.c:161: AT message: RESP: command:+CIEV, value: 2,0
```

```
$ bluealsa-aplay --profile-sco -v 2C:FD:A1:21:0D:B4
Selected configuration:
  BlueALSA service: org.bluealsa
  PCM device: default
  PCM buffer time: 500000 us
  PCM period time: 100000 us
  Bluetooth device(s): 2C:FD:A1:21:0D:B4
  Profile: SCO
bluealsa-aplay: ../../utils/aplay.c:801: Starting main loop

bluealsa-aplay: ../../utils/aplay.c:525: Creating PCM worker 2C:FD:A1:21:0D:B4
bluealsa-aplay: ../../utils/aplay.c:356: Starting PCM loop
```

killerkalamari, 2019-11-07:
There are some feature differences between the bluealsa (+BRSF value 2528) and my Android phone (+BRSF value 2919) AG's. Here is a decoding:

```
                                        bluealsa   Android
/* AG feature flags */                  2528       2919
#define HFP_AG_FEAT_3WC    (1 << 0)     False      True
#define HFP_AG_FEAT_ECNR   (1 << 1)     False      True
#define HFP_AG_FEAT_VOICE  (1 << 2)     False      True
#define HFP_AG_FEAT_RING   (1 << 3)     False      False
#define HFP_AG_FEAT_VTAG   (1 << 4)     False      False
#define HFP_AG_FEAT_REJECT (1 << 5)     True       True
#define HFP_AG_FEAT_ECS    (1 << 6)     True       True
#define HFP_AG_FEAT_ECC    (1 << 7)     True       False
#define HFP_AG_FEAT_EERC   (1 << 8)     True       True
#define HFP_AG_FEAT_CODEC  (1 << 9)     False      True
#define HFP_AG_FEAT_HFIND  (1 << 10)    False      False
#define HFP_AG_FEAT_ESCO   (1 << 11)    True       True
```

I believe that difference explains why I don't see these lines on the Android phone:
```
bluealsa: ../../src/rfcomm.c:106: Sending AT message: SET: command:+BAC, value:1
bluealsa: ../../src/at.c:161: AT message: RESP: command:, value:OK
bluealsa: ../../src/rfcomm.c:124: HFP state transition: 2 -> 3
```

On the bluealsa AG, HFP_AG_FEAT_CODEC is NO, due to my not enabling mSBC. Therefore, rfcomm.c:311 automatically knows to use CVSD. And it doesn't send +BAC, because of rfcomm.c:635. I don't see a problem there, it was just a difference in output I noticed between the two.

Another difference is that bluealsa performs this state transition, but with the the Android phone AG it doesn't:
```
bluealsa: ../../src/rfcomm.c:124: HFP state transition: 9 -> 13
```
State 9 is HFP_SLC_CONNECTED, and state 13 is HFP_CONNECTED. I see the reason at rfcomm.c:675, where if HFP_AG_FEAT_CODEC is set then there is a break, whereas that doesn't happen otherwise and it falls through to line 681 where it sets the state and then does a bluealsa_dbus_transport_update. I experimented with commenting out rfcomm.c:675 `if (t->rfcomm.hfp_features & HFP_AG_FEAT_CODEC)`, but it didn't seem to make any difference.

Arkq, 2019-11-08:
The HFP connection establishment with a phone or another bluealsa was not tested extensively. Currently I'm refactoring this part of code. Maybe in a week or two I will publish some enhancements. Right now I've changed the codec selection logic and other minor parts and it seems that the connection works much better. I have to test bluealsa-bluealsa interaction, though. Also, I think, I will add test coverage for such scenario. So, stay tuned :)

borine, 2019-11-09:
@killerkalamari - I see that in the hfp-ag output above, you have modified the bluealsa source to give a SCO mtu of 64; but the hfp-hf debug output unfortunately does not show the mtu. Have you also modified the mtu in the file sco.c as well as ba-transport.c, or are are you using the default of 48? You say the hfp-hf case is playing sound correctly so I guess whatever value for mtu you used must be the correct one. For my own information (i'm still investigating the whole sco mtu mess on linux) could you confirm the mtu value used in your hfp-hf test case? Thanks.

killerkalamari, 2019-11-09:
@borine You're right, I did make a single change to MTU (sorry, I am not sure which file right now, but I can look it up Monday), but only to bluealsa HFP-AG; the Raspberry Pi HFP-HF is stock.

Arkq, 2019-11-09:
@killerkalamari I've pushed some "work in progress" to master. Right now it should be possible to connect phone (Audio Gateway) to bluealsa. I've tested it with few devices (android 6.0 and iPhone 13.) and it seems to work. If you've got access to other phones it would be great if you could check connection with them. Playing between bluez-alsa and bluez-alsa might still not work. The problem is with connection acquisition logic in the SCO thread. I will have to rework it as well, because in the current form it causes more troubles than receiving data from AG (e.g. it's hard to implement proper PCM drain).

@borine Lately, I've been working around mSBC setup, and I've found out that there is a [socket option `sco_options` which returns MTU](https://github.com/Arkq/bluez-alsa/blob/d1e3d438c668a551200bcf38601bcacdd6c6e051/src/hci.c#L114). Unfortunately, this call also returns wrong value. I haven't check a lot of kernel code around SCO MTU, but it definitely seems like kernel bug...

borine, 2019-11-09:
@Arkq I'm currently thinking that the mtu value obtained from the kernel is unreliable only for USB HCI adapters; UART HCI ones seem to provide a correct value. This is based on my own very limited testing and the few forum posts for bluez and pulseaudio that give sufficient info. If I'm right then the function hci_sco_get_mtu() could make use of the struct hci_dev_info for the adapter to use code such as:
```
	switch (a->hci.type & 0x0f) {
		case HCI_USB:
			if (voice.setting == BT_VOICE_TRANSPARENT)
				options.mtu = 24;
			else
				options.mtu = 48;
			break;
//		[ any other special cases handled here ]
	}
```
where *a* is a pointer the the adapter info.
This is all just conjecture at present - I have no idea if TRANSPARENT streams  require special treatment for UART, for example.
However, if it turns out to be a practical approach, it would mean more embedded boards would have SCO work out-of-the-box with bluealsa rather than needing code patches.

@killerkalamari  your HFP-AG profile host requiring mtu == 64 is that an on-board UART adapter? And the HFP-HF profile host with mtu == 48 (ie default) is that an external USB adapter?
If both are on-board, it would be really helpful if you are able to test the HFP-HF host again, but with bluealsa modified to use an MTU of 64. The place to edit the code has unfortunately just moved again because of the re-factoring, so it might be best to wait a few weeks until @Arkq has the current set of SCO changes complete.

I'm also thinking it might be advantageous to add a command-line option allowing a savvy user to select the correct sco mtu policy for their adapter(s). Pulseaudio does this (actually with a module parameter). It would need more thought on the exact options to offer.

Once the current round of code re-factoring is complete I would be happy to submit a PR with some code with which to invite discussion and maybe even persuade some volunteers to test with a variety of adapters.

Arkq, 2019-11-11:
I think that most of the refactoring part is done already. Some enhancement might be required, though, because connection process is not reliable... But I think that the changes will be rather cosmetic.

@borine I've got also an UART based BT and the MTU reported by it is also 64 bytes. I haven't tested running bluealsa with such MTU, though. With 48/24 it seems to work fine. However, I will make test with original MTU and see how it will behave :)

borine, 2019-11-11:
@Arkq UART adapters tend to have an mtu size that fills their buffer (eg 64:1 is common for Broadcom/Cypress while 120:1 is common for Texas Instruments). USB tend to have larger buffers, but often smaller mtus (eg 50:8 for Qualcomm/Atheros, 64:8 for Intel).
The value of 64:8 is often generated by the kernel btusb driver as a quirk for controllers that it claims report wrong values. But for me, using a kernel mtu for any USB device crashes the controller whether "quirked" or not.

killerkalamari, 2019-11-11:
This is the patch I was using, took it out for now:

```
--- bluez-alsa-2cd6e4686f7808276480b430fb37df55dfdcc02b.orig/src/ba-transport.c	2019-10-17 13:32:38.942308319 -0700
+++ bluez-alsa-2cd6e4686f7808276480b430fb37df55dfdcc02b/src/ba-transport.c	2019-10-17 14:05:31.618254533 -0700
@@ -673,7 +673,7 @@ int ba_transport_drain_pcm(struct ba_tra
 	 * arbitrary time before releasing transport. In order to make it right,
 	 * there is a requirement for an asynchronous release mechanism, which
 	 * is not implemented - it requires a little bit of refactoring. */
-	usleep(200000);
+	usleep(1000000);
 
 	debug("PCM drained");
 	return 0;
@@ -839,8 +839,8 @@ static int transport_acquire_bt_sco(stru
 
 	/* XXX: It seems, that the MTU values returned by the HCI interface
 	 *      are incorrect (or our interpretation of them is incorrect). */
-	t->mtu_read = 48;
-	t->mtu_write = 48;
+	t->mtu_read = 64;
+	t->mtu_write = 64;
 
 	if (t->type.codec == HFP_CODEC_MSBC)
 		t->mtu_read = t->mtu_write = 24;
```

killerkalamari, 2019-11-11:
I've re-tested the various combos.

Newly working:
- bluealsa to bluealsa

Continues to work:
- bluealsa to TaoTronics
- Android to bluealsa

Still playing silence:
- bluealsa to soundbar

By my original bug report, this is resolved, thanks! I was hoping the resolution for bluealsa would also fix the soundbar, but no such luck, I guess. I know you're still in the middle of improving things, so I'll keep watching the commits. In case it's useful, here's the output I get when trying to play to the soundbar:

```
$ sudo bluealsa -p hfp-ag 
bluealsa: ba-adapter.c:151: Freeing adapter: hci0
bluealsa: bluez.c:1105: Creating hands-free profile object: /org/bluez/HFP/AudioGateway
bluealsa: bluez.c:1135: Registering hands-free profile: /org/bluez/HFP/AudioGateway
bluealsa: main.c:353: Acquiring D-Bus service name: org.bluealsa
bluealsa: main.c:358: Starting main dispatching loop

bluealsa: bluez.c:1073: Called: org.bluez.Profile1.NewConnection()
bluealsa: sco.c:154: Created SCO dispatcher [ba-sco-dispatch]: hci0
bluealsa: bluez.c:997: RFCOMM: HFP Audio Gateway configured for device 6B:F4:54:06:AC:D1
bluealsa: ba-transport.c:635: State transition: 0 -> 2
bluealsa: rfcomm.c:593: Starting loop: RFCOMM: HFP Audio Gateway
bluealsa: ba-transport.c:982: Created new thread [ba-rfcomm]: RFCOMM: HFP Audio Gateway
bluealsa: ba-transport.c:635: State transition: 0 -> 2
bluealsa: ba-transport.c:982: Created new thread [ba-sco]: HFP Audio Gateway (CVSD)
bluealsa: sco.c:194: Starting SCO loop: HFP Audio Gateway (CVSD)
bluealsa: sco.c:69: Starting SCO dispatcher loop: hci0
bluealsa: at.c:161: AT message: SET: command:+BRSF, value:031
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:+BRSF, value:2528
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
bluealsa: rfcomm.c:124: HFP state transition: 0 -> 2
bluealsa: at.c:161: AT message: TEST: command:+CIND, value:(null)
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:+CIND, value:("call",(0,1)),("callsetup",(0-3)),("service",(0-1)),("signal",(0-5)),("roam",(0-1)),("battchg",(0-5)),("callheld",(0-2))
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
bluealsa: rfcomm.c:124: HFP state transition: 2 -> 5
bluealsa: at.c:161: AT message: GET: command:+CIND, value:(null)
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:+CIND, value:0,0,0,0,0,0,0
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
bluealsa: rfcomm.c:124: HFP state transition: 5 -> 7
bluealsa: at.c:161: AT message: SET: command:+CMER, value:3,0,0,1
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
bluealsa: rfcomm.c:124: HFP state transition: 7 -> 8
bluealsa: rfcomm.c:124: HFP state transition: 8 -> 9
bluealsa: rfcomm.c:124: HFP state transition: 9 -> 13
bluealsa: at.c:161: AT message: TEST: command:+CHLD, value:(null)
bluealsa: Unsupported AT message: TEST: command:+CHLD, value:(null)
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:ERROR
bluealsa: at.c:161: AT message: SET: command:+CMEE, value:1
bluealsa: Unsupported AT message: SET: command:+CMEE, value:1
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:ERROR
bluealsa: at.c:161: AT message: SET: command:+CLIP, value:1
bluealsa: Unsupported AT message: SET: command:+CLIP, value:1
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:ERROR
bluealsa: at.c:161: AT message: SET: command:+CCWA, value:1
bluealsa: Unsupported AT message: SET: command:+CCWA, value:1
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:ERROR
bluealsa: at.c:161: AT message: SET: command:+NREC, value:0
bluealsa: Unsupported AT message: SET: command:+NREC, value:0
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:ERROR
bluealsa: at.c:161: AT message: SET: command:+VGS, value:15
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
bluealsa: at.c:161: AT message: SET: command:+VGM, value:15
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
bluealsa: at.c:161: AT message: SET: command:+XAPL, value:ABCD-1234-0100,10
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:+XAPL=BlueALSA,0
[... 43 lines cut ...]
  period_time  : 125000
  tstamp_mode  : NONE
  tstamp_type  : GETTIMEOFDAY
  period_step  : 1
  avail_min    : 2756
  period_event : 0
  start_threshold  : 11025
  stop_threshold   : 11025
  silence_threshold: 0
  silence_size : 0
  boundary     : 6206523236469964800
Slave: BlueALSA PCM: /org/bluealsa/hci0/dev_6B_F4_54_06_AC_D1/sco
BlueALSA BlueZ device: /org/bluez/hci0/dev_6B_F4_54_06_AC_D1
BlueALSA Bluetooth codec: 0x1
bluealsa-pcm.c:264: /org/bluealsa/hci0/dev_6B_F4_54_06_AC_D1/sco: Starting
bluealsa-pcm.c:137: /org/bluealsa/hci0/dev_6B_F4_54_06_AC_D1/sco: Starting IO loop: 8
bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_6B_F4_54_06_AC_D1/sco: Initializing SW
bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_6B_F4_54_06_AC_D1/sco: Initializing SW
bluealsa-pcm.c:169: /org/bluealsa/hci0/dev_6B_F4_54_06_AC_D1/sco: IO thread paused: 4
bluealsa-pcm.c:305: /org/bluealsa/hci0/dev_6B_F4_54_06_AC_D1/sco: Stopping
bluealsa-pcm.c:105: /org/bluealsa/hci0/dev_6B_F4_54_06_AC_D1/sco: IO thread cleanup
bluealsa-pcm.c:305: /org/bluealsa/hci0/dev_6B_F4_54_06_AC_D1/sco: Stopping
bluealsa-pcm.c:384: /org/bluealsa/hci0/dev_6B_F4_54_06_AC_D1/sco: Freeing HW
bluealsa-pcm.c:334: /org/bluealsa/hci0/dev_6B_F4_54_06_AC_D1/sco: Closing
```

killerkalamari, 2019-11-11:
Audio gateway (obsolete Samsung Artik):
```
$ sudo hciconfig hci0 version
hci0:	Type: Primary  Bus: UART
	BD Address: 70:2C:1F:32:15:85  ACL MTU: 1021:8  SCO MTU: 64:1
	HCI Version: 4.0 (0x6)  Revision: 0xe0
	LMP Version: 4.0 (0x6)  Subversion: 0x6119
	Manufacturer: Broadcom Corporation (15)
```

Hands-free (Pi 3):
```
sudo hciconfig hci0 version
hci0:	Type: Primary  Bus: UART
	BD Address: B8:27:EB:87:AD:65  ACL MTU: 1021:8  SCO MTU: 64:1
	HCI Version: 4.1 (0x7)  Revision: 0x168
	LMP Version: 4.1 (0x7)  Subversion: 0x2209
	Manufacturer: Broadcom Corporation (15)
```

I'll go ahead and change it to 64 and see what happens.

killerkalamari, 2019-11-11:
bluealsa to bluealsa output with the following patch applied to both AG and HF devices. Plays okay, the audio of longer files is momentarily garbled periodically, but I'm guessing that's just a feature of HFP:
```
--- bluez-alsa-ce960be902ea9dd9d7dd211afd0ec232433b3eda.orig/src/ba-transport.c	2019-11-11 13:18:58.868404453 -0700
+++ bluez-alsa-ce960be902ea9dd9d7dd211afd0ec232433b3eda/src/ba-transport.c	2019-11-11 13:34:46.102692599 -0700
@@ -713,7 +713,7 @@ int ba_transport_drain_pcm(struct ba_tra
 	 * arbitrary time before releasing transport. In order to make it right,
 	 * there is a requirement for an asynchronous release mechanism, which
 	 * is not implemented - it requires a little bit of refactoring. */
-	usleep(200000);
+	usleep(1000000);
 
 	debug("PCM drained");
 	return 0;
@@ -877,7 +877,9 @@ static int transport_acquire_bt_sco(stru
 
 	debug("New SCO link: %s: %d", batostr_(&t->d->addr), t->bt_fd);
 
+	debug("MTU before: R:%zu W:%zu", t->mtu_read, t->mtu_write);
 	t->mtu_read = t->mtu_write = hci_sco_get_mtu(t->bt_fd);
+	debug("MTU after: R:%zu W:%zu", t->mtu_read, t->mtu_write);
 
 	return t->bt_fd;
 
--- bluez-alsa-ce960be902ea9dd9d7dd211afd0ec232433b3eda.orig/src/hci.c	2019-11-11 13:18:51.684417398 -0700
+++ bluez-alsa-ce960be902ea9dd9d7dd211afd0ec232433b3eda/src/hci.c	2019-11-11 13:43:24.745751499 -0700
@@ -128,7 +128,7 @@ unsigned int hci_sco_get_mtu(int sco_fd)
 	/* XXX: It seems, that the MTU value returned by kernel
 	 *      is incorrect (or our interpretation of it). */
 
-	options.mtu = 48;
+	options.mtu = 64;
 	if (voice.setting == BT_VOICE_TRANSPARENT)
 		options.mtu = 24;
```

```
$ sudo bluealsa -p hfp-ag 
bluealsa: ba-adapter.c:151: Freeing adapter: hci0
bluealsa: bluez.c:1105: Creating hands-free profile object: /org/bluez/HFP/AudioGateway
bluealsa: bluez.c:1135: Registering hands-free profile: /org/bluez/HFP/AudioGateway
bluealsa: main.c:353: Acquiring D-Bus service name: org.bluealsa
bluealsa: main.c:358: Starting main dispatching loop

bluealsa: bluez.c:1073: Called: org.bluez.Profile1.NewConnection()
bluealsa: sco.c:167: Checking Broadcom internal SCO routing
bluealsa: sco.c:173: Current SCO interface setup: 0 2 0 1 1
bluealsa: sco.c:175: Setting SCO routing via transport interface
bluealsa: sco.c:201: Created SCO dispatcher [ba-sco-dispatch]: hci0
bluealsa: sco.c:77: Starting SCO dispatcher loop: hci0
bluealsa: bluez.c:997: RFCOMM: HFP Audio Gateway configured for device B8:27:EB:87:AD:65
bluealsa: ba-transport.c:635: State transition: 0 -> 2
bluealsa: rfcomm.c:698: Starting loop: RFCOMM: HFP Audio Gateway
bluealsa: ba-transport.c:949: Created new thread [ba-rfcomm]: RFCOMM: HFP Audio Gateway
bluealsa: ba-transport.c:635: State transition: 0 -> 2
bluealsa: ba-transport.c:949: Created new thread [ba-sco]: HFP Audio Gateway (CVSD)
bluealsa: sco.c:242: Starting SCO loop: HFP Audio Gateway (CVSD)
bluealsa: at.c:161: AT message: SET: command:+BRSF, value:628
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:+BRSF, value:2272
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
bluealsa: rfcomm.c:125: RFCOMM: HFP Audio Gateway state transition: 0 -> 2
bluealsa: at.c:161: AT message: TEST: command:+CIND, value:(null)
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:+CIND, value:("call",(0,1)),("callsetup",(0-3)),("service",(0-1)),("signal",(0-5)),("roam",(0-1)),("battchg",(0-5)),("callheld",(0-2))
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
bluealsa: rfcomm.c:125: RFCOMM: HFP Audio Gateway state transition: 2 -> 5
bluealsa: at.c:161: AT message: GET: command:+CIND, value:(null)
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:+CIND, value:0,0,0,0,0,0,0
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
bluealsa: rfcomm.c:125: RFCOMM: HFP Audio Gateway state transition: 5 -> 7
bluealsa: at.c:161: AT message: SET: command:+CMER, value:3,0,0,1,0
bluealsa: rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
[... 118 lines cut ...]
bluealsa: ../../src/rfcomm.c:125: RFCOMM: HFP Hands-Free state transition: 8 -> 9
bluealsa: ../../src/rfcomm.c:859: RFCOMM poll timeout
bluealsa: ../../src/rfcomm.c:627: Updating microphone gain: 15
bluealsa: ../../src/rfcomm.c:106: Sending AT message: SET: command:+VGM, value:15
bluealsa: ../../src/at.c:161: AT message: RESP: command:, value:OK
bluealsa: ../../src/rfcomm.c:125: RFCOMM: HFP Hands-Free state transition: 9 -> 9
bluealsa: ../../src/rfcomm.c:658: Updating speaker gain: 15
bluealsa: ../../src/rfcomm.c:106: Sending AT message: SET: command:+VGS, value:15
bluealsa: ../../src/at.c:161: AT message: RESP: command:, value:OK
bluealsa: ../../src/rfcomm.c:125: RFCOMM: HFP Hands-Free state transition: 9 -> 9
bluealsa: ../../src/sco.c:103: New incoming SCO link: 70:2C:1F:32:15:85: 8
bluealsa: ../../src/hci.c:126: SCO link socket MTU: 8: 64
bluealsa: ../../src/sco.c:412: SCO poll error status: 0x18
bluealsa: ../../src/ba-transport.c:898: Closing SCO: 8
```

```
$ bluealsa-aplay --profile-sco 70:2C:1F:32:15:85
bluealsa-aplay: ../../utils/aplay.c:801: Starting main loop

bluealsa-aplay: ../../utils/aplay.c:525: Creating PCM worker 70:2C:1F:32:15:85
bluealsa-aplay: ../../utils/aplay.c:356: Starting PCM loop

bluealsa-aplay: ../../utils/aplay.c:373: Device marked as inactive: 70:2C:1F:32:15:85
```

closed 2019-11-13 by killerkalamari