arkq/bluez-alsa #269: Bad use of fuction strcpy

Smulmeister, 2019-11-23:
Hi, 

There are two instances in your code of bad practice. The use of the function strcpy is bad practise in C. I would like to advise you to change that in your code. 

The exact implementation of strcpy can be found here:
https://github.com/torvalds/linux/blob/master/lib/string.c#L89

Here is a explanation why it is dangerous: 
https://en.wikibooks.org/wiki/C_Programming/string.h/strcpy

The two instances in your code are:
File: src/at.c - Line 39
File: src/asound/bluealsa-ctl.c - Line 215

If you have any questions let me know, good luck!

Kinds Regards

Maurits

Arkq, 2019-11-25:
> The use of the function strcpy is bad practise in C.

According to whom? :D Because according to me, the usage of this function is perfectly fine (if one knows how to use it, but anyway if one doesn't know how to use any memory copping functions, then usage of `mem*()` or `str*()` might be considered a bad practice).
 
> The two instances in your code are:
> File: src/at.c - Line 39
> File: src/asound/bluealsa-ctl.c - Line 215

Post here why the usage of strcpy is incorrect. If you've run e.g. symbolic execution please provide output from such evaluation.

killerkalamari, 2019-12-02:
@Smulmeister On bluealsa-ctl.c line 215 it's copying a ba_pcm device_path to a bt_dev device_path. But, look at bluealsa-ctl.c line 50, where the size of the bt_dev device path is set from the ba_pcm device_path. This one appears safe by design.

@Arkq The at.c strcpy could theoretically overflow with a long enough command or value, since msg is fixed at 256 in rfcomm.c rfcomm_write_at(). In at.c, the at_build() function could be modified to also require the length of buffer as a parameter. Then, at_build() will have enough information to verify that buffer can hold "len(command) + len(value) + 6 bytes". As far as I can tell, rfcomm_write_at() is the only caller, so it shouldn't be a big change.

Arkq, 2019-12-03:
@killerkalamari I think that in these both cases the usage of strcpy it is (rather) safe. There is indeed one potential overflow, though :/ But despite that one place, I'd rather make test coverage against such cases in general than fixing one by one if I spot some. That's why I've asked about symbolic execution. It will be a nice addition to the overall security of this project. Unfortunately, I've got rather no experience with symbolic execution, but someday I'd try to research it a little bit as a homework exercise :)