arkq/bluez-alsa #270: funny issue, aplay works but mpg123 not

sherifomran, 2019-11-24:
i made some adjustments to my system, in which the dbus was not working and it has now been fixed. But now i don't hear sound with mpg123 but i hear it with aplay
 
any ideas how to debug it ?

sherifomran, 2019-11-25:
if i check the bluealsa service i see BT Socket write error: bad file descriptor

borine, 2019-11-26:
For reliable use with mpg123, you need bluealsa version 2.0.0 or later. I suspect the error message
>BT socket write error: bad file descriptor

is from an earlier release of bluealsa which had a bug triggered by mpg123.

 If you compiled bluealsa yourself, I would recommend you obtain the latest source from the master branch in this repository and try again with that.

borine, 2021-06-04:
mpg123 playback was fixed with bluealsa v2.0.0

closed 2021-06-04 by borine