arkq/bluez-alsa #273: For mplayer is influenced by bluealsa

songyannan, 2019-12-02:
Platform:  rock-chip (rk3308) + bluez 5.0 
When I use bluealsa (1.2.0) with mplayer plays video and aplay plays sound  in my project, it seemed a bug is that bluetooth works maybe appearing nosound, I am sure the reason is the version of bluealsa. I use bluealsa(2.0.0) to solve this problem,but another bug appeared, which is aplay and mplayer work ablout 4 hours will trigger oom then bluealsa will be killed ?

Arkq, 2019-12-02:
Which profile you are using for playback, sco or a2dp? Also, which codec is
used for audio transfer (it is visible in bluealsa logs if compiled with
--enable-debug, or via DBus interface of bluealsa). Also, please check this
#215 (which is already in 2.0.0 release).

songyannan, 2019-12-03:
my profile is a2dp,   but I don't know which message to sure my codec. Here is my bluealsa log 
--------------------------
bluealsa: ctl.c:342: PCM requested for E0:B6:55:DC:56:15 type 1 stream 0
bluealsa: transport.c:834: New transport: 15 (MTU: R:672 W:716)
bluealsa: ctl.c:630: +-+-
bluealsa: bluez.c:1037: Signal: PropertiesChanged: org.bluez.MediaTransport1
bluealsa: transport.c:690: State transition: 0 -> 2
bluealsa: io.c:366: Starting IO loop: A2DP Source (SBC)
bluealsa: transport.c:104: Created new IO thread: A2DP Source (SBC)
bluealsa: transport.c:690: State transition: 2 -> 2
bluealsa: ctl.c:630: +-+-
----------------------------
Is useful ?

songyannan, 2019-12-04:
@Arkq 

my profile is a2dp, but I don't know which message to sure my codec. Here is my bluealsa log
bluealsa: ctl.c:342: PCM requested for E0:B6:55:DC:56:15 type 1 stream 0
bluealsa: transport.c:834: New transport: 15 (MTU: R:672 W:716)
bluealsa: ctl.c:630: +-+-
bluealsa: bluez.c:1037: Signal: PropertiesChanged: org.bluez.MediaTransport1
bluealsa: transport.c:690: State transition: 0 -> 2
bluealsa: io.c:366: Starting IO loop: A2DP Source (SBC)
bluealsa: transport.c:104: Created new IO thread: A2DP Source (SBC)
bluealsa: transport.c:690: State transition: 2 -> 2
bluealsa: ctl.c:630: +-+-
Is useful ?

borine, 2021-06-04:
This issue has had no activity for some time, and refers to an outdated version of bluez-alsa. If the OOM problem still exists in the latest release, please feel free to re-open this issue

closed 2021-06-04 by borine