arkq/bluez-alsa #274: default location for 20-bluealsa.conf is wrong

borine, 2019-12-04:
Using a default configuration of bluealsa with alsa-lib 1.1.8, I find the alsa config file is installed as:
`/usr/etc/alsa/conf.d/20-bluealsa.conf`
It should be
`/etc/alsa/conf.d/20-bluealsa.conf`
otherwise the alsa config routines will not find it.
This happens because the confgure.ac script uses `${prefix}` which defaults to `/usr`
My knowledge of autoconf is not sufficient to propose a correct solution, so I'll just leave it here as an issue :smile:

Arkq, 2019-12-04:
> I've just added a note to the wiki install from source page advising readers to set the correct directory when configuring.

Thanks, I'd try to fix it. Also, may thanks for the installation guide. Hi-quality wiki :+1:

infirit, 2019-12-04:
This is what I think happens if you don't pass ``--sysconfdir=/etc`` to ``configure``. Distro's will typically pass this by default in their build environment and is often forgotten about :smile:.

borine, 2019-12-05:
I've just looked at the configure.ac for the alsa-plugins repo [here](https://github.com/alsa-project/alsa-plugins/blob/master/configure.ac#L234)
They define an option `--with-alsalconfdir=dir [path where ALSA local add-on config files are stored]`
so that it can be different from the global sysconfdir - defaults to ${sysconfdir} if not set. This is exactly equivalent to bluealsa's `--with-alsaconfdir=dir` (shame about the slight difference in name though) - so perhaps this is a non-issue after all!

borine, 2019-12-05:
@infirit thanks for that clarification. I have updated the wiki again with this new information, and am closing this issue as I now think bluealsa's approach is consistent with other alsa plugin projects.

closed 2019-12-05 by borine