arkq/bluez-alsa #276: JBL Bar 3.1 soundshift issue

k3s4, 2019-12-09:
Hello,

I want to use soundshift feature together with bluealsa from my RPi3. Mainly it's about change automatically sound input to bluetooth when paired devices send music to soundbar but for some reason it's not working with bluealsa. Is that feature somehow connected with AVRCP ? 

root@raspberrypi:/var/log# amixer -D bluealsa contents
numid=1,iface=MIXER,name='JBL Bar 3.1 - A2DP Playback Switch'
  ; type=BOOLEAN,access=rw------,values=2
  : values=on,on
numid=2,iface=MIXER,name='JBL Bar 3.1 - A2DP Playback Volume'
  ; type=INTEGER,access=rw------,values=2,min=0,max=127,step=1
  : values=127,127

I have tried to change a2dp playback switch from on to off but it's not working - no source input change..

anyone have similar issues?

k3s4, 2020-01-18:
anyone ? :)

borine, 2021-06-04:
Better late than never ...

When using ALSA directly from applications without an audio server such as pulseaudio, there is no automatic switching from one device to another when the first becomes unavailable. There are some notes on what can be done in a wiki article here: https://github.com/Arkq/bluez-alsa/wiki/Using-bluealsa-as-default-ALSA-PCM

The bluealsa mixer switch you tried is actually just a mute switch, nothing to do with re-routing audio to alternate devices.

closed 2021-07-08 by borine