arkq/bluez-alsa #278: sco usb and uart differences

borine, 2019-12-18:
There are significant differences between the behaviour of UART and USB hci adapters. I think these differences require bluealsa to implement specific code for each. The main underlying distinction is that UART imposes no "packet" boundaries of its own - the size of a single transfer is limited only by the size of the buffers on each side; whereas USB imposes both size and timing constraints that are in addition to buffer size limits.

**Consider USB first:**

For SCO data (*not* A2DP) bluez uses the USB "isochronous" interface configured with a transaction interval of 1ms, and performs transfers at 3ms intervals. It is this 3ms interval that determines the size and timing of reads and writes to a SCO socket.

For CVSD voice, bluealsa writes raw mono 16bit 8kHz linear PCM. Doing the math gives a data rate of 16000 bytes per second, or **48 bytes per 3ms interval**. Magic number 48 explained!

For mSBC, bluealsa writes encoded, compressed, mono 16bit 16kHz PCM. The raw linear PCM stream works out at 32000 bytes per second, and mSBC encoding compresses this at 60 bytes of output for every 240 bytes of input. So we end up with 3 * 32 / 4 = **24 bytes per 3ms interval**. Magic number 24 explained :smile:

These magic numbers should not be interpreted as "maximums" (as in "mtu"). They are fixed requirements and if bluealsa fails to deliver exacly that number on time then the data rate will fall below that required to maintain the audio stream.

Note that the 3ms interval is measured by the USB clock, so trying to apply finer-grained timing adjustments using nanosleep() in the bluealsa SCO thread loop is counter-productive. Instead, bluealsa must concentrate on always being ready to read/write the correct chunk of data whenever the socket is ready.

@Arkq: As an aside, you are aware of a patch submitted on the kernel bluetooth mailing list that is proposing to modify bluez' implementation of mSBC streams on usb - latest update here:
https://www.spinics.net/lists/linux-bluetooth/msg82230.html . If that proposal is ever adopted, then the chunk size for mSBC will change from 24 to 60. Bluealsa will then have to determine which version of the btusb driver is loaded to know which value to use - I can't find any API that allows user space to query for the value.

**Now consider UART:**

The UART bus does not constrain the size or timing of individual transfers in any way. Thus the HCI can optimize data flow according to its own buffer capacity and latency requirements.

For CVSD, the most efficient strategy is to transfer as much data as the controller buffer can hold with each write. The only other constraint is that audio frames must not be fragmented, which means that the chunk size must be an even number. The buffer size is obtained by
`getsockopt(sco_fd, SOL_SCO, SCO_OPTIONS, &options, &len)`

For many UART adapters with smallish SCO buffers (eg broadcom typically uses 64 bytes) it is possible to maintain a sufficient data rate with smaller transfers (eg 48 bytes). However, for adapters designed with larger buffers (apparently some Texas Instruments ones have a 150 byte buffer) using 48 byte chunks results in dropped frames and very poor sound quality. Therefore bluealsa should always transfer chunks of size *sco_mtu* for CVSD streams on UART adapters.

For mSBC, (and any other codec using "transparent" voice) the most efficient strategy is to transfer one full codec frame at a time. So that is a chunk size of 60 bytes, always containing one complete mSBC frame.

I haven't tested the need for nanosleep() with UART yet. It is possible that using poll() is sufficient to achieve the correct data rate for SCO, but I haven't been able to confirm this just by reading the BT core spec and bluez code.

I am hoping to be able to implement some of this analysis experimentally, and will submit a PR if it goes well. However, that will not be before the new year now, so I wanted to raise this as an issue so that any other interested parties can comment and point out any mistakes or omissions; maybe others could even try some code to see if bluealsa would benefit from this approach.

Arkq, 2019-12-18:
Many thanks for such extensive research! Magic numbers explanation is most entertaining :)

> Note that the 3ms interval is measured by the USB clock, so trying to apply finer-grained timing adjustments using nanosleep() in the bluealsa SCO thread loop is counter-productive.

I'm not sure for USB bluetooth, but for UART bluetooth, incorrect timing has caused mSBC issues (missing packets). Before the 9eb717f commit, the delay was not working, so the write was triggered by data availability or SCO socket writeability. With such setup data flow was not linear (in the function of time) - there were linear parts and stalling parts. After fixing the delay, the data flow becomes linear and the mSBC "quality" was improved. But maybe these two events are not related....

For USB bluetooth, I'm not able to force mSBC playback, so I'm not sure whether correct delay accounting is important. I will try to investigate it though. I might try to remove delay logic and see whether CVSD will still work.

> If that proposal is ever adopted, then the chunk size for mSBC will change from 24 to 60.

It would be best to avoid that.... I'm experimenting with various options for detecting MTU from reading, or FIONREAD ioctl. But the result are not consistent. And of course it is not general solution, so it will not be merged into the master, but maybe it will lead me somewhere...

borine, 2019-12-19:
>for UART bluetooth, incorrect timing has caused mSBC issues (missing packets)

Ok, the bluetooth core spec is not as clear on timing requirements for UART as it is on USB.  So I think you have demonstrated here that bluealsa is responsible for correct timing on a UART HCI bus. 

This table summarizes the results of my analysis. *sco_mtu* stands for the value obtained from getsockopt().

| Bus Type | CVSD | mSBC |
| -------- | ---- | ---- |
| UART | chunk size: sco_mtu<br>timing delay: yes | chunk size: 60 bytes<br>timing delay: yes |
| USB | chunk size: 48 bytes<br>timing delay: no | chunk size: 24 bytes<br>timing delay: no |

Of course, other bus types exist, and this analysis needs to be extended to cover them also. I am unable to test any other types, but may do some more reading on them next year.

borine, 2019-12-27:
It turns out that, for  bluez SCO sockets, the POLLOUT event is **always** ready, so it cannot be used to control the timing of outgoing packets.

Next I will look at using the method adopted by pulseaudio which is to use the POLLIN event as the trigger for both read() and write() on a SCO socket. If that worls then the timing delay becomes unncessary for both USB and UART. I'll report back when I have some test results.

Arkq, 2019-12-27:
I might already say, that it will not work, because pollin does not trigger
if there is no incoming data (case where bluealsa talks to bluealsa in only
one direction). For headsets it will most likely work, because headsets
will send mic audio.

borine, 2019-12-27:
> pollin does not trigger if there is no incoming data (case where bluealsa talks to bluealsa in only
one direction)

Yes, absolutely. That was why I originally doubted the possibility of using a read() to establish the HCI SCO packet size. But *if* bluealsa to bluealsa is the *only* problem case (a big if that's yet to be proven) then it can be fixed by sending silence when no pcm input is present.

A bigger issue for me now is what to do when the sco socket is ready for output but there is not a full packet of data available. For CVSD I can easily pad with silence, but for mSBC I think that may not be possible. Similar issue when SCO input arrives but the buffer does not have space for it - perhaps simply drop the overflow for CVSD but no such easy answer for mSBC. I intend to think over that for a while.

borine, 2020-01-12:
After trialling all the ideas above, I have to concede that none have produced any measurable improvement over the existing bluealsa SCO code.

One addition that does at least remove the underruns that I see with msbc is the "Packet Loss Concelment" code from the bluetooth HFP spec. I needed to replace their "short" type declarations with "uint8_t" or "int16_t" in various places, but other than that it works well with bluealsa msbc to replace any missing audio frames and so keep the bit rate correct. 

The licencing conditions of the example code in the HFP spec is unclear to me, so I don't know if I am permitted to post it here as a PR. 

@Arkq I am happy to post my modifications to src/mcbc.[ch] to use the PLC code if you are interested, but you would have to satisfy yourself that it is OK to copy the spec example code, and obtain that directly.

Arkq, 2020-01-18:
> After trialling all the ideas above, I have to concede that none have produced any measurable improvement over the existing bluealsa SCO code.

D'oh....

> I am happy to post my modifications to src/mcbc.[ch] to use the PLC code if you are interested, but you would have to satisfy yourself that it is OK to copy the spec example code, and obtain that directly.

For personal use it will be "OK". But this project is used commercially as well, and if licensing might be an issue, then I would rather not merge it. Or at least add it as additional feature via configure. Anyway, I think that some reference of copyright statement from the HFP spec might be required...

borine, 2020-01-25:
I'm closing this issue now as I think there is no more I can do with it. Current bluealsa support for SCO audio links is good (better than pulseaudio I think) and I now believe the remaining issues are caused by bluez code, not bluealsa. 

Hopefully a free license implementation of msbc packet loss concealment will be released by someone one day as that would definitely help bluealsa handle radio interference with sco msbc better.

closed 2020-01-25 by borine

Arkq, 2020-01-25:
> I'm closing this issue now as I think there is no more I can do with it. Current bluealsa support for SCO audio links is good (better than pulseaudio I think) and I now believe the remaining issues are caused by bluez code, not bluealsa.

OK, many thanks for your work around SCO configuration. I think, that indeed, right now, SCO is more or less stable. In the near future, I have a plan to separate SCO reading and writing (as a preliminary for non-symmetrical A2DP codecs like FastStream). I will try to use two separate threads, which might simplify some actions (e.g. PCM drain). It might add a little bit of scheduling overhead, but I hope it will smooth out reading and writing even more (on SMP of course).