arkq/bluez-alsa #279: Stuttering audio

rcr777, 2019-12-22:
Hello everyone! Already asked on askubuntu, but no luck. Wanna try to get some help here.

My setup: NAS Cubietruck (Ubuntu Xenial with Armbian Linux 3.4.113-sun7i) with integrated bluetooth sends audio with mpd via bluealsa to bluetooth receiver (connected to speaker). Setup described here https://askubuntu.com/questions/1197072/sending-audio-to-bluetooth-speaker-with-bluealsa

Setup works - MPD can play the music but audio is really stuttering and actually not listenable. Changing distance between NAS and bluetooth receiver brings nothing so it's not a problem of poor signal. No luck with adding `auto_resample "no`" and `use_mmap "yes"` in audio_output of `mpd.conf` as recommended here https://wiki.archlinux.org/index.php/Music_Player_Daemon/Troubleshooting#High_CPU_usage_with_ALSA. Wifi on NAS is dissabled (reported to interfere with bluetooth on raspi systems).

htop shows CPU load initially up to 100% then about 7-13% when mpd is playing. Doesn't seem to be a performance problem. 

Am I doing something wrong. Any ideas?

borine, 2019-12-22:
Can you provide some more information on the versions of the various software components you are using, and whether you built from source or obtained from ubuntu repository or elsewhere?
```
bluetoothd --version
aplay --version
bluealsa --version
mpd --version
```
Ubuntu Xenial is I think quite old now, so there may be many issues with the bluetooth stack in the versions in the Xenial repository that have been fixed in more recent versions.

Quality of sound can also depend on how the bluetooth adapter is configured. I'm not familiar with the cubieboard; can you also post the processor model and bluetooth module model from the hardware specs?

rcr777, 2019-12-22:
Hello, Borine. I've built bluealsa from source with `../configure --enable-aac --enable-ofono --enable-debug`. 

`root@Artisan:~# bluetoothd --version
5.37
root@Artisan:~# aplay --version
aplay: version 1.1.0 by Jaroslav Kysela <perex@perex.cz>
root@Artisan:~# bluealsa --version
2.0.0
root@Artisan:~# mpc version
mpd version: 0.19.0
root@Artisan:~#mpd --version
Music Player Daemon 0.19.12

Copyright (C) 2003-2007 Warren Dukes <warren.dukes@gmail.com>
Copyright (C) 2008-2014 Max Kellermann <max@duempel.org>
This is free software; see the source for copying conditions.  There is NO
warranty; not even MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Database plugins:
 simple proxy upnp

Storage plugins:
 local smbclient nfs

Neighbor plugins:
 smbclient upnp

Decoders plugins:
 [mad] mp3 mp2
 [mpg123] mp3
 [vorbis] ogg oga
 [oggflac] ogg oga
 [flac] flac
 [opus] opus ogg oga
 [sndfile] wav aiff aif au snd paf iff svx sf voc w64 pvf xi htk caf sd2
 [audiofile] wav au aiff aif
 [dsdiff] dff
 [dsf] dsf
 [faad] aac
 [mpcdec] mpc
 [wavpack] wv
 [modplug] 669 amf ams dbm dfm dsm far it med mdl mod mtm mt2 okt s3m stm ult umx xm
 [mikmod] amf dsm far gdm imf it med mod mtm s3m stm stx ult uni xm
 [sidplay] sid mus str prg P00
 [wildmidi] mid
 [fluidsynth] mid
 [adplug] amd d00 hsc laa rad raw sa2
 [ffmpeg] 16sv 3g2 3gp 4xm 8svx aa3 aac ac3 afc aif aifc aiff al alaw amr anim apc ape asf atrac au aud avi avm2 avs bap bfi c93 cak cin cmv cpk daud dct divx dts dv dvd dxa eac3 film flac flc fli fll flx flv g726 gsm gxf iss m1v m2v m2t m2ts m4a m4b m4v mad mj2 mjpeg mjpg mka mkv mlp mm mmf mov mp+ mp1 mp2 mp3 mp4 mpc mpeg mpg mpga mpp mpu mve mvi mxf nc nsv nut nuv oga ogm ogv ogx oma ogg omg opus psp pva qcp qt r3d ra ram rl2 rm rmvb roq rpl rvc shn smk snd sol son spx str swf tgi tgq tgv thp ts tsp tta xa xvid uv uv2 vb vid vob voc vp6 vmd wav webm wma wmv wsaud wsvga wv wve
 [gme] ay gbs gym hes kss nsf nsfe sap spc vgm vgz
 [pcm]

Output plugins:
 shout null fifo pipe alsa roar ao oss openal pulse jack httpd recorder

Encoder plugins:
 null vorbis opus lame wave flac

Archive plugins:
 [bz2] bz2
 [zzip] zip
 [iso] iso

Input plugins:
 file alsa archive curl ffmpeg smbclient nfs mms cdio_paranoia

Playlist plugins:
 extm3u m3u pls xspf asx rss soundcloud cue embcue

Protocols:
 file:// http:// https:// mms:// mmsh:// mmst:// mmsu:// gopher:// rtp:// rtsp:// rtmp:// rtmpt:// rtmps:// smb:// nfs:// cdda:// alsa://`

Cubietruck has this CPU inside:  Allwinner Tech SOC A20 ARM Cortex-A7 Dual-Core T

Unfortunately there's no info about bluetooth module in technical documentation. Not sure whether it helps, but this command I could quickly google in this regards:
`root@Artisan:~# lspci -nnk | grep -iA2 net; lsusb; hciconfig -a; dmesg | egrep -                                       i 'blue|firm'
pcilib: Cannot open /proc/bus/pci
lspci: Cannot find any working access method.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 1058:25f3 Western Digital Technologies, Inc.
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
hci0:   Type: BR/EDR  Bus: UART
        BD Address: 43:29:B1:55:01:01  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING PSCAN
        RX bytes:5164 acl:31 sco:0 events:496 errors:0
        TX bytes:685094 acl:857 sco:0 commands:48 errors:0
        Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
        Link policy: RSWITCH SNIFF
        Link mode: SLAVE ACCEPT
        Name: 'Artisan'
        Class: 0x1c0000
        Service Classes: Rendering, Capturing, Object Transfer
        Device Class: Miscellaneous,
        HCI Version: 4.0 (0x6)  Revision: 0x1000
        LMP Version: 4.0 (0x6)  Subversion: 0x220e
        Manufacturer: Broadcom Corporation (15)

[    2.328001] Registered led device: blue:ph21:led1
[   11.933167] Bluetooth: Core ver 2.16
[   11.947622] Bluetooth: HCI device and connection manager initialized
[   11.951503] Bluetooth: HCI socket layer initialized
[   11.958293] Bluetooth: L2CAP socket layer initialized
[   11.963965] Bluetooth: SCO socket layer initialized
[   11.975636] Bluetooth: HCI UART driver ver 2.2
[   11.979620] Bluetooth: HCI H4 protocol initialized
[   11.983667] Bluetooth: HCI BCSP protocol initialized
[   11.987580] Bluetooth: HCILL protocol initialized
[   11.991646] Bluetooth: HCIATH3K protocol initialized
[   12.002328] Bluetooth: RFCOMM TTY layer initialized
[   12.009217] Bluetooth: RFCOMM socket layer initialized
[   12.013853] Bluetooth: RFCOMM ver 1.11
[   12.022201] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[   16.278794] [ap6210] Final fw_path=/lib/firmware/ap6210/fw_bcm40181a2.bin
[   16.284490] [ap6210] Final nv_path=/lib/firmware/ap6210/nvram_ap6210.txt
[   16.307198] [ap6210] download firmware /lib/firmware/ap6210/fw_bcm40181a2.bin
[   16.665709] [ap6210] Firmware up: op_mode=0x0005, Broadcom Dongle Host Driver                                        mac=ac:83:f3:a8:72:fd
[   16.735412] [ap6210] Firmware version = wl0: Mar 10 2014 14:59:50 version 5.9                                       0.195.89.18 FWID 01-b13c09c7
[   30.030345] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   30.034574] Bluetooth: BNEP filters: protocol multicast`

borine, 2019-12-23:
I can see from the hciconfig output above that your bluetooth is a Broadcom/Cypress UART adapter. I don't know how to get detailed model numbers for a UART device other than from the board hardware specs.
One confusing entry (for me at least) in the dmesg output is the line
`[   11.991646] Bluetooth: HCIATH3K protocol initialized`
that looks like its for a Qualcom/Atheros device, which doesn't make sense for a Broadcom bluetooth module.

Anyway, you software versions look OK; although mpd 0.19.x is now very old and unsupported by the mpd developer, I *think* it probably works OK with bluealsa 2.0.0. I would strongly recommend that you put mpd to one side for the moment and use aplay for testing at least until you have confirmed that the system layers are working.

"Stuttering" audio almost always indicates underruns on the output device, so my first suspicion would be the configuration of the UART and bluetooth module. If they are not right then you will not get sufficient or steady throughput on the UART bus. Ignore any advice about editing configuration or startup files for alsa, bluealsa, or bluez - they are unlikely to be the cause and losing the default settings may introduce other problems. If possible, start with a fresh system install; if that can't be done then make sure you revert changes advised by the links you give above.

In particular, use the default bluealsa alsa config that is installed when you install you bluealsa build. Do not edit it. With alsa 1.1.0 it needs to be installed as:
`/usr/share/alsa/alsa.conf.d/20-bluealsa.conf`
Do not use .asoundrc or any other alsa config changes at this stage - you can add convenience entries to ~/.asoundrc or wherever *after* you have a default configuration working.

You do not need any special configuration or command line args for the bluetooth daemon either. My systemd unit file for bluetooth starts the daemon as:
`ExecStart=/usr/lib/bluetooth/bluetoothd`
Certainly do not use `--plugin=a2dp` as that was for Bluez 4 but you are using Bluez 5.

Next you need to investigate the UART setup. That is specific to your SoC and bluetooth module, so it is not possible to give a simple correct answer here. If there is a cubieboard user group that might be a good place to seek advice.

I use Raspbian on a RPi Zero W with good results. Raspian confiures the UART using the bluez utility "hciattach". I can see its settings using:
```
borine@pizero:~ $ ps ax | grep hciattach
  318 ?        S      0:00 /usr/bin/hciattach /dev/serial1 bcm43xx 3000000 flow - b8:27:eb:2b:8d:c3
```
The raspbian hci attach not only configures the UART serial settings, but also uploads a firmware file to the bluetooth module. These two actions have to be done together, as well as setting the bt address, so the whole process is performed by a single program. Ubuntu may not use hciattach: some distributions (eg Arch) have deprecated it and use btattach instead. Unfortunately many users have reported problems with btattach on the RPi. That's why you may need more specific help for your board and distribution.

rcr777, 2019-12-23:
Dear borine, thank you for your suggestions and advice.  I've already tried to play mp3 with aplay via bluealsa, the sound quality is even worse as with mpd. Anyway I'll go through your suggestion step by step and will report the result.

rcr777, 2019-12-23:
Fast google search gave me this: https://linux-sunxi.org/Cubietruck/Bluetooth

Is it what I'm looking for?

rcr777, 2019-12-23:
I've found bluetooth module of Cubietruck: https://linux-arm-kernel.infradead.narkive.com/iGZuabVX/patch-2-7-gpiolib-support-purely-name-based-gpiod-lookup-in-device-trees

`The CubieTruck has an AMPAK AP6210 WiFi+Bluetooth module. The Bluetooth part is a BCM20710 device connected to UART2 in the A20 SoC.`

Does this make something for hciattach clear?

borine, 2019-12-23:
>  I've already tried to play mp3 with aplay

aplay does not play mp3 files. You'll need a .wav test file for use with aplay.
The reason I recommend aplay for testing is that it is a very simple application with no configuration required.

I forgot to mention in my previous post that it is best to run bluealsa (as root) with no arguments for this initial testing - it defaults to a2dp output which is what you need to test first. Then use aplay thus (assuming your test file is music.wav):
```
aplay --verbose -D bluealsa:DEV=43:29:B1:55:01:01,PROFILE=a2dp music.wav
```
The cubietruck looks to be sufficiently different from the RPi that my experience with RPi may not be as useful to you as I had hoped. Try to find out if any one else has successfully used bluetooth audio on cubietruck with Ubuntu - either with bluealsa or pulseaudio - and check your UART set up against theirs. Me trying to guess how to do it may end up leading you down the wrong path.

rcr777, 2019-12-23:
With reversed bluealsa config and started without any arguments bluealsa and bluetooth aplay plays the same way (with stuttering) as mpd used to:
```

root@Artisan:~# aplay --verbose -D bluealsa:DEV=EB:06:EF:A6:A9:D4,PROFILE=a2dp /ncsdata/Music/ss.wav
D: ../../../src/asound/bluealsa-pcm.c:625: /org/bluealsa/hci0/dev_EB_06_EF_A6_A9_D4/a2dp: Setting constraints
Playing WAVE '/ncsdata/Music/ss.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
D: ../../../src/asound/bluealsa-pcm.c:344: /org/bluealsa/hci0/dev_EB_06_EF_A6_A9_D4/a2dp: Initializing HW
D: ../../../src/asound/bluealsa-pcm.c:371: /org/bluealsa/hci0/dev_EB_06_EF_A6_A9_D4/a2dp: FIFO buffer size: 4096
D: ../../../src/asound/bluealsa-pcm.c:377: /org/bluealsa/hci0/dev_EB_06_EF_A6_A9_D4/a2dp: Selected HW buffer: 6 periods x 16380 bytes <= 98284 bytes
D: ../../../src/asound/bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_EB_06_EF_A6_A9_D4/a2dp: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_EB_06_EF_A6_A9_D4/a2dp: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_EB_06_EF_A6_A9_D4/a2dp: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:414: /org/bluealsa/hci0/dev_EB_06_EF_A6_A9_D4/a2dp: Prepared
D: ../../../src/asound/bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_EB_06_EF_A6_A9_D4/a2dp: Initializing SW
Plug PCM: Rate conversion PCM (48000, sformat=S16_LE)
Converter: libspeex (builtin)
Protocol version: 10002
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 44100
  exact rate   : 44100 (44100/1)
  msbits       : 16
  buffer_size  : 22574
  period_size  : 3763
  period_time  : 85333
  tstamp_mode  : NONE
  tstamp_type  : GETTIMEOFDAY
  period_step  : 1
  avail_min    : 3763
  period_event : 0
  start_threshold  : 22574
  stop_threshold   : 22574
  silence_threshold: 0
  silence_size : 0
  boundary     : 1479409664
Slave: BlueALSA PCM: /org/bluealsa/hci0/dev_EB_06_EF_A6_A9_D4/a2dp
BlueALSA BlueZ device: /org/bluez/hci0/dev_EB_06_EF_A6_A9_D4
BlueALSA Bluetooth codec: 0
D: ../../../src/asound/bluealsa-pcm.c:264: /org/bluealsa/hci0/dev_EB_06_EF_A6_A9_D4/a2dp: Starting
D: ../../../src/asound/bluealsa-pcm.c:137: /org/bluealsa/hci0/dev_EB_06_EF_A6_A9_D4/a2dp: Starting IO loop: 8
^CAborted by signal Interrupt...
aplay: pcm_write:1940: write error: Interrupted system call
D: ../../../src/asound/bluealsa-pcm.c:305: /org/bluealsa/hci0/dev_EB_06_EF_A6_A9_D4/a2dp: Stopping
D: ../../../src/asound/bluealsa-pcm.c:105: /org/bluealsa/hci0/dev_EB_06_EF_A6_A9_D4/a2dp: IO thread cleanup
D: ../../../src/asound/bluealsa-pcm.c:384: /org/bluealsa/hci0/dev_EB_06_EF_A6_A9_D4/a2dp: Freeing HW
D: ../../../src/asound/bluealsa-pcm.c:334: /org/bluealsa/hci0/dev_EB_06_EF_A6_A9_D4/a2dp: Closing
```

mpd now remains silent 

```
root@Artisan:~# mpc play
Parov Stelar - All Night (Original)
[playing] #1/1   0:00/2:46 (0%)
volume:100%   repeat: off   random: off   single: off   consume: off
root@Artisan:~# mpc stop
volume:100%   repeat: off   random: off   single: off   consume: off
ERROR: Failed to open audio output
```

Anyway I'll try to dig in direction (UART setup) you've pointed. Thanks!

borine, 2019-12-23:
Having removed your edit from the bluealsa config file, you will need to update your mpd.conf file:
```
audio_output {
        type            "alsa"
        name            "ALSA Bluetooth Headset"
        device          "bluealsa:DEV=43:29:B1:55:01:01,PROFILE=a2dp"
        mixer_type    "software"
}
```

rcr777, 2019-12-30:
Well then, no idea on Armbian forum: https://forum.armbian.com/topic/12515-stuttering-audio-when-sending-to-bluetooth-receiver-via-bluealsa/ 
On https://www.cubieforums.com/index.php it is impossible to sign up (for unknown reason) and thus ask a question. 

Does it make sense to try hciattach as you did, borine? As long as you have `bcm43xx` in your hciattach settings which I guess stands for Broadcom bluetooth module, it shouldn't make dramatic difference? I'l found also updated firmware can be found at https://github.com/armbian/firmware/tree/master/ap6210 Should I try just to update firmware first?

borine, 2019-12-30:
> Does it make sense to try hciattach as you did, borine? As long as you have bcm43xx in your hciattach settings which I guess stands for Broadcom bluetooth module, it shouldn't make dramatic [...]

I don't know whether Raspbian modifies hciattach in any way, Assuming Armbian and Raspbian use the same build of hciattach, then the params you need will depend on the requirements of your bluetooth module. The number 3000000 in my hciattach command line is the baud rate for the UART serial device, you would need to determine which baud rates are supported by your module and choose a suitable one (possibly the highest available? - I don't know how to choose the "best" value).

>Should I try just to update firmware first?

Have you been able to determine what UART settings are currently in use by your bt module?

I think its generally best to always use the latest available firmware, however whether that works in this case I have no idea.

borine, 2021-06-21:
Many changes have been made to both RPi firmware and bluez-alsa since this issue was last updated. It is very likely that the problem no longer exists. Please open a new issue if you have a problem with the latest bluez-alsa code on Armbian as the information here is now outdated.

closed 2021-06-21 by borine