arkq/bluez-alsa #280: Couldn't get BlueALSA transport: No such device on Raspberry Pi

jiserra, 2019-12-29:
I tried everything, I read all the issues, I've spent more than 6 hours with this and I can't figure out what's happening, why I can't make this work.

I have a BT record player. I want to record the signal into a wav file on my Raspberry Pi with Stretch. I successfully connected to the record player using `bluetoothctl`.

But when I try to run `arecord -D bluealsa:HCI=hci0,DEV=F4:04:4C:11:DF:37,PROFILE=a2dp music/test.wav` it fails with

```
ALSA lib bluealsa-pcm.c:680:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA transport: No such device
arecord: main:788: audio open error: No such device
```

I'm desperate, I read every article and issue and I don't know what else to do.

borine, 2019-12-30:
according to
```
bluealsa --help
```

you need to start bluealsa with
```
bluealsa -p a2dp-sink
```
You appear to be using an old version of bluealsa, so there may be other issues, but hopefully the above will get you past this immediate issue,

jiserra, 2019-12-30:
Thanks for answering! If I execute `bluealsa -p a2dp-sink` then I get:
```
bluealsa: Couldn't initialize controller thread: Bad file descriptor
```

What version should I be running? Currently I have 1.3.0

borine, 2019-12-30:
Early versions of bluealsa created a directory for socket files, usually /run/bluealsa - sometimes that was left behund if the demon did not shutdown cleanly.If that directory exists, delete it before starting bluealsa.

Arkq, 2019-12-30:
@jiserra please use master branch or at least version 2.0.0.

jiserra, 2020-01-02:
@Arkq Thanks for the answer. I could finally make version 2.0.0, but when I try to run `arecord -D bluealsa capture.wav`, it says

```
ALSA lib ../../../src/asound/bluealsa-pcm.c:719:(_snd_pcm_bluealsa_open) Unknown field interface
arecord: main:828: audio open error: Invalid argument
```

I'm so sorry if you already answered this, I tried searching for answers in the issues and I couldn't find any.

Arkq, 2020-01-03:
Version 2.0.0 is not compatible with 1.*.*, if you've got custom asoundrc file you will have to upgrade it. Remove "interface" field and look at README for how to properly setup custom asoundrc file.

jiserra, 2020-01-14:
I sincerely tried everything and I can't make this to work… I never felt so frustrated in my life. Thanks anyway.

b23prodtm, 2020-03-26:
To free cache and run bluealsa again in daemon mode from hci0, that's worked many times 👍 

```
rm -rf /var/run/bluealsa/
/usr/bin/bluealsa -i hci0 -p a2dp-source -p a2dp-sink &
``` 

This is _bluealsa --help_ output :
```
Usage:
  bluealsa [OPTION]...

Options:
  -h, --help            print this help and exit
  -V, --version         print version and exit
  -B, --dbus=NAME       D-Bus service name suffix
  -S, --syslog          send output to syslog
  -i, --device=hciX     HCI device to use
  -p, --profile=NAME    enable BT profile
  --a2dp-force-mono     force monophonic sound
  --a2dp-force-audio-cd force 44.1 kHz sampling
  --a2dp-keep-alive=SEC keep A2DP transport alive
  --a2dp-volume         control volume natively
  --sbc-quality=NB      set SBC quality to NB
  --aac-afterburner     enable afterburner
  --aac-latm-version=NB set LATM syntax version
  --aac-vbr-mode=NB     set VBR mode to NB
  --mp3-quality=NB      set encoder quality to NB
  --mp3-vbr-quality=NB  set VBR quality to NB

Available BT profiles:
  - a2dp-source Advanced Audio Source (AAC, MP3, SBC)
  - a2dp-sink   Advanced Audio Sink (AAC, MP3, SBC)
  - hfp-ofono   Hands-Free handled by oFono (hf & ag)
  - hfp-hf      Hands-Free (v1.7)
  - hfp-ag      Hands-Free Audio Gateway (v1.7)
  - hsp-hs      Headset (v1.2)
  - hsp-ag      Headset Audio Gateway (v1.2)

By default only output profiles are enabled, which includes A2DP Source and
HSP/HFP Audio Gateways. If one wants to enable other set of profiles, it is
required to explicitly specify all of them using `-p NAME` options.
```

closed 2021-06-21 by borine