arkq/bluez-alsa #281: Cannot play sound (Unknown PCM bluealsa)

darek-arch, 2020-01-04:
Hello, I cannot make my bluetooth headset work.
`[darek@LAPTOP ~]$ aplay -D bluealsa:SRV=org.bluealsa,DEV=00:18:09:67:57:8C,PROFILE=a2dp ./Downloads/Alesis-Fusion-Clean-Guitar-C3.wav 
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM bluealsa:SRV=org.bluealsa,DEV=00:18:09:67:57:8C,PROFILE=a2dp
aplay: main:828: audio open error: Nie ma takiego pliku ani katalogu`

My OS is Arch Linux
`[darek@LAPTOP ~]$ uname -a
Linux LAPTOP 5.4.7-arch1-1 #1 SMP PREEMPT Tue, 31 Dec 2019 17:20:16 +0000 x86_64 GNU/Linux
`

Device connects ok:
`Device 00:18:09:67:57:8C (public)
	Name: MDR-ZX220BT
	Alias: MDR-ZX220BT
	Class: 0x00240404
	Icon: audio-card
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)`

Bluezalsa logs:
`ty 04 20:58:14 LAPTOP systemd[1]: Started BluezAlsa proxy.
sty 04 20:58:14 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/ofono.c:628: Checking oFono service presence
sty 04 20:58:15 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/bluez.c:750: Creating media endpoint object: /org/bluez/hci0/A2DP/MPEG24/Source/1
sty 04 20:58:15 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/bluez.c:781: Registering media endpoint: /org/bluez/hci0/A2DP/MPEG24/Source/1
sty 04 20:58:15 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/bluez.c:750: Creating media endpoint object: /org/bluez/hci0/A2DP/MPEG24/Source/2
sty 04 20:58:15 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/bluez.c:781: Registering media endpoint: /org/bluez/hci0/A2DP/MPEG24/Source/2
sty 04 20:58:15 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/bluez.c:750: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/1
sty 04 20:58:15 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/bluez.c:781: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/1
sty 04 20:58:15 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/bluez.c:750: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/2
sty 04 20:58:15 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/bluez.c:781: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/2
sty 04 20:58:15 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/ba-adapter.c:151: Freeing adapter: hci0
sty 04 20:58:15 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/bluez.c:1068: Creating hands-free profile object: /org/bluez/HSP/AudioGateway
sty 04 20:58:15 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/bluez.c:1098: Registering hands-free profile: /org/bluez/HSP/AudioGateway
sty 04 20:58:15 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/bluez.c:1068: Creating hands-free profile object: /org/bluez/HFP/AudioGateway
sty 04 20:58:15 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/bluez.c:1098: Registering hands-free profile: /org/bluez/HFP/AudioGateway
sty 04 20:58:15 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/main.c:371: Acquiring D-Bus service name: org.bluealsa
sty 04 20:58:15 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/main.c:376: Starting main dispatching loop
sty 04 20:58:15 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/bluez.c:1257: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
sty 04 20:58:15 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/ba-adapter.c:151: Freeing adapter: hci0
sty 04 20:58:15 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/bluez.c:1257: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
sty 04 20:59:44 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/bluez.c:1036: Called: org.bluez.Profile1.NewConnection()
sty 04 20:59:44 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/sco.c:201: Created SCO dispatcher [ba-sco-dispatch]: hci0
sty 04 20:59:44 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/bluez.c:958: RFCOMM: HFP Audio Gateway configured for device 00:18:09:67:57:8C
sty 04 20:59:44 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/ba-transport.c:635: State transition: 0 -> 2
sty 04 20:59:44 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/sco.c:77: Starting SCO dispatcher loop: hci0
sty 04 20:59:44 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/ba-transport.c:947: Created new thread [ba-rfcomm]: RFCOMM: HFP Audio Gateway
sty 04 20:59:44 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/ba-transport.c:635: State transition: 0 -> 2
sty 04 20:59:44 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/rfcomm.c:808: Starting loop: RFCOMM: HFP Audio Gateway
sty 04 20:59:44 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/sco.c:242: Starting SCO loop: HFP Audio Gateway (CVSD)
sty 04 20:59:44 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/ba-transport.c:947: Created new thread [ba-sco]: HFP Audio Gateway (CVSD)
sty 04 20:59:44 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/at.c:161: AT message: SET: command:+BRSF, value:144
sty 04 20:59:44 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/rfcomm.c:106: Sending AT message: RESP: command:+BRSF, value:2272
sty 04 20:59:44 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
sty 04 20:59:44 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/rfcomm.c:125: RFCOMM: HFP Audio Gateway state transition: 0 -> 2
sty 04 20:59:44 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/at.c:161: AT message: TEST: command:+CIND, value:(null)
sty 04 20:59:44 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/rfcomm.c:106: Sending AT message: RESP: command:+CIND, value:("service",(0-1)),("call",>
sty 04 20:59:44 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/rfcomm.c:106: Sending AT message: RESP: command:(null), value:OK
sty 04 20:59:44 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/rfcomm.c:125: RFCOMM: HFP Audio Gateway state transition: 2 -> 5
sty 04 20:59:44 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/bluez.c:1257: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
sty 04 20:59:44 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/bluez.c:1257: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
sty 04 20:59:44 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/at.c:161: AT message: GET: command:+CIND, value:(null)
sty 04 20:59:44 LAPTOP bluealsa[371]: /usr/bin/bluealsa: D: ../../src/rfcomm.c:106: Sending AT message: RESP: command:+CIND, value:0,0,0,0,0,0,5
[... 45 lines cut ...]
#User=bluealsa
#Group=audio
ExecStart=/usr/bin/bluealsa

[Install]
WantedBy=multi-user.target`

Asoundrc
`[darek@LAPTOP ~]$ cat ./.asoundrc
pcm.!default {
    type hw
    card PCH
}

ctl.!default {
    type hw
    card PCH
}

defaults.bluealsa.service "org.bluealsa"
defaults.bluealsa.device "00:18:09:67:57:8C"
defaults.bluealsa.profile "a2dp"
defaults.bluealsa.delay 10000`

How can I debug it further?

borine, 2020-01-05:
This looks like the same as #280 : old bluealsa release.

darek-arch, 2020-01-05:
Thank you for replying. It seems I use the latest version
`[darek@LAPTOP ~]$ bluealsa -V
2.0.0`

borine, 2020-01-05:
Hmm, in that case, it seems that libasound is unaware of the bluealsa pcm plugin, so does not recognize the "bluealsa:" device type. Perhaps the bluealsa installation is incomplete. Typically the alsa plugins directory is:
`/usr/lib/x86_64-linux-gnu/alsa-lib/`

although that may depend on your distribution.
There should be a file `libasound_module_pcm_bluealsa.so` in the alsa plugins directory.

darek-arch, 2020-01-05:
I have it like this:
[root@LAPTOP darek]# ls -l /usr/lib/alsa-lib/
razem 524
-rw-r--r-- 1 root root 140536 12-01 09:35 libasound_module_ctl_bluealsa.a
-rwxr-xr-x 1 root root   1105 12-01 09:35 libasound_module_ctl_bluealsa.la
-rwxr-xr-x 1 root root 113848 12-01 09:35 libasound_module_ctl_bluealsa.so
-rw-r--r-- 1 root root 144486 12-01 09:35 libasound_module_pcm_bluealsa.a
-rwxr-xr-x 1 root root   1105 12-01 09:35 libasound_module_pcm_bluealsa.la
-rwxr-xr-x 1 root root 120728 12-01 09:35 libasound_module_pcm_bluealsa.so

borine, 2020-01-05:
My suspicion, based on the error message 
>audio open error: Nie ma takiego pliku ani katalogu 

 is that these files are in  the wrong directory. I'm not familiar with Arch, so I can't say for certain where they should be

darek-arch, 2020-01-05:
I found the problem. The config file, 20-bluealsa.conf, was located in the wrong directory. It was in /usr/etc/alsa/conf.d instead of /etc/alsa/conf.d. After moving the file it works!
Thank you for your help.

closed 2020-01-05 by darek-arch