arkq/bluez-alsa #282: E: Couldn't acquire transport: GDBus.Error:org.bluez.Error.Failed: Input/output error

jpg85, 2020-01-11:
Running as root with SELinux in permissive mode. All processes running as root to assure proper permissions.

Using example aplay command: aplay -D bluealsa:SRV=org.bluealsa,DEV=48:D6:D5:CE:D2:7C,PROFILE=a2dp test.wav

Configure line:
../configure --enable-aac --enable-ldac --enable-mp3lame --enable-msbc --disable-hcitop --disable-rfcomm --enable-test --with-dbusconfdir=/etc/dbus-1 --sysconfdir=/etc --enable-mpg123 --enable-debug --enable-upower --enable-rfcomm

Log from bluealsa:
/usr/bin/bluealsa: D: ../../src/main.c:383: Starting main dispatching loop
/usr/bin/bluealsa: D: ../../src/bluez.c:1264: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
/usr/bin/bluealsa: D: ../../src/bluez.c:1264: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
/usr/bin/bluealsa: D: ../../src/bluez.c:719: Called: org.bluez.MediaEndpoint1.SelectConfiguration()
/usr/bin/bluealsa: D: ../../src/bluez.c:1264: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
/usr/bin/bluealsa: D: ../../src/bluez.c:719: Called: org.bluez.MediaEndpoint1.SetConfiguration()
/usr/bin/bluealsa: D: ../../src/bluez.c:638: A2DP Source (AAC) configured for device 48:D6:D5:CE:D2:7C
/usr/bin/bluealsa: D: ../../src/bluez.c:641: Configuration: channels: 2, sampling: 48000
/usr/bin/bluealsa: D: ../../src/ba-transport.c:715: State transition: 0 -> 0
/usr/bin/bluealsa: D: ../../src/bluez.c:764: Creating media endpoint object: /org/bluez/hci0/A2DP/MPEG24/Source/3
/usr/bin/bluealsa: D: ../../src/bluez.c:793: Registering media endpoint: /org/bluez/hci0/A2DP/MPEG24/Source/3
/usr/bin/bluealsa: D: ../../src/bluealsa-dbus.c:139: Called: org.bluealsa.Manager1.GetPCMs()
/usr/bin/bluealsa: D: ../../src/bluealsa-dbus.c:375: Called: org.bluealsa.PCM1.Open()
/usr/bin/bluealsa: E: Couldn't acquire transport: GDBus.Error:org.bluez.Error.Failed: Input/output error
/usr/bin/bluealsa: D: ../../src/bluez.c:719: Called: org.bluez.MediaEndpoint1.ClearConfiguration()
/usr/bin/bluealsa: D: ../../src/bluez.c:678: Disconnecting media endpoint: /org/bluez/hci0/A2DP/MPEG24/Source/1
/usr/bin/bluealsa: D: ../../src/ba-transport.c:312: Freeing transport: A2DP Source (AAC)
/usr/bin/bluealsa: D: ../../src/ba-transport.c:1000: Closing PCM: 10

Note that exact same setup works properly on version 2.0.0. I've reverted to that version.

I'm using Fedora 31, latest updated packages. I don't intend to debug further, just wanted to make others aware of the incompatibility.

Arkq, 2020-01-11:
> Note that exact same setup works properly on version 2.0.0. I've reverted to that version.

Will you be able to `git bisect` this issue? It would be really helpful to know after which commit master branch does not work.

jpg85, 2020-01-11:
I ran three different bisects and came to three different conclusions. I'm completely deleting the build directory and rebuilding from scratch every time, so I'm confident I'm starting from clean.

Whatever the error, I don't think I'll be able to narrow it down at the moment. I haven't had it fail on the 2.0.0 release commit yet, but have had it fail on the commit right after. Yet, I've had it work on the latest commit, so I'm not sure if it is just luck of the draw.

Arkq, 2020-01-13:
> I haven't had it fail on the 2.0.0 release commit yet, but have had it fail on the commit right after.

The commit just after 2.0.0 is not in the bluealsa server code, but in the utility (not used in your setup). So, the bug must be somewhere before 2.0.0. I suspect some timing (race) issue...

borine, 2021-06-22:
> Running as root with SELinux

May be coincidence, but the error message:
> `/usr/bin/bluealsa: E: Couldn't acquire transport: GDBus.Error:org.bluez.Error.Failed: Input/output error`

happens when bluealsa cannot obtain the unix file descriptors fromD-Bus

This is the same problem diagnosed in issues #398 and #361 as being caused by SELinux lacking some needed policy. Given that this issue has been idle for some time I'm closing it here. If new information becomes available that suggests some cause other than SELinux,  please re-open. See the referenced issues for the policy changes that worked there.

closed 2021-06-22 by borine