arkq/bluez-alsa #283: Ldac how to check?

nicosea, 2020-01-11:
I am streaming music with BlueALSA to Sony MDR-1000x headphones m, which supports Ldac codec.
I successfully installed BlueALSA with ldac module, but how can I check that BlueALSA is streaming to the headphones in Ldac?
Is there a command or a log file to check it? How?
Thank you.

Arkq, 2020-01-11:
I think that the simplest way is to see logs (but bluez-alsa has to be build with debug enabled). Optionally, you can run `gdbus call --system -d org.bluealsa -o /org/bluealsa -m org.bluealsa.Manager1.GetPCMs` (or any other application which can issue D-Bus calls, e.g. `qdbusviewer`) and look for 'Codec' value.

E.g.
```
$ gdbus call --system -d org.bluealsa -o /org/bluealsa -m org.bluealsa.Manager1.GetPCMs
({objectpath '/org/bluealsa/hci0/dev_F8_87_F1_B8_30_85/sco':   <---- SCO connection
  {'Device': <objectpath '/org/bluez/hci0/dev_F8_87_F1_B8_30_85'>,
   'Modes': <['source', 'sink']>,
   'Format': <uint16 32784>,
   'Channels': <byte 0x01>,
   'Sampling': <uint32 0>,
   'Codec': <uint16 0>,   <----- codec not selected
   'Delay': <uint16 10>,
   'Volume': <uint16 3855>},
 '/org/bluealsa/hci0/dev_F8_87_F1_B8_30_85/a2dp':    <-------- A2DP connection
  {'Device': <objectpath '/org/bluez/hci0/dev_F8_87_F1_B8_30_85'>,
   'Modes': <['sink']>,
   'Format': <uint16 32784>,
   'Channels': <byte 0x02>,
   'Sampling': <uint32 44100>,
   'Codec': <uint16 2>,        <-----------  AAC codec
   'Delay': <uint16 150>,
   'Volume': <uint16 1799>}},)
```

List of A2DP codec values: https://github.com/Arkq/bluez-alsa/blob/master/src/a2dp-codecs.h#L35, for LDAC you shall see 0x2DFF (11775).

closed 2020-01-11 by Arkq

nicosea, 2020-01-13:
I get the following:

`({objectpath '/org/bluealsa/hci0/dev_04_5D_4B_E9_2B_F3/a2dp': {'Device': <objectpath '/org/bluez/hci0/dev_04_5D_4B_E9_2B_F3'>, 'Modes': <['source']>, 'Format': <uint16 32784>, 'Channels': <byte 0x02>, 'Sampling': <uint32 48000>, 'Codec': <uint16 0>, 'Delay': <uint16 154>, 'Volume': <uint16 32639>}},)`

So no LDAC codec but it is strange because during installation no errors... I reinstalled it log enabled, how can I check the logs and see there if Ldac is used?

Arkq, 2020-01-13:
Look for log line like follows (but with LDAC):
```
D: ../../src/bluez.c:638: A2DP Source (AAC) configured for device 1C:48:F9:9D:81:5C
```

nicosea, 2020-01-13:
Sorry, but I am not so good, I understood what to look for, but how can I open the log file? Path?
where is the log file stored? And do I need to open it with cat or nano?

Arkq, 2020-01-13:
It depends how you start `bluealsa`. By default it prints logs to stderr. When launched with `-S` it writes logs to syslog. If you start bluealsa via systemd, look in journalctl, etc... It all depends :) The easiest way it is to start bluealsa manually in terminal.