arkq/bluez-alsa #284: How to decrease latency between usb microphone input and bluetooth speaker output?

FKED, 2020-01-12:
In my setup I have a microphone attached via a usb soundcard as well as a bluetooth headphone. 
I would like to feedback the incoming audio to the headset in "realtime" ideally.

The following command gives me audio feedback to the headphones however the delay is around 1 second. 
[code] ~ $ arecord -D "hw:1,0" -f S16_LE -r 44100 | aplay -f S16_LE -r 441000
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 44100 Hz, Mono
Playing WAVE 'stdin' : Signed 16 bit Little Endian, Rate 44100 Hz, Mono
[/code]

my ~/.asoundrc looks like this: 
[code]pcm.btreceiver {
    type plug
    slave.pcm {
        type bluealsa
        device "44:18:FD:C3:FA:5E"
        profile "a2dp"
    }
    hint {
        show on
        description "Bluetooth Receiver"
    }
}

pcm.!default {
    type plug
    slave.pcm "btreceiver"
}
[/code]

I've also tried the alsaloop command but I can't get it to run. It gives me this error:
[code]underrun for playback btreceiver[/code]
 
Is it possible to minimize latency via bluealsa? How to approach this? 
Or am I running into constraints of the bluetooth protocol here? 

I've also tested the bluetooth headset while watching Youtube videos and latency is actually pretty low / almost no lipsyncing. 

Many thanks!

borine, 2020-01-13:
arecord by default uses an input buffer size of 500ms, and does not start to deliver samples until that buffer is full; similarly aplay uses an output buffer of 500ms and does not begin playback until that is full. That gives a total start delay of 1s, as you have discovered. To prove that this is not a bluealsa issue, try it with an internal speaker.

You can adjust buffer size and start delay size with command line options to aplay and arecord, but realize that:
- a small buffer is required for low latency
- a large buffer is required to avoid overruns/underruns

So you have to find a compromise: zero latency is not possible in the real world.

FKED, 2020-01-13:
@borine thanks for your answer! 

I understand realtime is relative. In my application ideally I would like to stay below 200ms if possible. Right now it's impossible for me to estimate whether this is realistic or not. 

I've expimented with a wired setup vs a bluetooth setup as suggested. In the wired setup I'm plugging in the headphones to the same USB soundcard as the microphone is plugged into. 

This gives me about 100ms delay (hard to estimate but) as expected: 
`arecord -D hw:1,0 -fS16_LE -r48000 -B50000 | aplay -D plughw:1,0 -fS16_LE -r48000 -B50000`

This however gives me about a 1000ms delay: 
`arecord -D hw:1,0 -fS16_LE -r48000 -B50000 | aplay -D btreceiver -fS16_LE -r48000 -B50000`
("btreceiver" is default PCM device, but I included it for clarity sake) 

This makes me wonder if I'm approaching setting the buffer for blueasla correctly. Can I even do this directly in the command line like shown above?

Arkq, 2020-01-13:
When using A2DP, it might happen, that the delay will be increased due to headset being not responsive. Please, try SCO PCM type just to verify that. Also, you might use any player application with "pause" feature. Start playing, hit pause and see how much delay there is. Then unpause, play few seconds and pause again. If I'm correct, during the second pause, the delay will be shorter. If that's the case, there might be some workaround, e.g. use `--a2dp-keep-alive=10`. Then start playback (might be silence, anything) which will open A2DP connection, stop playback (e.g. terminate application), and finally start actual playback setup (yor arecord|aplay forwarding).

FKED, 2020-01-13:
@Arkq thanks for the tips. 

With the sco profile I can definitely get lower latency - estimating around 500ms.  
I also noticed here when starting the playback a first time the latency is higher then starting it a second time after terminating. Couldn't replicate this effect with a2dp profile. 

Will try to find a player that supports pausing. Maybe a stupid question but how do I know the delay when just playing audio? When watching a youtube video like [this](https://www.youtube.com/watch?v=ucZl6vQ_8Uo) in the desktop environment the BT headphones seem pretty much spot on. I'm guessing under the hood that's also blueasla right?

Arkq, 2020-01-18:
> Maybe a stupid question but how do I know the delay when just playing audio?

You hit "pause" and wait when the audio will actually stop playing. Then you will know how much latency there was :).

> When watching a youtube video like this in the desktop environment the BT headphones seem pretty much spot on.

First of all, are you using bluealsa for desktop? If so, then there is a delay reporting (in bluealsa), which tells browser how much delay is in the audio processing. Then, ALSA client (here browser) shall account this delay for AV sync.

borine, 2021-10-02:
This issue has been idle for over 18 months now. The delay experienced when piping between arecord and aplay can be avoided by using alsaloop. For example, with a target delay or 100 ms, try:
```
alsaloop -C hw:1,0 -f s16_le -c 2 -r 44100 -P bluealsa:44:18:FD:C3:FA:5E -t 100000
```
It may be possible to decrease the target latency (-t option) slightly, but if alsaloop cannot achieve the target it tends to fall over.

I'm closing this now as I believe that there is nothing that BlueALSA can do  to reduce delays introduced by arecord / aplay.

closed 2021-10-02 by borine