arkq/bluez-alsa #285: Detect volume changes from devices not sending MediaTransport volume updates..

juliandc1969, 2020-01-14:
Hi, 

Normally I use DBus and a modified bluealsa to catch the MediaTransport PropertiesChanged signals from connected devices ( iPad etc ) so as to catch them and use the volume level elsewhere ( for example LCD display showing current volume ). All is fine with the few iOS devices I have but Android devices like ASUS Nexus and a late model Huawei mobile don't emit any DBus volume information whatsoever. Bluealsa in debug dumps nothing for these Androids,  dbus-monitor --system shows nothing except property changes regarding transport state. 
I can however on all devices "send" DBus volume changes back to the individual device ( iPad ) so in that direction the device receives AVRCP updates and updates it personal volume control. 
So:
Huawei -> bluealsa -> rendered volume changed but no DBus volume signals      <== what I'm after..
DBus send new volume -> bluealsa -> Huawei ->  volume changed on mobile = fine!

My question is if the phone is not emitting volume changes or my version of Bluez5 is not compatible in some way is there a way to extract the volume level from another area of bluealsa or bluealsa-aplay?  The the rendered volume does change, but no volume signals are emitted from connected Android device.

Is the volume coded into each audio packet?

Thanks.

Arkq, 2020-01-18:
> Is the volume coded into each audio packet?

No, volume is not encoded in the audio packets. However, Android somehow knows that the remote device is not able to control volume, so Android handles it by itself. To be honest I don't know why... I suspect, that in order to support volume update from Android it is required to connect some additional Bluetooth profile (but I might be wrong, because I did not investigate this part thoroughly). Anyway, the volume control thingy is not "production-ready". It just works, but in order to support all kind of devices (and support capture volume native control, see https://github.com/Arkq/bluez-alsa/pull/188#issuecomment-470217238 for more details) more work will be required and currently I've got no time for that. So, some PR with solution for this issue, or some "thinking" is welcome :)

juliandc1969, 2020-01-19:
Hi
Re: Android it is required to connect some additional Bluetooth profile.. any idea what? It's not AVRCP because my device offers this profile and iOS grabs and uses it; so at least that's available. Are you suggesting some other profile Android could use to send volume updates to my board but it's not being offered by my board?

Cheers.

juliandc1969, 2020-01-23:
Turns out my later Android had Absolute Volume disabled and turning it back on resulted in all the DBus signals I need Bluealsa end. The other Android device is just too old early version 6.

Cheers.

closed 2020-01-23 by juliandc1969